evcc-io/evcc #3953: API missing "vehicleTitle" in loadpoint after restart

powelllens, 2022-07-26:
### Describe the bug

After a EVCC restart the value "vehicleTitle" will not be published / is not available. After an initial change of the connected car (via auto car detect or manual) the value will appear and stay in the API. Even after resetting the value to "Gastfahrzeug" the value will remain but stay empty (which is most likely the intended behavior).

This might be intentional but could result in bad or unexpected behavior for API users.

Overall the behavior is not consistent, if no data is available the value should either not be published or it should always be published and stay empty (vehiclesTitle = "").

Same behavior for:
- vehicleCapacity
- vehicleOdometer
- vehiclePresent
- vehicleRange
- vehicleSoC
- vehicleTitle

It should be mentioned that the error-handling should always be implemented on the API user side.

### Steps to reproduce

Restart EVCC, check API - Value will be missing.
`(...),"targetSoC":100,"title":"Hof","vehicles":["(...)`
Change the Car and the Value will appear and remain online until EVCC will be restarted.
`"targetSoC":80,"title":"Hof","vehicleCapacity":0,"vehicleOdometer":0,"vehiclePresent":false,"vehicleRange":0,"vehicleSoC":0,"vehicleTitle":"","vehicles":["`

### Configuration details

- not relevant

### Log details

- not relevant

### What type of operating system are you running?

Docker container

### Version

0.97

andig, 2022-07-27:
> After a EVCC restart the value "vehicleTitle" will not be published / is not available.

Is the vehicle even detected? If not it can't be published. Logfile?

powelllens, 2022-07-27:
After the restart the car is not detected and the API value is not published. If a car is detected the API will publish the name but this value remains "online"/"published" even if the car is disconnected. Value remains `vehicleTitle=""`

After restart, no car detected, no values published (as expected):
```
{
  "result": {
    "auth": {
      "vehicles": {}
    },
    "batteryConfigured": false,
    "bufferSoC": 0,
    "currency": "EUR",
    "gridConfigured": true,
    "gridPower": 0,
    "homePower": 0,
    "loadpoints": [
      {
        "activePhases": 3,
        "chargeCurrent": 0,
        "chargeCurrents": [
          0,
          0,
          0
        ],
        "chargeDuration": 0,
        "chargePower": 0,
        "chargedEnergy": 0,
        "charging": false,
        "connected": false,
        "connectedDuration": 0,
        "enabled": false,
        "loadpoint": 1,
        "maxCurrent": 20,
        "minCurrent": 6,
        "minSoC": 0,
        "mode": "off",
        "phases": 3,
        "targetSoC": 100,
        "title": "Hof",
        "vehicles": [
          "car1",
          "car2"
        ]
      }
    ],
    "prioritySoC": 0,
    "pvConfigured": false,
    "pvPower": 0,
    "residualPower": 0,
    "savingsSince": 1658915536,
    "siteTitle": "Zuhause",
    "sponsor": "max.mustermann@gmail.com",
    "tariffFeedIn": 0,
    "tariffGrid": 0
  }
}

```

After one car have been detected and removed:
(Car information remain published even if content is empty)

```{
  "result": {
    "auth": {
      "vehicles": {}
    },
    "batteryConfigured": false,
    "bufferSoC": 0,
    "currency": "EUR",
    "gridConfigured": true,
    "gridPower": 0,
    "homePower": 0,
    "loadpoints": [
      {
        "activePhases": 3,
        "chargeCurrent": 0,
        "chargeCurrents": [
          0,
          0,
          0
        ],
        "chargeDuration": 0,
        "chargePower": 0,
        "chargeRemainingDuration": -1,
        "chargedEnergy": 0,
        "charging": false,
        "connected": false,
        "connectedDuration": 0,
        "enabled": false,
        "loadpoint": 1,
        "maxCurrent": 16,
        "minCurrent": 7,
        "minSoC": 20,
        "mode": "minpv",
        "phases": 3,
        "pvAction": "inactive",
        "pvRemaining": 0,
        "targetSoC": 80,
        "title": "Hof",
        "vehicleCapacity": 0,
        "vehicleOdometer": 0,
        "vehiclePresent": false,
        "vehicleRange": 0,
        "vehicleSoC": 0,
        "vehicleTitle": "",
        "vehicles": [
          "car1",
          "car2"
        ]
      }
    ],
    "prioritySoC": 0,
    "pvConfigured": false,
    "pvPower": 0,
    "residualPower": 0,
    "savingsSince": 1658867801,
    "siteTitle": "Zuhause",
    "sponsor": "max.mustermann@gmail.com",
    "tariffFeedIn": 0,
    "tariffGrid": 0
  }
}```

andig, 2022-07-27:
> After the restart the car is not detected

Please retry on master, should be fixed.

closed 2022-07-27 by andig