evcc-io/evcc #3956: Fahrzeug wird nicht erkannt seit Version 0.97

Robbe64, 2022-07-27:
### Describe the bug

Seit ich die Version 0.97 nutze wird das Fahrzeug nicht mehr automatisch erkannt. Ich kann es dank des neuen Features zwar manuell identifiezieren aber automatisch geht es nicht mehr.
Erst wenn ich EVCC neu starte wird das Fahrzeug erkannt.

### Steps to reproduce

Ich nutze die Identifizierung mit RFID-Karte bei der Easee.
Anschließend muss ich nur das Auto anschließen.

### Configuration details

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/9195957/evcc.yaml.txt)

### Log details

Ohne Neustart -> Auto wird nicht identifiziert

[site  ] DEBUG 2022/07/27 08:30:53 ----,
[lp-1  ] DEBUG 2022/07/27 08:30:53 charge power: 0W,
[site  ] DEBUG 2022/07/27 08:30:53 pv power: 1794W,
[site  ] DEBUG 2022/07/27 08:30:53 battery power: 6W,
[site  ] DEBUG 2022/07/27 08:30:53 grid power: -1462W,
[site  ] DEBUG 2022/07/27 08:30:53 battery soc 0: 37%,
[site  ] DEBUG 2022/07/27 08:30:53 site power: -1356W,
[lp-1  ] DEBUG 2022/07/27 08:30:53 charge currents: [0.008 0.008 0.008]A,
[lp-1  ] DEBUG 2022/07/27 08:30:53 charger status: B,
[lp-1  ] INFO 2022/07/27 08:30:53 car connected,
[lp-1  ] DEBUG 2022/07/27 08:30:53 vehicle api refresh,
[lp-1  ] DEBUG 2022/07/27 08:30:53 pv timer elapse,
[lp-1  ] DEBUG 2022/07/27 08:30:53 pv timer inactive,
[lp-1  ] WARN 2022/07/27 08:30:53 charger out of sync: expected disabled, got enabled,
[lp-1  ] DEBUG 2022/07/27 08:30:53 pv charge current: 5.9A = 0A + 5.9A (-1356W @ 1p),
[lp-1  ] DEBUG 2022/07/27 08:30:53 pv enable timer reset,
[lp-1  ] DEBUG 2022/07/27 08:30:53 pv timer inactive,
[push  ] DEBUG 2022/07/27 08:30:54 telegram: sending to ...,
[site  ] DEBUG 2022/07/27 08:31:03 ----,
[lp-1  ] DEBUG 2022/07/27 08:31:03 charge power: 0W,
[site  ] DEBUG 2022/07/27 08:31:03 pv power: 1813W,
[site  ] DEBUG 2022/07/27 08:31:03 battery power: 6W,
[site  ] DEBUG 2022/07/27 08:31:03 grid power: -1433W,
[site  ] DEBUG 2022/07/27 08:31:03 battery soc 0: 37%,
[site  ] DEBUG 2022/07/27 08:31:03 site power: -1327W,
[lp-1  ] DEBUG 2022/07/27 08:31:03 charge currents: [0.008 0.008 0.008]A,
[lp-1  ] DEBUG 2022/07/27 08:31:03 charger status: B,
[lp-1  ] DEBUG 2022/07/27 08:31:03 pv charge current: 5.77A = 0A + 5.77A (-1327W @ 1p),
[site  ] DEBUG 2022/07/27 08:31:13 ----

Mit Neustart -> Auto wird identifiziert

[site  ] DEBUG 2022/07/27 08:51:23 ----,
[lp-1  ] DEBUG 2022/07/27 08:51:23 charge power: 1371W,
[site  ] DEBUG 2022/07/27 08:51:23 pv power: 2008W,
[site  ] DEBUG 2022/07/27 08:51:23 battery power: 6W,
[site  ] DEBUG 2022/07/27 08:51:23 grid power: -376W,
[site  ] DEBUG 2022/07/27 08:51:23 battery soc 0: 37%,
[site  ] DEBUG 2022/07/27 08:51:23 site power: -270W,
[lp-1  ] DEBUG 2022/07/27 08:51:23 charge currents: [6.04 0.015 0.008]A,
[lp-1  ] DEBUG 2022/07/27 08:51:23 detected phases: 1p,
[lp-1  ] DEBUG 2022/07/27 08:51:23 charger status: C,
[lp-1  ] DEBUG 2022/07/27 08:51:23 pv charge current: 7.17A = 6A + 1.17A (-270W @ 1p),
[lp-1  ] DEBUG 2022/07/27 08:51:23 max charge current: 7A,
[main  ] INFO 2022/07/27 08:51:34 evcc 0.97,
[main  ] INFO 2022/07/27 08:51:34 using config file /etc/evcc.yaml,
[main  ] INFO 2022/07/27 08:51:34 listening at :7070,
[site  ] INFO 2022/07/27 08:51:42 site config:,
[site  ] INFO 2022/07/27 08:51:42   meters:      grid ✓ pv ✓ battery ✓,
[site  ] INFO 2022/07/27 08:51:42     grid:      power ✓ energy ✗ currents ✗,
[site  ] INFO 2022/07/27 08:51:42     pv 1:      power ✓ energy ✗ currents ✗,
[site  ] INFO 2022/07/27 08:51:42     battery 1: power ✓ energy ✗ currents ✗ soc ✓,
[lp-1  ] INFO 2022/07/27 08:51:42 loadpoint 1:,
[lp-1  ] INFO 2022/07/27 08:51:42   mode:        pv,
[lp-1  ] INFO 2022/07/27 08:51:42   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗,
[lp-1  ] INFO 2022/07/27 08:51:42   meters:      charge ✓,
[lp-1  ] INFO 2022/07/27 08:51:42     charge:    power ✓ energy ✗ currents ✓,
[lp-1  ] INFO 2022/07/27 08:51:42   vehicles:    ✓,
[lp-1  ] INFO 2022/07/27 08:51:42     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗,
[lp-1  ] INFO 2022/07/27 08:51:42     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗,
[lp-1  ] DEBUG 2022/07/27 08:51:42 max charge current: 6A,
[site  ] DEBUG 2022/07/27 08:51:42 ----,
[lp-1  ] DEBUG 2022/07/27 08:51:42 charge power: 1371W,
[site  ] DEBUG 2022/07/27 08:51:42 pv power: 2044W,
[site  ] DEBUG 2022/07/27 08:51:42 battery power: 6W,
[site  ] DEBUG 2022/07/27 08:51:42 grid power: -384W,
[site  ] DEBUG 2022/07/27 08:51:43 battery soc 0: 37%,
[site  ] DEBUG 2022/07/27 08:51:43 site power: -278W,
[lp-1  ] DEBUG 2022/07/27 08:51:43 charge currents: [6.04 0.015 0.008]A,
[lp-1  ] DEBUG 2022/07/27 08:51:43 charger status: C,
[lp-1  ] INFO 2022/07/27 08:51:43 start charging ->,
[lp-1  ] DEBUG 2022/07/27 08:51:43 charger vehicle id: 04DA682ABD7280,
[lp-1  ] INFO 2022/07/27 08:51:43 vehicle updated: unknown -> eGolf,
[lp-1  ] DEBUG 2022/07/27 08:51:43 set min current: 7,
[lp-1  ] DEBUG 2022/07/27 08:51:43 set max current: 16,
[lp-1  ] DEBUG 2022/07/27 08:51:43 set target soc: 90,
[lp-1  ] DEBUG 2022/07/27 08:51:43 vehicle status: C,
[lp-1  ] DEBUG 2022/07/27 08:51:43 vehicle soc: 61%,
[lp-1  ] DEBUG 2022/07/27 08:51:43 vehicle range: 77km,
[lp-1  ] DEBUG 2022/07/27 08:51:43 pv charge current: 7.21A = 6A + 1.21A (-278W @ 1p),
[push  ] DEBUG 2022/07/27 08:51:43 telegram: sending to ...,
[lp-1  ] DEBUG 2022/07/27 08:51:43 climater active: false, target temp: 15.5°C, outside temp: 0.0°C,
[lp-1  ] DEBUG 2022/07/27 08:51:43 max charge current: 7A,
[site  ] DEBUG 2022/07/27 08:51:43 ----

### What type of operating system are you running?

Docker container

### Version

0.97

rivengh, 2022-07-27:
Hatte bei mir auch das gleiche Problem. Bereite gerade den Pull request mit dem Fix vor.

Ursache ist diese Condition hier, die immer `True` ist, wenn kein default vehicle konfiguriert ist:

https://github.com/evcc-io/evcc/blob/55910059edb59cd48d50b59d1bad10c86f9e0784/core/loadpoint.go#L421-L424

andig, 2022-07-27:
Fix ist drin- probiert ihr mal master? Hab das bei mir auch entdeckt und gleich noch ein paar Tests ergänzt...

closed 2022-07-27 by andig

rivengh, 2022-07-27:
LGTM

Robbe64, 2022-07-28:
Kann ich bestätigen. In dem nightly funktioniert es wieder.