evcc-io/evcc #3977: % Charge not refreshed when charging ends

mlnoga, 2022-07-28:
### Describe the bug

When instructing evcc to charge until 80% SoC is reached, it faithfully does so. However, while the vehicle itself then displays 80% SoC on the dash, the evcc web interface continues to display 79%. 

Suggested fix: poll the vehicle SoC one more time when charging ends (instead of waiting for `interval` to elapse). 

### Steps to reproduce

Connect car with SoC below 80%, and and wait for it to complete charging. 

### Configuration details

Here's the relevant excerpt:

```template: id
mode: pv
targetSoC: 80

estimate : false```

Not reproducing the full file here, as it contains login details.

### Log details

No errors, just outdated values being displayed. 

### What type of operating system are you running?

Linux

### Version

evcc version 0.97.1

andig, 2022-07-28:
@mlnoga hast Du das Limit im Auto gesetzt oder in evcc? War das verhalten schon immer so oder ist es neu?

/cc @premultiply

mlnoga, 2022-07-28:
Ich habe das Limit im Fahrzeug und im evcc gesetzt. 

Kann leider nicht sagen, seit wann das Verhalten auftritt. Auf meinem System hat der Abruf des Ladestands seit einigen Updates nicht funktioniert. Ich habe erst heute früh gemerkt,  dass ich meine evcc.yaml von Hand auf den Template-Mechanismus umstellen muss. Seitdem funktioniert der Abruf wieder. In der Mittagspause habe ich dann den Schiefstand um 1% bei Ladeende bemerkt.

premultiply, 2022-07-28:
Dann bitte das Limit im Auto auf 100% setzen.

andig, 2022-07-29:
> Ich habe das Limit im Fahrzeug und im evcc gesetzt.

@mlnoga das kann nicht funktionieren, da der Ladevorgang aus Sicht evcc grundlos stopp.t Soc aktualisieren machen wir nur beim Laden.

@premultiply Eine Aktualisierung bei jedem Ladestopp den *wir* nicht verursacht haben wäre noch eine Option (einschl. vorherigem API Refresh)- aber woran erkennen wir die?

andig, 2022-07-29:
Idee Logik: wenn Übergang Status C->B und Wallbox "Enabled", dann 1x SoC Update einschl. API-Refresh.

premultiply, 2022-08-10:
Problem: Wann genau erfolgt dieses SoC-Update mit Refresh bzw. wann liefert es endlich aktuelle Daten?
Da bräuchte man eigentlich auch wieder mehrere Versuche mehrere Minuten lang wie bei der Fahrzeugerkennung...

premultiply, 2022-08-10:
Mutmaßlich dürfte das in den meisten Fällen (bei asynchroner API) dann zu keinem Mehrwert führen weil die Anfrage zu früh kommt bevor die API neue Daten vom Fahrzeug erhalten hat.

premultiply, 2022-08-23:
Aber einen Versuch könnte es trotzdem Wert sein.

andig, 2022-09-01:
Estimator is now enabled by default. Let's see if that helps.

closed 2022-09-01 by andig