evcc-io/evcc #3981: Ladedauer im UI läuft trotz "aus"

andig, 2022-07-28:
Loadpoint steht auf "Aus".
Fahrzeug wird angesteckt.
Loadpoint bleibt "Aus" (soll so sein)
"Dauer" im UI fängt an zu zählen.

War das schon immer so?
Bug or feature?

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/3980_

VolkerK62, 2022-07-28:
Noch als Ergänzung:
Wenn ich dann im angesteckten Zustand `restart evcc` mache, bleibt die Dauer danach bei `--` und fängt nicht an zu zählen.

Tichu78, 2022-10-16:
Gleiches Problem bei mir egal in welchem mode. v.0.104.2

Tichu78, 2022-10-17:
### core\loadpoint.go
```
func (lp *LoadPoint) publishChargeProgress() {
...
lp.publish("chargeDuration", lp.chargeDuration)
...
}
```
called by
```
func (lp *LoadPoint) Update(sitePower float64, cheap, batteryBuffered bool) {
...
// update progress and soc before status is updated
lp.publishChargeProgress()
...
}
```
called by
### **core\site.go**
```
func (site *Site) update(lp Updater) {
...
if sitePower, err := site.sitePower(totalChargePower); err == nil {
  lp.Update(sitePower, cheap, site.batteryBuffered)
...
}
```
called by
```
// Run is the main control loop. It reacts to trigger events by
// updating measurements and executing control logic.
func (site *Site) Run(stopC chan struct{}, interval time.Duration) {
...
	for {
		select {
		case <-ticker.C:
			site.update(<-loadpointChan)
		case lp := <-site.lpUpdateChan:
			site.update(lp)
		case <-stopC:
			return
		}
	}
}
```
Is it possible that changing the vehicleDetectTicker logic in the loadpoint triggers the lpUpdateChan to update the site causing into updating the chargeDuration at the end?

andig, 2022-11-10:
Closed in https://github.com/evcc-io/evcc/commit/1d51100c1ebba22b36a9d3c62ae186fa717fb415. Bitte morgen im Nightly verifizieren. Es scheint als wäre hier ein lange bestehender Bug durch Änderung der umgebenden Logik ans Tageslicht gekommen.

closed 2022-11-10 by andig