evcc-io/evcc #4008: OCPP - cannot create charger 'ocpp': configuration change rejected

Hofyyy, 2022-07-31:
### Describe the bug

If I configure the "meter = true" for a Wallbe Pro I am not able to use OCPP.
If I do not use meter all is fine.

### Steps to reproduce

  - name: wallbe
    type: ocpp # Wallbe charger
    stationid: wallbeP9741 # ModBus address
    idtag: 1010
    meter: true

### Configuration details

  - name: wallbe
    type: ocpp # Wallbe charger
    stationid: wallbeP9741 # ModBus address
    idtag: 1010
    meter: true

### Log details

XXX@Sebastians-MacBook-Air evcc_0.98_macOS_all % ./evcc
[main  ] INFO 2022/07/31 19:10:49 evcc 0.98
[main  ] INFO 2022/07/31 19:10:49 using config file /Users/XXX/Downloads/evcc_0.98_macOS_all/evcc.yaml
[main  ] INFO 2022/07/31 19:10:49 listening at :7070
[ocpp  ] DEBUG 2022/07/31 19:12:09 chargepoint connected: wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:09 handling incoming CALL [1, BootNotification] from wallbeP9741
[ocpp-cp] TRACE 2022/07/31 19:12:09 *core.BootNotificationRequest: &{ChargeBoxSerialNumber: ChargePointModel:Smart Charge Control ChargePointSerialNumber:00000009741 ChargePointVendor:wallbe FirmwareVersion:BE-03.01.29 Iccid: Imsi: MeterSerialNumber: MeterType:}
[ocpp  ] TRACE 2022/07/31 19:12:09 sent CALL RESULT [1] for wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:09 handling incoming CALL [2, StatusNotification] from wallbeP9741
[ocpp-cp] TRACE 2022/07/31 19:12:09 *core.StatusNotificationRequest: &{ConnectorId:0 ErrorCode:NoError Info: Status:Available Timestamp:2022-07-31 19:12:09 +0200 CEST VendorId: VendorErrorCode:0}
[ocpp  ] TRACE 2022/07/31 19:12:09 sent CALL RESULT [2] for wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:09 enqueued CALL [130530088, GetConfiguration] for wallbeP9741
[ocpp  ] DEBUG 2022/07/31 19:12:09 dispatched request 130530088 for wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:09 started timeout timer for wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:10 handling incoming CALL RESULT [130530088] from wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:10 completed request 130530088 for wallbeP9741
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 AuthorizeRemoteTxRequests (true): true
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ClockAlignedDataInterval (true): 0
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ConnectionTimeOut (false): 60
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 GetConfigurationMaxKeys (true): 0
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 HeartbeatInterval (false): 60
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 LocalAuthorizeOffline (true): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 LocalPreAuthorize (true): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 MeterValuesAlignedData (true): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 MeterValuesSampledData (true): Voltage.L1,Voltage.L2,Voltage.L3,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Import.Register,Power.Active.Import
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 StopTransactionOnEVSideDisconnect (true): true
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 StopTransactionOnInvalidId (true): true
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 StopTxnAlignedData (true): false
[ocpp  ] TRACE 2022/07/31 19:12:10 wallbeP9741 ready to transmit again
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 StopTxnSampledData (true): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 MeterValueSampleInterval (false): 300
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 NumberOfConnectors (true): 1
[ocpp  ] TRACE 2022/07/31 19:12:10 timeout canceled for wallbeP9741
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ConnectorPhaseRotation (true): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 SupportedFeatureProfiles (true): Core,Reservation,SmartCharging,RemoteTrigger
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ResetRetries (false): 3
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 TransactionMessageAttempts (false): 3
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 TransactionMessageRetryInterval (false): 60
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 UnlockConnectorOnEVSideDisconnect (true): true
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ChargeProfileMaxStackLevel (true): 1
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ChargingScheduleAllowedChargingRateUnit (true): Current
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ChargingScheduleMaxPeriods (true): 1
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 MaxChargingProfilesInstalled (true): 1
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 PricePerKwh (false): 0
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 AllowOfflineTxForFreemode (true): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 WebSocketPingInterval (false): 60
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 RFID_ShowInReverseOrder (false): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 RFID_HighRegisterFirst (false): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ConnectorMaximumCurrent (false): 32
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ConnectorMinimumCurrent (false): 13
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 GlobalMaximumCurrent (false): 16
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 MeterValueSendTotalEnergy (false): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 BootloaderVersion (true): B-3.03
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 AllowOfflineTxForUnknownId (false): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 OCPPSocketEncryption (false): false
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 OCPPSocketServer (false): 192.168.20.144
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 OCPPSocketPort (false): 8887
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 OCPPSocketPath (false): wallbeP9741
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 APNName (false): internet.telekom
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 APNUser (false): telekom
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 APNPassword (false): telekom
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 TimeZone (false): 
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 TimeZonePOSIX (false): 
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 EMOC_PublicKey_Conn_1 (true): unknown
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 Current values 
	Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Temperature != 
	Voltage.L1,Voltage.L2,Voltage.L3,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Import.Register,Power.Active.Import
[ocpp  ] TRACE 2022/07/31 19:12:10 enqueued CALL [986991204, ChangeConfiguration] for wallbeP9741
[ocpp  ] DEBUG 2022/07/31 19:12:10 dispatched request 986991204 for wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:10 started timeout timer for wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:10 handling incoming CALL [3, StatusNotification] from wallbeP9741
[ocpp-cp] TRACE 2022/07/31 19:12:10 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Available Timestamp:2022-07-31 19:12:09.001 +0200 CEST VendorId: VendorErrorCode:0}
[ocpp  ] TRACE 2022/07/31 19:12:10 sent CALL RESULT [3] for wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:10 handling incoming CALL RESULT [986991204] from wallbeP9741
[ocpp  ] TRACE 2022/07/31 19:12:10 completed request 986991204 for wallbeP9741
[ocpp-wallbeP9741:1] TRACE 2022/07/31 19:12:10 ChangeMeterConfigurationRequest *core.ChangeConfigurationConfirmation: &{Status:Rejected}
[main  ] FATAL 2022/07/31 19:12:10 cannot create charger 'wallbe': cannot create charger 'ocpp': configuration change rejected
[ocpp  ] TRACE 2022/07/31 19:12:10 wallbeP9741 ready to transmit again

### What type of operating system are you running?

Docker container

### Version

0.98

Hofyyy, 2022-07-31:
@xantalor:
Wie besprochen.
Ich brauche deinen Fix :-P

Hofyyy, 2022-07-31:
Der Fehler besteht seit den letzten Versionen und ist schon besprochen und getestet. Hat nichts mit der 0.98 Release zu tun. Hab nur geschaut ob er durch Zufall mit den anderen OCPP Patches verschwunden ist, oder sich verändert hat.

andig, 2022-08-01:
> Der Fehler 

Was ist hier der Fehler? Es scheint, als würden wir die Meßwerte nicht flexibel genug verarbeiten und die Box wehrt sich gegen Umkonfiguration?

    Current values
    Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Temperature !=
    Voltage.L1,Voltage.L2,Voltage.L3,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Import.Register,Power.Active.Import

@xantalor es scheint als würde uns auch noch die Möglichkeit fehlen, ausgehende Werte zu loggen?

andig, 2022-08-01:
Wenn Du kannst gerne PR ausprobieren, damit sollte es gehen!

Hofyyy, 2022-08-01:
@andig
kannst du den taggen? oder wie komme ich an ein macos binary? sorry ich hab noch keine dev umgebung.

andig, 2022-08-01:
Da hilft nur auschecken und `make`. Is auf MacOS supereinfach.

Hofyyy, 2022-08-01:
> Da hilft nur auschecken und `make`. Is auf MacOS supereinfach. Go-Installation mit `homebrew`.

Dein Wort in Gottes ohr. gucke ich mit an. #M1 Mac

Hofyyy, 2022-08-01:
Also ohne Auto sieht es so aus:

[main  ] INFO 2022/08/01 19:00:11 evcc 0.98 (c9bd7ae3)
[main  ] INFO 2022/08/01 19:00:11 using config file /Users/sebastianhofmann/Documents/GitHub/evcc/evcc.yaml
[main  ] INFO 2022/08/01 19:00:11 listening at :7070
[ocpp  ] DEBUG 2022/08/01 19:00:49 chargepoint connected: wallbeP9741
[ocpp  ] DEBUG 2022/08/01 19:00:49 dispatched request 1028764257 for wallbeP9741
[ocpp  ] DEBUG 2022/08/01 19:00:50 dispatched request 2481399560 for wallbeP9741
[lp-1  ] WARN 2022/08/01 19:00:52 poll mode '{connected 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] INFO 2022/08/01 19:00:52 site config:
[site  ] INFO 2022/08/01 19:00:52   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/08/01 19:00:52     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/08/01 19:00:52     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/08/01 19:00:52     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/08/01 19:00:52 loadpoint 1:
[lp-1  ] INFO 2022/08/01 19:00:52   mode:        minpv
[lp-1  ] INFO 2022/08/01 19:00:52   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/08/01 19:00:52   meters:      charge ✓
[lp-1  ] INFO 2022/08/01 19:00:52     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2022/08/01 19:00:52   vehicles:    ✓
[lp-1  ] INFO 2022/08/01 19:00:52     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/08/01 19:00:52 vehicle updated: unknown -> unavailable
[site  ] DEBUG 2022/08/01 19:00:52 ----
[lp-1  ] ERROR 2022/08/01 19:00:52 charge meter: not available
[site  ] DEBUG 2022/08/01 19:00:52 pv power: 128W
[site  ] DEBUG 2022/08/01 19:00:52 battery power: -77W
[site  ] DEBUG 2022/08/01 19:00:52 grid power: 1071W
[site  ] DEBUG 2022/08/01 19:00:52 battery soc 0: 78%
[site  ] DEBUG 2022/08/01 19:00:52 site power: 994W
**[lp-1  ] ERROR 2022/08/01 19:00:52 charge meter: not available**
[lp-1  ] DEBUG 2022/08/01 19:00:52 charger status: A
[site  ] DEBUG 2022/08/01 19:01:02 ----

Hofyyy, 2022-08-01:
Trace
[trace.txt](https://github.com/evcc-io/evcc/files/9235856/trace.txt)

andig, 2022-08-01:
Mhhm:

```
[ocpp  ] TRACE 2022/08/01 19:04:12 enqueued CALL [3655570544, TriggerMessage] for wallbeP9741
[ocpp  ] DEBUG 2022/08/01 19:04:12 dispatched request 3655570544 for wallbeP9741
[ocpp  ] TRACE 2022/08/01 19:04:12 started timeout timer for wallbeP9741
[ocpp  ] TRACE 2022/08/01 19:04:12 handling incoming CALL [3, StatusNotification] from wallbeP9741
[ocpp-cp] TRACE 2022/08/01 19:04:12 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Available Timestamp:2022-08-01 19:04:12.001 +0200 CEST VendorId: VendorErrorCode:0}
[ocpp  ] TRACE 2022/08/01 19:04:12 sent CALL RESULT [3] for wallbeP9741
[ocpp  ] TRACE 2022/08/01 19:04:12 TriggerMessageRequest *remotetrigger.TriggerMessageConfirmation: <nil>
[ocpp  ] DEBUG 2022/08/01 19:04:12 chargepoint disconnected: wallbeP9741
[ocpp  ] TRACE 2022/08/01 19:04:12 timeout canceled for wallbeP9741
```

Keine Ahnung welcher Timeout da in der gleichen Sekunde wieder abläuft? Finden wir vielleicht mal eine Chance uns abends per Email zu einer VSCode Live Share Session zu verabreden wo ich mal remote bei Dir debuggen kann?

xantalor, 2022-08-01:
Ich würde sagen, das Problem liegt daran, dass die Wallbe nicht erlaubt die Meter Values zu setzen, dafür muss ich was umbauen. Kam leider noch net dazu. @Hofyyy können gerne mal ne Session machen.

andig, 2022-08-01:
@xantalor der Umbau ist schon da- die liefert ja alle Werte, nur gibt sie eine andere Reihenfolge an. Ich glaub daran liegts nicht.

xantalor, 2022-08-01:
> @xantalor der Umbau ist schon da- die liefert ja alle Werte, nur gibt sie eine andere Reihenfolge an. Ich glaub daran liegts nicht.

Ah hatte ich noch nicht gesehen. Aktuell etwas busy 😬

andig, 2022-08-05:
Erkenntnis: Box liefert nur Zählerwerte wenn Auto angeschlossen ist

closed 2022-09-11 by andig