evcc-io/evcc #4014: BMW - Odometer in Meilen statt Kilometer

blenni, 2022-07-31:
### Describe the bug

Bei BMW wird Laufstrecke in Meilen statt in Kilometer abgerufen:

```
pi@raspberrypi:~ $ evcc vehicle
[main  ] INFO 2022/07/31 21:20:56 evcc 0.98
[main  ] INFO 2022/07/31 21:20:56 using config file /etc/evcc.yaml
[mqtt  ] INFO 2022/07/31 21:20:56 connecting evcc-12345 at tcp://192.168.178.37:1883
[mqtt  ] DEBUG 2022/07/31 21:20:59 tcp://192.168.178.37:1883 connected

i3
--
SoC:           81%
Charge status: A
Range:         263km
Odometer:      6433km
Capacity:      39kWh
```
[image]
[image]

### Steps to reproduce

`evcc vehicle` mit einem BMW-Fahrzeug

### Configuration details

```
vehicles:
- name: i3
  type: bmw
  title: BMW i3s
  capacity: 39 # kWh
  user: ...
  password: ...
  vin: ...
```

### Log details

```
Jul 31 21:16:45 raspberrypi systemd[1]: Started evcc.
Jul 31 21:16:50 raspberrypi evcc[29199]: [lp-2  ] WARN 2022/07/31 21:16:50 poll interval '45m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:16:51 loadpoint 2:
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:16:51   mode:        pv
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:16:51   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:16:51   meters:      charge ✓
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:16:51     charge:    power ✓ energy ✓ currents ✓
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:16:51   vehicles:    ✓
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:16:51     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:16:51 vehicle updated: unknown -> BMW i3s
Jul 31 21:16:51 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:16:51 charge power: 0W
Jul 31 21:17:01 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:01 charge power: 0W
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 vehicle odometer: 6433km
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 charge currents: [0 0 0]A
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 charger status: B
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] INFO 2022/07/31 21:17:04 car connected
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 pv timer elapse
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 pv timer inactive
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 vehicle status: A
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 vehicle soc: 81%
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 vehicle range: 263km
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 pv charge current: 0A = 0A + -4.19A (964W @ 1p)
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 pv enable timer reset
Jul 31 21:17:04 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:04 pv timer inactive
Jul 31 21:17:11 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:11 charge power: 0W
Jul 31 21:17:21 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:21 charge power: 0W
Jul 31 21:17:21 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:21 vehicle odometer: 6433km
Jul 31 21:17:21 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:21 charge currents: [0 0 0]A
Jul 31 21:17:21 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:21 charger status: B
Jul 31 21:17:21 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:21 pv charge current: 0A = 0A + -4.06A (934W @ 1p)
Jul 31 21:17:31 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:31 charge power: 0W
Jul 31 21:17:41 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:41 charge power: 0W
Jul 31 21:17:41 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:41 charge currents: [0 0 0]A
Jul 31 21:17:41 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:41 charger status: B
Jul 31 21:17:41 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:41 pv charge current: 0A = 0A + -4.04A (928W @ 1p)
Jul 31 21:17:51 raspberrypi evcc[29199]: [lp-2  ] DEBUG 2022/07/31 21:17:51 charge power: 0W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.98

andig, 2022-08-01:
Aber Range stimmt? Dann gibt BMW da wohl ein Sammelsurium an Einheiten zurück, ist schnell korrigiert.

andig, 2022-08-01:
Closed in https://github.com/evcc-io/evcc/commit/47287956a62e141a05d844d4b31a9b462df7e47a

closed 2022-08-01 by andig

blenni, 2022-08-01:
> Aber Range stimmt? Dann gibt BMW da wohl ein Sammelsurium an Einheiten zurück, ist schnell korrigiert.

Ja, die Reichweite stimmt.

Danke :)