evcc-io/evcc #4041: Fahrzeug wird sporadisch nicht erkannt

Robbe64, 2022-08-04:
### Describe the bug

Gelegentlich wird das Fahrzeug nicht erkannt. Ich konnte noch keine Regemäßigkeit erkennen. Nach einem Neustart wird es jedoch erkannt.

### Steps to reproduce

?

### Configuration details

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/9257559/evcc.yaml.txt)

### Log details

[site  ] DEBUG 2022/08/04 09:00:00 ----
[lp-1  ] DEBUG 2022/08/04 09:00:00 charge power: 0W
[site  ] DEBUG 2022/08/04 09:00:00 pv power: 3017W
[site  ] DEBUG 2022/08/04 09:00:01 battery power: 5W
[site  ] DEBUG 2022/08/04 09:00:01 grid power: -2759W
[site  ] DEBUG 2022/08/04 09:00:01 battery soc 0: 34%
[site  ] DEBUG 2022/08/04 09:00:01 site power: -2654W
[lp-1  ] DEBUG 2022/08/04 09:00:01 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/08/04 09:00:01 charger status: A
[site  ] DEBUG 2022/08/04 09:00:30 ----
[lp-1  ] DEBUG 2022/08/04 09:00:30 charge power: 0W
[site  ] DEBUG 2022/08/04 09:00:30 pv power: 3303W
[site  ] DEBUG 2022/08/04 09:00:30 battery power: 6W
[site  ] DEBUG 2022/08/04 09:00:30 grid power: -3024W
[site  ] DEBUG 2022/08/04 09:00:31 battery soc 0: 34%
[site  ] DEBUG 2022/08/04 09:00:31 site power: -2918W
[lp-1  ] DEBUG 2022/08/04 09:00:31 charge currents: [1.22 0.009 0.009]A
[lp-1  ] DEBUG 2022/08/04 09:00:31 charger status: C
[lp-1  ] INFO 2022/08/04 09:00:31 start charging ->
[lp-1  ] WARN 2022/08/04 09:00:31 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2022/08/04 09:00:31 available power 2918W < 4140W min 3p threshold
[lp-1  ] DEBUG 2022/08/04 09:00:31 start phase scale1p timer
[lp-1  ] DEBUG 2022/08/04 09:00:31 phase scale1p in 7m0s
[lp-1  ] DEBUG 2022/08/04 09:00:31 pv charge current: 7.45A = 3.22A + 4.23A (-2918W @ 3p)
[lp-1  ] DEBUG 2022/08/04 09:00:31 site power -2918W <= 0W enable threshold
[lp-1  ] DEBUG 2022/08/04 09:00:31 pv enable timer start: 3m0s
[lp-1  ] DEBUG 2022/08/04 09:00:31 pv enable in 3m0s
[push  ] DEBUG 2022/08/04 09:00:31 telegram: sending to ...
[site  ] DEBUG 2022/08/04 09:01:00 ----

Nach einem Neustart:

[main  ] INFO 2022/08/04 09:28:15 evcc 0.98
[main  ] INFO 2022/08/04 09:28:15 using config file /etc/evcc.yaml
[main  ] INFO 2022/08/04 09:28:15 listening at :7070
[site  ] INFO 2022/08/04 09:28:23 site config:
[site  ] INFO 2022/08/04 09:28:23   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/08/04 09:28:23     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/08/04 09:28:23     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/08/04 09:28:23     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/08/04 09:28:23 loadpoint 1:
[lp-1  ] INFO 2022/08/04 09:28:23   mode:        pv
[lp-1  ] INFO 2022/08/04 09:28:23   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/08/04 09:28:23   meters:      charge ✓
[lp-1  ] INFO 2022/08/04 09:28:23     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2022/08/04 09:28:23   vehicles:    ✓
[lp-1  ] INFO 2022/08/04 09:28:23     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2022/08/04 09:28:23     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] DEBUG 2022/08/04 09:28:23 max charge current: 6A
[site  ] DEBUG 2022/08/04 09:28:23 ----
[lp-1  ] DEBUG 2022/08/04 09:28:23 charge power: 3056W
[site  ] DEBUG 2022/08/04 09:28:23 pv power: 4688W
[site  ] DEBUG 2022/08/04 09:28:23 battery power: 6W
[site  ] DEBUG 2022/08/04 09:28:23 grid power: -2989W
[site  ] DEBUG 2022/08/04 09:28:24 battery soc 0: 34%
[site  ] DEBUG 2022/08/04 09:28:24 site power: -2883W
[lp-1  ] DEBUG 2022/08/04 09:28:24 charge currents: [13.5 0.019 0.009]A
[lp-1  ] DEBUG 2022/08/04 09:28:24 charger status: B
[lp-1  ] INFO 2022/08/04 09:28:24 car connected
[lp-1  ] DEBUG 2022/08/04 09:28:24 vehicle api refresh
[lp-1  ] DEBUG 2022/08/04 09:28:24 pv timer elapse
[lp-1  ] DEBUG 2022/08/04 09:28:24 pv timer inactive
[lp-1  ] DEBUG 2022/08/04 09:28:24 charger vehicle id: 04DA682ABD7280
[lp-1  ] INFO 2022/08/04 09:28:24 vehicle updated: unknown -> eGolf
[lp-1  ] DEBUG 2022/08/04 09:28:24 set min current: 8
[lp-1  ] DEBUG 2022/08/04 09:28:24 set max current: 16
[lp-1  ] DEBUG 2022/08/04 09:28:24 set target soc: 90
[lp-1  ] DEBUG 2022/08/04 09:28:24 vehicle status: C
[lp-1  ] DEBUG 2022/08/04 09:28:24 vehicle soc: 75%
[lp-1  ] DEBUG 2022/08/04 09:28:24 vehicle range: 132km
[lp-1  ] DEBUG 2022/08/04 09:28:24 pv charge current: 12.5A = 0A + 12.5A (-2883W @ 1p)
[lp-1  ] DEBUG 2022/08/04 09:28:24 climater active: false, target temp: 15.5°C, outside temp: 0.0°C
[lp-1  ] DEBUG 2022/08/04 09:28:24 pv timer reset
[lp-1  ] DEBUG 2022/08/04 09:28:24 pv timer inactive
[lp-1  ] DEBUG 2022/08/04 09:28:24 max charge current: 12A
[site  ] DEBUG 2022/08/04 09:28:24 ----

### What type of operating system are you running?

Docker container

### Version

0.98

andig, 2022-08-04:
Damit kann ich leider nichts anfangen:

> charger vehicle id: 04DA682ABD7280

RFID wird erkannt. Aber:

> charger out of sync: expected disabled, got enabled

Keine RFID, dafür Fehler. War überhaupt eine angehalten? Logfile der Chargerkommunikation?

Generell: issuetemplate bitte vollstädig ausfüllen, dafür haben wir es entwickelt...

andig, 2022-08-04:
Setz mal bitte

```
levels:
  easee: trace
```

und dann bitte nochmal testen ob die RFID erkannt wird.

andig, 2022-08-04:
> Also wenn die Easee die RFID nicht weitergibt, dann können wir das Issue schließen. Was die Easee-API aktuell zurück gibt ist oft einfach nur falsch. Jetzt gerade ist garkein Auto angeschlossen. Trotzdem meldet die Easee-API es wäre eines angeschlossen und gibt den RFID des eGolf zurück.

Da liegt die Vermutung nahe, dass der Root Cause bei der Easee liegt. Ist eigentlich genug Material um dort ein Ticket zu eröffnen, oder? Hier schließen wir bis es etwas Neues gibt?

closed 2022-08-04 by andig

Robbe64, 2022-08-04:
Ja, gut. Ticket habe ich ja bereits eröffnet. Liegt gerade bei deren Entwicklungsabteilung. Freitag soll ich ne Rückmeldung bekommen.
Danke trotzdem für die Hilfe!