evcc-io/evcc #4050: Cannot create vehicle bug for BMW i4 after v0.98 upgrade

Wishmaster83, 2022-08-05:
### Describe the bug

Hi,
i have a BMW i4 which i successfully integrated into evcc a while ago and it was running just fine (detecting the car and showing the SoC).
But after the upgrade to v.0.98 the car is not recognized anymore and i get the following error message:

"vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot find vehicle: WBY31AW010FN****"

I have set up the configuration for the BMW exactly according to the instructions and haven't touched it after it was working.

### Steps to reproduce

Again, all i did was upgrading to V. 0.98

### Configuration details

meters:
- type: template
  template: powerfox-poweropti
  usage: grid
  user: w.wagner1308@gmail.com
  password: waldo8583.pwrfx
  name: grid1
- # name: pv2
  type: custom
  # usage: pv
  power:
    source: modbus
    id: 1
    uri: 192.168.1.140:502
    rtu: true # serial modbus rtu (rs485) device connected using simple ethernet adapter
    # register details
    register:
      address: 32290 # Active generation power
      type: holding
      decode: int32
  name: pv2

chargers:
- type: template
  template: amtron
  id: 255
  host: 192.168.1.139
  port: 502
  modbus: tcpip
  name: wallbox4

vehicles:
- name: ev3
  type: template
  template: bmw
  title: BMW i4 M50
  user: w.wagner1308@gmail.com
  password: waldo8583.bmw
  vin: WBY31AW010FN1****
  capacity: 80

loadpoints:
- title: Stellplatz
  charger: wallbox4
  vehicles:
  - ev3
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: false

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.265 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.123 # [currency]/kWh

site:
  title: Home
  meters:
    grid: grid1
    pvs:
    - pv2

### Log details

evcc --log debug
[main  ] INFO 2022/08/05 10:46:51 evcc 0.98
[main  ] INFO 2022/08/05 10:46:51 using config file /home/pi/evcc.yaml
[main  ] INFO 2022/08/05 10:46:51 listening at :7070
[site  ] INFO 2022/08/05 10:46:52 site config:
[site  ] INFO 2022/08/05 10:46:52   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/08/05 10:46:52     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/08/05 10:46:52     pv 1:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/08/05 10:46:52 loadpoint 1:
[lp-1  ] INFO 2022/08/05 10:46:52   mode:        now
[lp-1  ] INFO 2022/08/05 10:46:52   charger:     power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/08/05 10:46:52   meters:      charge ✓
[lp-1  ] INFO 2022/08/05 10:46:52     charge:    power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/08/05 10:46:52   vehicles:    ✓
[lp-1  ] INFO 2022/08/05 10:46:52     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] DEBUG 2022/08/05 10:46:52 max charge current: 6A
[server] FATAL 2022/08/05 10:46:52 remove /tmp/evcc: operation not permitted
[site  ] DEBUG 2022/08/05 10:46:52 ----

### What type of operating system are you running?

Linux

### Version

0.98

andig, 2022-08-05:
Fixed in master

closed 2022-08-05 by andig

Wishmaster83, 2022-08-05:
> Fixed in master

Sorry for the stupid question (github noob here), but what does that mean?

andig, 2022-08-05:
Ist in 0.99 und im Nightly release mit drin 👍🏻

Wishmaster83, 2022-08-05:
> Ist in 0.99 und im Nightly release mit drin 👍🏻

Habs aktualisiert und getestet, läuft wieder 1A, tausend Dank für den schnellen fix!! :)