evcc-io/evcc #4053: Vorschlag: Anpassbare Zeiten für prioritySoC

SkyDiscovery, 2022-08-05:
Anpassbare Zeiten für prioritySoc, da es meistens ab z. B. 18 Uhr bei schlechterem Wetter nicht mehr geschafft wird z. B. von  65 auf 100 bzw. 80 zu laden, um über die Nacht zu kommen, wenn man das Auto lädt.

#24h formatted
prioritySoC: [18:70,19:80]

andig, 2022-08-07:
Was spricht dagegen, den PrioritySoC einfach extern von einer Heimautomatisierung zu steuern?

SkyDiscovery, 2022-08-07:
> Was spricht dagegen, den PrioritySoC einfach extern von einer Heimautomatisierung zu steuern?

Wir haben keine Heimautomatisierung. 

Mein Vorschlag soll die Möglichkeit bieten thresholds setzen zu können (vorwiegend für die Abendstunden) Damit man mit dem Speicher noch über die Nacht kommt und dieser dann auch ausreichend geladen ist. 

Mit PV kann das zwar gehen aber für Min+PV sollte man thresholds setzen können damit man wie gesagt genug Energie für den Hausbedarf über nacht hat indem dem zum vorangeschrittenen Tag die Ladung früher unterbrochen wird damit sich der Speicher auch wieder voll bzw. fast voll laden kann 

Es kann sein das ich hier priority und buffer verwechsle..

VolkerK62, 2022-08-07:
Basteltipp: entsprechende API-calls per cronjob schicken

andig, 2022-08-07:
Was für Thresholds? Ich dachte hier gehts um Zeiten?

andig, 2022-08-07:
Ich verschiebe mal in die Discussions für mehr Visibility.