evcc-io/evcc #4061: failed to download stamps: Get "https://raw.githubusercontent.com/neoPix/bluelinky-stamps/master/hyundai-014d222

RaveGun, 2022-08-07:
### Describe the bug

The version 0.99 is not able to configure the vehicle.

evcc[1342368]: [lp-1  ] ERROR 2022/08/07 10:41:35 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.githubusercontent.com/neoPix/bluelinky-stamps/master/hyundai-014d222>

### Steps to reproduce

Moved from 0.95 to 0.99

Moved back to 0.95 and it is working again.

### Configuration details

-

### Log details

-

### What type of operating system are you running?

Linux

andig, 2022-08-07:
Could you please share logfile of .95 and .99? I don‘t think that we‘ve made any changes there.

RaveGun, 2022-08-07:
```
root@hellsnuke:~/evcc# ./evcc 
[main  ] INFO 2022/08/07 14:23:26 evcc 0.95
[main  ] INFO 2022/08/07 14:23:26 using config file /root/evcc/evcc.yaml
[main  ] INFO 2022/08/07 14:23:26 listening at :7070
[lp-1  ] WARN 2022/08/07 14:23:28 ignoring phases config (3p) for switchable charger
[site  ] INFO 2022/08/07 14:23:28 site config:
[site  ] INFO 2022/08/07 14:23:28   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/08/07 14:23:28     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/08/07 14:23:28     pv 1:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/08/07 14:23:28 loadpoint 1:
[lp-1  ] INFO 2022/08/07 14:23:28   mode:        pv
[lp-1  ] INFO 2022/08/07 14:23:28   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/08/07 14:23:28   meters:      charge ✓
[lp-1  ] INFO 2022/08/07 14:23:28     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/08/07 14:23:28   vehicles:    ✓
[lp-1  ] INFO 2022/08/07 14:23:28     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/08/07 14:23:28 vehicle updated: unknown -> IQ5
```
This was the log for the 0.95.

Now that I switched over to 0.99 it started but the car is not connected right now.
Maybe it was just a fluke on the internet.
I just tried following the above mentioned link but I am still getting the 404.

Once the car is back in the garage I will plug the charger in to see how it is behaving.

VolkerK62, 2022-08-07:
I have also an Ioniq5. I have got no problems with 0.99.

andig, 2022-08-07:
Closing as temporary- nothing we could do.

closed 2022-08-07 by andig