evcc-io/evcc #4076: Min+PV and PV not working anymore

fountainedge, 2022-08-09:
Unfortunately since the last update, the Min+PV and the PV Mode are not doing anything if I start them. I am only able to charge „now“ or not. In the previous version it started to charge but failed. Installation is Senec+Easee+SmartEQ.

fountainedge, 2022-08-09:
....as I reported it now, 20 minutes later it started to charge at Min+PV. But it stopped after some time without a reason and did not start again. Only at manual switch off and on again and then wait some minutes, the Min+PV is working again. I don't know if it is the Smart EQ (with the 22kW Zoe charger) who makes the problem or if the refresh of data sometimes has delays which lead to this on/off symptomatic...?

andig, 2022-08-09:
There have been no changes. Closing due to missing details.

fountainedge, 2022-08-09:
bitte mal kurz die vier Bilder anschauen. Das passiert bei mir regelmäßig seit der aktuellen Version: 
Es fängt an zu laden mit eigentlich zu niedriger Leistung (Min 8A eingestellt), fährt hoch, zeigt dann völlig falsche Verbrauchswerte und fährt deshalb vermutlich wieder runter, bis es ganz abbricht. Das ganze passiert innerhalb von 3 refresh-Vorgängen, also in einer halben Minute.

[image]
[image]
[image]
[image]

andig, 2022-08-09:
> Das passiert bei mir regelmäßig seit der aktuellen Version

Das ändert nichts daran, dass jegliche angefragte Information fehlt.

andig, 2022-08-10:
@fountainedge gerne neuer Anlauf mit Konfiguration und Logfile!

fountainedge, 2022-08-10:
> @fountainedge gerne neuer Anlauf mit Konfiguration und Logfile!

ich versuch's - leider bin ich in Sachen Raspi und Befehle und so eher etwas noob-mäßig unterwegs und klaue mir die Befehle überall zusammen....
Daher auch auf die Schnelle die Screenshots des Verhaltens. Ich versuche mich einzulesen und dann das Verhalten zu reproduzieren und loggen zu lassen. Danke!

fountainedge, 2022-08-10:
So, einmal das Logfile und die EVCC.yaml.
Wechsel (nachts) von "PV" auf "Min+PV" ohne dass der Ladevorgang beginnt:

Aug 10 23:33:30 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:30 battery power: 603W
Aug 10 23:33:30 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:30 grid power: -21W
Aug 10 23:33:30 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:30 battery soc 0: 65%
Aug 10 23:33:30 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:30 site power: 582W
Aug 10 23:33:30 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:30 charge currents: [0.009 0.009 0.009]A
Aug 10 23:33:30 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:30 charger status: B
Aug 10 23:33:30 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:30 next soc poll remaining time: 58m49s
Aug 10 23:33:30 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:30 available power -582W < 4140W min 3p threshold
Aug 10 23:33:30 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:30 phase scale1p in 1m20s
Aug 10 23:33:30 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:30 pv charge current: 0A = 0A + -0.843A (582W @ 3p)
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 set charge mode: minpv
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 pv timer elapse
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 pv timer inactive
Aug 10 23:33:35 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:35 ----
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 charge power: 0W
Aug 10 23:33:35 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:35 pv power: 0W
Aug 10 23:33:35 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:35 battery power: 605W
Aug 10 23:33:35 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:35 grid power: -17W
Aug 10 23:33:35 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:35 battery soc 0: 65%
Aug 10 23:33:35 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:35 site power: 589W
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 charge currents: [0.009 0.009 0.009]A
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 charger status: B
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 next soc poll remaining time: 58m45s
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 available power -589W < 4140W min 3p threshold
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 phase scale1p in 1m15s
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 pv charge current: 0A = 0A + -0.853A (589W @ 3p)
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 max charge current: 6A
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 charger enable
Aug 10 23:33:35 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:35 wake-up timer: start
Aug 10 23:33:40 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:40 ----
Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 charge power: 0W
Aug 10 23:33:40 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:40 pv power: 0W
Aug 10 23:33:40 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:40 battery power: 600W
Aug 10 23:33:40 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:40 grid power: -28W
Aug 10 23:33:40 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:40 battery soc 0: 65%
Aug 10 23:33:40 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:40 site power: 572W
Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 charge currents: [0.009 0.009 0.009]A
Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 charger status: B
Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 next soc poll remaining time: 58m40s
Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 available power -572W < 4140W min 3p threshold
Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 phase scale1p in 1m10s
Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 pv charge current: 0A = 0A + -0.829A (572W @ 3p)
Aug 10 23:33:50 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:50 ----
Aug 10 23:33:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:50 charge power: 379W
Aug 10 23:33:50 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:50 pv power: 0W
Aug 10 23:33:50 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:50 battery power: 610W
Aug 10 23:33:50 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:50 grid power: 27W
Aug 10 23:33:50 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:50 battery soc 0: 65%
Aug 10 23:33:50 raspberrypi evcc[20687]: [site  ] DEBUG 2022/08/10 23:33:50 site power: 637W
Aug 10 23:33:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:50 charge currents: [7.03 7.07 7.02]A
Aug 10 23:33:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:50 charger status: C
Aug 10 23:33:50 raspberrypi evcc[20687]: [lp-1  ] INFO 2022/08/10 23:33:50 start charging ->
Aug 10 23:33:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:50 vehicle soc: 48%
[... cut ...]
Aug 10 23:34:00 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:34:00 vehicle soc: 48%
[... 207 lines cut ...]
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Electric-Eagle46
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.35 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1048 # [currency]/kWh`

VolkerK62, 2022-08-11:
`phases: 0` beim vehicle?

fountainedge, 2022-08-11:
> 

0 heißt doch Automatischer Wechsel, 1 und 3 wären fest vorgegebene Anzahl Phasen. Oder?

VolkerK62, 2022-08-11:
Beim LOADPOINT ... nicht beim vehicle (siehe Doku-Hinweis)

fountainedge, 2022-08-11:
> `phases: 0` beim vehicle? https://docs.evcc.io/docs/reference/configuration/vehicles#phases

hab das gerade mal geändert auf "3" und nen restart gemacht. Jetzt lädt die Mühle aktuell im Modus PV dreiphasig mit 6,3kW. Das würde soweit passen. Aber wie kam denn die 0 beim vehicle rein? die sollte eigentlich ja nur beim loadpoint drin sein für automatische Phasenumschaltung...
Ich beobachte mal den Tag über ob das restliche Verhalten jetzt auch passt, dann gebe ich Bescheid.
Danke @VolkerK62

andig, 2022-08-12:
> Wechsel (nachts) von "PV" auf "Min+PV" ohne dass der Ladevorgang beginnt:

Warum sagst Du das wenn Dein Logfile genau das Gegenteil belegt?

```
[lp-1  ] DEBUG 2022/08/10 23:33:35 set charge mode: minpv
[lp-1  ] DEBUG 2022/08/10 23:33:35 pv timer elapse
[site  ] DEBUG 2022/08/10 23:33:35 ----
[lp-1  ] DEBUG 2022/08/10 23:33:35 available power -589W < 4140W min 3p threshold
[lp-1  ] DEBUG 2022/08/10 23:33:35 phase scale1p in 1m15s
[lp-1  ] DEBUG 2022/08/10 23:33:35 charger enable
[site  ] DEBUG 2022/08/10 23:33:40 ----
[lp-1  ] DEBUG 2022/08/10 23:33:40 phase scale1p in 1m10s
[site  ] DEBUG 2022/08/10 23:33:50 ----
[lp-1  ] DEBUG 2022/08/10 23:33:50 charge currents: [7.03 7.07 7.02]A
[lp-1  ] DEBUG 2022/08/10 23:33:50 charger status: C
```

Es dauert einen Moment bis Dein Auto die Ladung beginnt und dann gehts los. Parallel startet der Timer um von 3p auf 1p umzuschalten (da bin ich nicht ganz sicher warum das nicht zum Start passiert, aber das ist ein anderes Thema).

Hier funktioniert alles wie erwartet...

closed 2022-08-12 by andig

fountainedge, 2022-08-12:
Der Ladevorgang beginnt eben nicht richtig, also es kommt einfach nicht dazu, dass das Auto wirklich mit "Min", also 6A bzw. 8A lädt.
Es wird versucht zu starten mit den 3x7A
`[lp-1  ] DEBUG 2022/08/10 23:33:50 charge currents: [7.03 7.07 7.02]A
[lp-1  ] DEBUG 2022/08/10 23:33:50 charger status: C`
5 Sekunden später ist es aber wieder bei 0,009 also 0A:
`Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 charge currents: [0.009 0.009 0.009]A
Aug 10 23:33:40 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:33:40 charger status: B`
Parallel läuft der Countdown zum einphasigen Laden runter ab 1:15min, aber sobald der auf 0 ist, passiert nichts:
`Aug 10 23:34:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:34:50 charge currents: [0.009 0.008 0.008]A
Aug 10 23:34:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:34:50 charger status: B
Aug 10 23:34:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:34:50 next soc poll remaining time: 59m9s
Aug 10 23:34:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:34:50 available power -567W < 4140W min 3p threshold
Aug 10 23:34:50 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:34:50 phase scale1p in 0s`
Dann wird kurz versucht, einphasig zu laden, aber mit zu geringem Strom (3,13A statt 6/8A), sodass das Auto das natürlich nicht akzeptiert:
`Aug 10 23:35:10 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:35:10 charge currents: [3.13 0.008 0.008]A
Aug 10 23:35:10 raspberrypi evcc[20687]: [lp-1  ] DEBUG 2022/08/10 23:35:10 charger status: B`
10 Sekunden später ist es wieder bei den quasi 0A und bleibt ab dann auch so.
Meine persönliche Logik (ohne mich mit der Programmierung im Hintergrund befasst zu haben bzw. das zu können) sagt mir, dass bei Min+PV kurz geprüft wird was seitens PV ankommt, und wenn da nichts kommt, startet der Ladevorgang mit den eingestellten 1x6A.
Nichts für Ungut, ich finde das EVCC tool wirklich Bombe und bin froh, dass sich jemand diese Mühe macht und in dieser Frequenz updatet, dafür nochmal herzlichen Dank! 
Ich denke nur, dass ich solche Feststellungen hier anbringen muss. Wenn ich da selbst irgendeinen Fehler eingebaut haben sollte, ist es natürlich mein Problem.
Ich habe nur in den letzten Tagen öfters mal die Symptomatik gehabt, dass trotz verfügbarer PV-Leistung einfach nicht richtig geladen wird, oder bei Min+PV es zu keinem echten Ladevorgang kommt.
Lasst uns das Thema hier auf geschlossen lassen, denn seit die "0" bei den vehicle Phasen in ne 3 geändert wurde, hat es bisher getan.

premultiply, 2022-08-12:
Den bezogenen Strom legt das Ladesystem des Fahrzeug selbst fest. Die Wallbox signalisiert immer nur eine Freigabe und eine PhasenstromOBERGRENZE.

VolkerK62, 2022-08-12:
@fountainedge  die Sache mit `charge currents` sieht bei meiner Wallbox ähnlich aus. 
Wenn ich volle Pulle (16A) lade, zeigt sie 15,1 A an.

fountainedge, 2022-08-12:
> Den bezogenen Strom legt das Ladesystem des Fahrzeug selbst fest. Die Wallbox signalisiert immer nur eine Freigabe und eine PhasenstromOBERGRENZE.

das heißt EVCC macht ein Angebot und der Smart Onboard Charger lehnt es ab? Würde zum viel diskutierten zickigen Ladeverhalten des Smart passen, sobald er keine 11 oder 22kW bekommt :-(

fountainedge, 2022-08-12:
> @fountainedge die Sache mit `charge currents` sieht bei meiner Wallbox ähnlich aus. Wenn ich volle Pulle (16A) lade, zeigt sie 15,1 A an.

ja, wenn ich mit Volllast (16A) lade, sind es ja auch keine nominalen 11kW sondern meist "nur" 10,5kW.

premultiply, 2022-08-13:
Das sind auch nominal nicht 11 kW sondern 230 V * 16 A = 11,04 kVA.
Die Wirkleistung ist immer kleiner als die Scheinleistung.