evcc-io/evcc #4081: Bluelink: Problem mit Fahrzeugerkennung

VolkerK62, 2022-08-09:
### Describe the bug

Fahrzeug wird nicht (bzw. nur selten) automatisch erkannt. 
https://github.com/evcc-io/evcc/discussions/4030#discussioncomment-3322620

### Steps to reproduce

Kein Defaultfahrzeug ausgewählt. 
Fahrzeug angesteckt. 
Im UI wird "Gastfahrzeug verbunden" angezeigt. 
Fahrzeug muss per Hand ausgewählt werden. 

### Configuration details

```
vehicles:
- name: auto
  type: template
  template: hyundai
  title: (T)Raumschiff      # Wird in der Benutzeroberfläche angezeigt
  user: xxx  # Benutzername um auf das Gerät zuzugreifen
  password: xxx      # Passwort des Benutzerkontos
  vin: xxx
  capacity: 72
  language: de            
  phases: 3 
  #cloud: true
  ###onIdentify
  mode: pv
  minSoC: 0
  targetSoC: 99
```

```
loadpoints:
- title: Carport
  charger: wallbox
  vehicles: 
  - auto
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  enable:           # Verhalten zum Start des PV-ladens
    threshold: 0     
    delay: 1m        
  disable:          # Verhalten zum Stop des PV-Ladens
    threshold: 100
    delay: 1m        
  soc:
    min: 0
    target: 99
    estimate: true  
    poll:
      mode: always
      interval: 180m
```

### Log details

[Bluelink20220809.txt](https://github.com/evcc-io/evcc/files/9290066/Bluelink20220809.txt)

13:09:16 car connected
13:11:37 Fahrzeug manuell ausgewählt

### What type of operating system are you running?

Linux

### Version

0.99

andig, 2022-08-09:
@VolkerK62 isso. Was ist Deine Erwartung in Anbetracht der (Nicht-)Fähigkeiten der Schnittstelle?

VolkerK62, 2022-08-09:
Du immer mit deinen schelmischen Fragen ;-)

Nach dem manuellen Auswählen, kommen SoC und Range und Odometer (geprüft: das sind aktuelle Werte).
Der Status ist aber immer "A", auch später, wenn geladen wird.
Es werden also aktuellen Werte übertragen, nur der Status ist nicht aktuell.
Wird der überhaupt übertragen? Ich habe vergeblich versucht, das in der Hyundai-Antwort zu finden.
Würde es einen Fehler geben, wenn kein Status übertragen wird, oder wird der dann per default auf "A" gesetzt?
Wird bei `evcc vehicle` der gleichen API Aufruf geschickt?

andig, 2022-08-09:
> Wird bei evcc vehicle der gleichen API Aufruf geschickt?

Ja

> Es werden also aktuellen Werte übertragen, nur der Status ist nicht aktuell.

Gibts das als trace log vom Auto beim laden?

andig, 2022-08-09:
Gibts da eine Statusabfrage die ich übersehe?

VolkerK62, 2022-08-09:
Ah, du meinst ein `evcc -l trace vehicle`  während das Auto lädt?
Das hab ich nicht. Werd ich nächstes Mal dran denken. Jetzt ist Auto voll.

In dem ersten logfile wird 2x die API abgefragt. Einmal beim connect und dann beim manuellen Fahrzeugwechsel. Mehr hab ich nicht.

premultiply, 2022-08-09:
Nee das muss schon ein Log des laufenden Prozesses sein, denn mit einem neuen Prozess (evcc vehicle) triggert man ja eine komplett neue Abfrage an die nichts mit dem laufenden Dienst zu tun hat.

Das Blöde ist ja dass evcc die Fahrzeug-APIs entsprechend pollen muss und gleichzeitig die Fahrzeug-APIs teilweise sehr träge sind so dass regelmäßig bei einem lokalen Connect-Event der Wallbox die jeweilige API noch gar nicht den Status auf Seite des Fahrzeugs aktualisiert hat. Hier muss man daher mehrfach Pollen (wie oft und wie lang?) bis eine API einen Timestamp liefert der zeitlich >= dem Connect-Event der Wallbox entspricht und einen passenden Ladezustand liefert.
Das kann von Sekunden über Minuten bis zu Niemals dauern.

Letztendlich überschneidet sich das vermutlich zeitlich, ist von temporären API-Problemen sowie von der Mobilfunksituation des Fahrzeugs beeinflusst. Und jeder Hersteller macht es anders, gepaart mit permanenten Änderungen, Bugs, Reverse-Engineering usw. Und zu oft anfragen darf man ja auch nicht sonst läuft man ja gleich wieder Gefahr (temporär) ausgesperrt zu werden was dann noch kontraproduktiver ist.

Vielleicht muss man auch nochmal ganz Grundlegend in das entsprechende Konzept und den zugehörigen Code seitens evcc schauen. Nicht auszuschließen dass dort auch noch ein paar Aspekte verbessert werden könnten.

VolkerK62, 2022-08-09:
Das erklärt, warum es manchmal funktioniert und meistens nicht.
Das erklärt auch, warum Range, SoC und Odometer aktuell waren (das Auto hat mehrere Stunden gestanden).
Kann ich noch irgendetwas machen/testen/ausprobieren?

andig, 2022-08-09:
> Ist doch `level: trace`. Ich habe dann ein paar Minuten geladen. Hier ist die Fortsetzung des log. Da ist aber nix mehr von Hyundai. [Bluelink20220809_2.txt](https://github.com/evcc-io/evcc/files/9292857/Bluelink20220809_2.txt)
> 
> Oder habe ich dich falsch verstanden?

Ich sehe im Log nichts von `hyundai`, dafür jede Menge unnützes `influx, cache, daheim, modbus` etc. Falls logging an war wird nix abgefragt.

> Das erklärt, warum es manchmal funktioniert und meistens nicht.

Es gibt 2 Möglichkeiten:

- Refreshproblem: das geht dann meist bei neuer Session. Zusammenhänge lassen sich am Besten mit gleichzeitigen `trace` logs erkennen.
- Auto sendet nur A- dann hast Du Pech gehabt oder Auto sendet korrekt wir dekodieren es falsch- nur mit `trace` log der Fahrzeugkommunikation zu erkennen.

Es fehlen also weiter verwertbare Daten :(

VolkerK62, 2022-08-10:
Ok, ich hatte alles auf `trace`. Ich habe meine `log levels` jetzt so angepasst, dass nur noch `hyundai` auf `trace` ist. Dann sollte der unnütze Kram demnächst raus sein. Wenn der Akku wieder etwas Platz hat, werde ich einen nächsten Versuch starten.

VolkerK62, 2022-08-11:
nächster Test:  Der Tip (an anderer Stelle) von @premultiply  `soc:poll:always:10m` einzustellen funktioniert nicht. Da das Fahrzeug nicht erkannt wird, findet wohl auch kein soc polling statt.
Es ist schon etwas crazy.
Auto war angesteckt, aber selbst in der Bluelink App wurde es als "nicht verbunden" angezeigt. Tür auf, Tür zu über die App hat funktioniert, also Verbindung war da. Auch als ich das Laden im UI (Zustand Gastfahrzeug) gestartet habe, fand auch keine Erkennung statt- Erst als ich die App aktualisiert habe, wurd dort "wird geladenn" angezeigt und dann wurde das Auto auch in evcc erkannt.
Es ist fast so, als würde das Fahrzeug den Stecker nicht erkennen, aber würde es dann laden?

```
Aug 11 10:53:21 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:21 charge power: 0W
Aug 11 10:53:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:36 charge power: 0W
Aug 11 10:53:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:36 charge currents: [0 0 0]A
Aug 11 10:53:37 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:37 charger status: B
Aug 11 10:53:37 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:37 charge start energy: 880kWh
Aug 11 10:53:37 raspberrypi evcc[16995]: [lp-1  ] INFO 2022/08/11 10:53:37 car connected
Aug 11 10:53:37 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:37 vehicle api refresh
Aug 11 10:53:37 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:37 pv timer elapse
Aug 11 10:53:37 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:37 pv timer inactive
Aug 11 10:53:37 raspberrypi evcc[16995]: [hyundai] TRACE 2022/08/11 10:53:37 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Aug 11 10:53:37 raspberrypi evcc[16995]: [hyundai] TRACE 2022/08/11 10:53:37 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":49.310094,"lon":8.539169,"alt":0,"type":0},"head":150,"speed":{"value":0,"unit":0},"accuracy":{"hdop":0,"pdop":0},"time":"20220810162757"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":93,"batteryPlugin":0,"remainTime2":{"etc1":{"value":2,"unit":1},"etc2":{"value":120,"unit":1},"etc3":{"value":30,"unit":1},"atc":{"value":120,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":453,"unit":1},"totalAvailableRange":{"value":453,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"0000","timeSection":0}},"end":{"day":0,"time":{"time":"0000","timeSection":0}}},"targetSOClist":[{"targetSOClevel":100,"plugType":1},{"targetSOClevel":80,"plugType":0}]},"batteryPreconditioning":false},"ign3":false,"hoodOpen":false,"transCond":true,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":86,"batState":0,"sjbDeliveryMode":0,"batSignalReferenceValue":{"batWarning":65},"powerAutoCutMode":2},"lampWireStatus":{"stopLamp":{"stopLampStatus":false,"leftLamp":false,"rightLamp":false},"headLamp":{"headLampStatus":false,"leftLowLamp":false,"rightLowLamp":false,"leftHighLamp":false,"rightHighLamp":false,"leftBifuncLamp":false,"rightBifuncLamp":false},"turnSignalLamp":{"turnSignalLampStatus":false,"leftFrontLamp":false,"rightFrontLamp":false,"leftRearLamp":false,"rightRearLamp":false}},"smartKeyBatteryWarning":false,"washerFluidStatus":false,"breakOilStatus":false,"sleepModeCheck":false,"time":"20220811095255","systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"odometer":{"value":9680.9,"unit":1}}},"msgId":"b370eac6-5f9b-4f51-9570-4d529b420d1a"}
Aug 11 10:53:37 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:37 vehicle status: A ((T)Raumschiff)
Aug 11 10:53:51 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:53:51 charge power: 0W
Aug 11 10:54:06 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:54:06 charge power: 0W
Aug 11 10:54:06 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:54:06 charge currents: [0 0 0]A
Aug 11 10:54:07 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:54:07 charger status: B
Aug 11 10:54:07 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:54:07 vehicle status: A ((T)Raumschiff)
Aug 11 10:54:21 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:54:21 charge power: 0W
Aug 11 10:54:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:54:36 charge power: 0W
Aug 11 10:54:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 10:54:36 charge currents: [0 0 0]A
[... cut ...]
Aug 11 11:03:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 11:03:36 set min current: 6
Aug 11 11:03:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 11:03:36 set max current: 16
Aug 11 11:03:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 11:03:36 set min soc: 0
Aug 11 11:03:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 11:03:36 set target soc: 99
Aug 11 11:03:36 raspberrypi evcc[16995]: [lp-1  ] DEBUG 2022/08/11 11:03:36 final charge energy: 0.2kWh
Aug 11 11:03:36 raspberrypi evcc[16995]: [lp-1  ] INFO 2022/08/11 11:03:36 stop charging <- 
```

andig, 2022-08-11:
> Es ist fast so, als würde das Fahrzeug den Stecker nicht erkennen, aber würde es dann laden?

Auf jeden Fall. Laden = elektrisch, "Erkennung" = Sicht der Bluelink Server. Wenn die App das auch nicht ausspuckt, dann ist es auch kein Mappingproblem bei evcc.

Long story short: ist durch uns leider nicht zu lösen :( In diesem Fall würde ich zu machen?

VolkerK62, 2022-08-11:
Ja klar. Liegt definitiv nciht an evcc.
Hast du evtl einen Tip, wie ich bei Bluelink evtl ein ticket aufmachen kann?

andig, 2022-08-11:
Fahrzeughesteller? Die App tut ja nicht ;)

closed 2022-08-11 by andig

xpcone, 2022-08-11:
Also hier tut die APP und der Ioniq5 wird seit dem Update 0.96 > 0.99 nicht mehr erkannt/die Verbindung zu Bluelink geht nicht mehr.
Ich bereite gerade das Downgrade vor

andig, 2022-08-11:
Da hat sich nichts geändert…

xpcone, 2022-08-11:
So Quertest durch, Kommando kommt nur gelegentlich zurück... 
Gerade bei meinem Händler angerufen und mal gefragt, wann der Bluelink-Kram endlcih wieder klappt. 
Die Antwort, sie haben gerade einige "technisches Herrausforderungen" und "sind bemüht, dass schnellstmöglich zu lösen".

VolkerK62, 2022-08-11:
Ich habe mein Problem gerade über die Hyundai Service Seite gemeldet. Mal gespannt ob und wann und was da zurückkommt.

VolkerK62, 2022-08-14:
zur Info. Mein Händler hat mir bestätigt, dass das ein bekannter Fehler ist und das daran gearbeitet wird.