evcc-io/evcc #4094: RCT Wechselrichter Kommunikationsprobleme

JAK0721, 2022-08-10:
### Describe the bug

Hallo,

ich bekomme vom RCT Wechselrichter permanent Fehlermeldungen. Die Anlage besteht aus:
POWER STORAGE DC 6.0
POWER BATTERY 7.6
Heidelberg Energy Control

danke und Grüße 
Jörg

grid meter energy: Invalid response Cmd Write (02) Id #INVALID (3AA565FC) Data [62 240 235 183 1 0 0 0 64 250 201 193 0 209 45 68 190 165 14 58 23 183 209 56 1 0 0 0 13 223 118 63 0 0 0 0]
9 minutes ago
grid meter energy: Invalid response Cmd Write (02) Id #INVALID (3AA565FC) Data [62 240 235 183 1 0 0 0 0 48 17 190 186 133 54 68 171 56 74 184 236 81 120 63 1 0 0 0 158 238 118 63 0 0 0 0]
10 minutes ago
grid meter energy: Invalid response Cmd Write (02) Id #INVALID (3AA565FC) Data [62 240 235 183 1 0 0 0 64 66 154 193 6 175 45 68 104 249 82 159 0 0 0 0 1 0 0 0 68 243 118 63 0 0 0 0]
11 minutes ago
grid meter energy: Invalid response Cmd Write (02) Id #INVALID (3AA565FC) Data [62 240 235 183 1 0 0 0 160 239 212 193 42 52 42 68 162 156 163 3 80 56 23 66 1 0 0 0 214 241 118 63 0 0 0 0]

### Steps to reproduce

Fehler tauchen sporadisch auf

### Configuration details

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NTQ1MDIyMTIsImlhdCI6MTY1OTg5NDIxMiwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IkpBSzA3MjEifQ.jR4gvxoo8o4E1hGUYqPbioKA7dMT-6YOHUbIkxzuNsE

meters:
- type: template
  template: rct-power
  usage: grid
  host: 192.168.0.225
  name: grid1
- type: template
  template: rct-power
  usage: pv
  host: 192.168.0.225
  name: pv2
- type: template
  template: rct-power
  usage: battery
  host: 192.168.0.225
  name: battery3

chargers:
- type: template
  template: heidelberg
  id: 2
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: 8E1
  modbus: rs485serial
  name: wallbox6

vehicles:
- type: template
  template: opel
  title: Corse e
  user: jkaina
  password: nicolas1205
  vin: VXKUHZKXZN4327572
  capacity: 50
  phases: 1
  mode: off
  minSoC: 0
  targetSoC: 100
  minCurrent: 6

### Log details

pi@raspberrypi:/etc $ evcc --log debug
[main  ] INFO 2022/08/10 18:59:29 evcc 0.99
[main  ] INFO 2022/08/10 18:59:29 using config file /etc/evcc.yaml
[main  ] INFO 2022/08/10 18:59:29 listening at :7070
[lp-1  ] WARN 2022/08/10 18:59:29 vehicle 'unavailable' does not support automatic detection
[site  ] INFO 2022/08/10 18:59:29 site config:
[site  ] INFO 2022/08/10 18:59:29   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/08/10 18:59:29     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/08/10 18:59:29     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/08/10 18:59:29     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/08/10 18:59:29 loadpoint 1:
[lp-1  ] INFO 2022/08/10 18:59:29   mode:        off
[lp-1  ] INFO 2022/08/10 18:59:29   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2022/08/10 18:59:29   meters:      charge ✓
[lp-1  ] INFO 2022/08/10 18:59:29     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/08/10 18:59:29   vehicles:    ✓
[lp-1  ] INFO 2022/08/10 18:59:29     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/08/10 18:59:29     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] DEBUG 2022/08/10 18:59:29 ----
[server] FATAL 2022/08/10 18:59:29 remove /tmp/evcc: operation not permitted

### What type of operating system are you running?

Linux

### Version

evcc version 0.99

andig, 2022-08-10:
/cc @mlnoga

andig, 2022-08-10:
Kann es sein, dass du 2 evcc laufen hast?

JAK0721, 2022-08-11:
Hallo,

wie meinst Du das? Ich hatte einen Testinstallation auf einem Raspberry B+ gemacht. Da es dort funktioniert hatte, aber auch mit den Fehlern in der Kommunikation.
Anschließend habe ich die Installation auf einem Raspberry Pi 4+, der B+ ist aber nicht mehr am Netz/Strom.

andig, 2022-08-11:
> operation not permitted

das passiert gerne wenn evcc doppelt läuft. Mach mal bitte ein

    evcc meter —log trace

andig, 2022-08-11:
Und ändere bitte das Passwort deines Autos! Das kennt jetzt jeder :(

andig, 2022-08-11:
Beim nochmaligen hinschauen: dein evcc läuft aufgrund der Fehlermeldung nicht. Datei bitte händisch löschen. Mit RCT das das nciht zu tun…

closed 2022-08-11 by andig