evcc-io/evcc #4096: Use English as base communication language

joseigor, 2022-08-11:
**Is your feature request related to a problem? Please describe.**
Hi Guys, 

First of all, you are doing a really good job here. But I would to know if there is the possibility to adopt English as the main base language for discussion on PR and Issues. The reason is that for those who do not speak German is very hard to follow the discussions.

The same for the Documentation and website (for those I can even contribute in my spare time).

BRs, 

Jose

RTTTC, 2022-08-11:
Mainly German people here.
Language does not pose a big problem for me - I am reading everything here using autotranslators, and posting in EN, seems to be working just fine for me. I do appreciate that this way I am accepted here. I would think it's not exactly polite to request that people speak YOUR language, when visiting them ;)

andig, 2022-08-12:
As said before- we won't force our users to any language. German is essential for our user base. If you want to support website and multilingual docs please contact us on slack.

closed 2022-08-12 by andig