evcc-io/evcc #4098: Phase scale wait on connecting car

panzerdev, 2022-08-12:
### Describe the bug

When a car is being connected and the charger is set to 3p but the current production is not enough for 3p the timer to scale to 1p is started instead of directly switching down to start charging. 

### Steps to reproduce

Unplug car when last charging phases was 3p. 

Connect car later.

See timer in logs for 3p to 1p scaling

### Configuration details

phases confugured to 0 in config

### Log details

today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 charge power: 0W
today at 15:52:58[site  ] DEBUG 2022/08/12 15:52:58 pv power: 3876W
today at 15:52:58[site  ] DEBUG 2022/08/12 15:52:58 grid power: -2345W
today at 15:52:58[site  ] DEBUG 2022/08/12 15:52:58 site power: -2345W
today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 charge currents: [0 0 0]A
today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 charger status: B
today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 next soc poll remaining time: 15m10s
today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 available power 2345W < 4140W min 3p threshold
today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 phase scale1p in 7m30s
today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 pv charge current: 3.4A = 0A + 3.4A (-2345W @ 3p)
today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 climater active: false, target temp: 22.0°C, outside temp: 22.0°C
today at 15:52:58[lp-1  ] DEBUG 2022/08/12 15:52:58 climater active: false, target temp: 22.0°C, outside temp: 22.0°C

### What type of operating system are you running?

Docker container

### Version

Commit [7047a11](https://github.com/evcc-io/evcc/commit/7047a11db0563ca3874b1a2b440a8f5bce5929b1)

Nightly

panzerdev, 2022-08-12:
Workaround until latest change was to manually trigger phase switching via API. Now this doesn't work like that anymore.

andig, 2022-08-12:
Good point. I'll need to think about this some more, maybe we could just expire the PV timer on disconnect.

panzerdev, 2022-08-13:
Or maybe scale to the right amount of phases immediately before starting to charge after connect? Maybe there is a bool indicating connected but not started to charge yet?

panzerdev, 2022-08-14:
What I also noticed it's that after a few phase changes it stops working and it won't switch anymore at all. I'll need to investigate why exactly a bit more after restart with default phases set to 0 things work again as expected

andig, 2022-08-14:
> it stops working

Yep, da brauch ich mehr Details...

panzerdev, 2022-08-15:
Started server with `phases: 0` and had it doing its thing yesterday. Starting 1 phase, switching to 3 and then back to 1.

[image]

Today I connected the car, the current phases value from the mqtt topic `evcc/loadpoints/1/phases` says `1` and it doesn't want to switch phases to 3. As you can see I have over 1.5kw left.

`[lp-1  ] DEBUG 2022/08/15 12:55:02 charge power: 3645W
today at 12:55:02[site  ] DEBUG 2022/08/15 12:55:02 pv power: 6657W
today at 12:55:02[site  ] DEBUG 2022/08/15 12:55:02 grid power: -1501W
today at 12:55:02[site  ] DEBUG 2022/08/15 12:55:02 site power: -1501W
today at 12:55:02[lp-1  ] DEBUG 2022/08/15 12:55:02 charge currents: [15.7 0 0]A
today at 12:55:02[lp-1  ] DEBUG 2022/08/15 12:55:02 detected phases: 1p
today at 12:55:02[lp-1  ] DEBUG 2022/08/15 12:55:02 charger status: C
today at 12:55:02[lp-1  ] DEBUG 2022/08/15 12:55:02 next soc poll remaining time: 9m50s
today at 12:55:02[lp-1  ] DEBUG 2022/08/15 12:55:02 vehicle soc: 62%
today at 12:55:02[lp-1  ] DEBUG 2022/08/15 12:55:02 vehicle range: 217km
today at 12:55:02[lp-1  ] DEBUG 2022/08/15 12:55:02 pv charge current: 22.5A = 16A + 6.53A (-1501W @ 1p)
today at 12:55:02[lp-1  ] DEBUG 2022/08/15 12:55:02 climater active: false, target temp: 22.0°C, outside temp: 22.0°C`

andig, 2022-08-15:
Ich hab keine Zeit für Suchspiele. Bitte vollständige Config und Logfile.
Hinweis: ich werde Issues ohne die angefragten Informationen künftig kommentarlos schließen. Das ewige Nachgefrage kostet mich zuviel Zeit :(

closed 2022-08-15 by andig

panzerdev, 2022-08-17:
https://github.com/evcc-io/evcc/issues/4116

Zeigt den Bug den ich auch versucht habe zu beschreiben. Scheinbar aber nicht ausführlich genug. Macht nix. Das originale Thema dieses Issues war eigentlich ein Anders und ich denke das Issue könnte wieder geöffnet werden.

Thema ist Auto anstecken und warten mit Ladebeginn wenn die phases noch auf 3 stehen aber nur für 1 phase Überschuss zur Verfügung steht.

andig, 2022-08-17:
Ich hab einfach keine Zeit zum suchen. Bis heute fehlt die passende Config. Bitte vollständiges Issue, dann kann ich mir das anschauen.