evcc-io/evcc #4116: 3p 1p switch does not work

Flexo0815, 2022-08-15:
### Describe the bug

EVCC starts with 3p. After some time ist reduces to 1p. But there is still enough power for 3p charging. Even with 6000W ist does not switch to 3p. 

### Steps to reproduce

It does not switch to 3p after reducing to 1p. No matter how much pv energy ist produced. 

### Configuration details

uri: XXX:7070 # uri for ui
interval: 10s # control cycle interval

log: debug

meters:
- type: template
  template: rct-power
  usage: grid
  host: XXX
  name: grid1
- type: template
  template: rct-power
  usage: pv
  host: XXX
  name: pv2
- type: template
  template: rct-power
  usage: battery
  host: XXX
  name: battery3

chargers:
- type: template
  template: easee
  user: XXX
  password: XXX
  charger: XXX
  name: wallbox5

vehicles:
- type: template
  template: tesla
  title: Hugo
  accessToken: XXX
  refreshToken: XXX
  vin: XXX
  capacity: 60
  phases: 3
  mode: pv
  minSoC: 0
  targetSoC: 90
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: ev4

loadpoints:
- title: Wallbox
  charger: wallbox5
  vehicle: ev4
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
    min: 0
    target: 90
    estimate: false
  enable:
[... 372 lines cut ...]
[lp-1  ] INFO 2022/08/15 12:27:42 loadpoint 1:
[lp-1  ] INFO 2022/08/15 12:27:42   mode:        pv
[lp-1  ] INFO 2022/08/15 12:27:42   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/08/15 12:27:42   meters:      charge ✓
[lp-1  ] INFO 2022/08/15 12:27:42     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2022/08/15 12:27:42   vehicles:    ✓
[lp-1  ] INFO 2022/08/15 12:27:42     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/08/15 12:27:42 vehicle updated: unknown -> Hugo
[lp-1  ] DEBUG 2022/08/15 12:27:42 set charge mode: pv
[lp-1  ] DEBUG 2022/08/15 12:27:42 set min current: 6
[lp-1  ] DEBUG 2022/08/15 12:27:42 set max current: 16
[lp-1  ] DEBUG 2022/08/15 12:27:42 set min soc: 0
[lp-1  ] DEBUG 2022/08/15 12:27:42 set target soc: 90
[lp-1  ] DEBUG 2022/08/15 12:27:42 max charge current: 6A
[site  ] DEBUG 2022/08/15 12:27:42 ----
[lp-1  ] DEBUG 2022/08/15 12:27:42 charge power: 1437W
[server] FATAL 2022/08/15 12:27:42 remove /tmp/evcc: operation not permitted

### What type of operating system are you running?

Linux

### Version

evcc version 0.100

Flexo0815, 2022-08-15:
here the moment, when the switch to 1p starts in the log file. there is a panic error?!

Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 charge power: 0W
Aug 15 12:52:31 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:31 pv power: 5882W
Aug 15 12:52:31 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:31 battery power: -3W
Aug 15 12:52:31 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:31 grid power: -5589W
Aug 15 12:52:31 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:31 battery soc 0: 90%
Aug 15 12:52:31 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:31 site power: -5592W
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 charge currents: [0.01 0.011 0.01]A
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 charger status: C
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] INFO 2022/08/15 12:52:31 start charging ->
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 vehicle soc: 91%
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 vehicle range: 393km
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 pv charge current: 10.1A = 2.01A + 8.1A (-5592W @ 3p)
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 max charge current: 10A
Aug 15 12:52:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:41 ----
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 charge power: 1435W
Aug 15 12:52:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:41 pv power: 5887W
Aug 15 12:52:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:41 battery power: -3W
Aug 15 12:52:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:41 grid power: -3250W
Aug 15 12:52:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:41 battery soc 0: 90%
Aug 15 12:52:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:41 site power: -3252W
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 charge currents: [6.05 0.011 0.01]A
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 detected phases: 1p
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 charger status: C
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 vehicle soc: 91%
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 vehicle range: 393km
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 pv charge current: 22.2A = 8.05A + 14.1A (-3252W @ 1p)
Aug 15 12:52:42 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:42 max charge current: 16A
Aug 15 12:52:48 raspberrypi evcc[11091]: recovering from panic in logger: expected string, got []interface {} from [ts 2022-08-15T11:52:48.920933998Z level info connect
Aug 15 12:52:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:51 ----
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 charge power: 2373W
Aug 15 12:52:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:51 pv power: 5884W
Aug 15 12:52:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:51 battery power: -3W
Aug 15 12:52:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:51 grid power: -3250W
Aug 15 12:52:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:51 battery soc 0: 90%
Aug 15 12:52:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:51 site power: -3252W
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 charge currents: [10 0.011 0.01]A
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 detected phases: 1p
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 charger status: C
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 vehicle soc: 91%
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 vehicle range: 393km
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 pv charge current: 26.2A = 12A + 14.1A (-3252W @ 1p)
Aug 15 12:53:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:53:01 ----
Aug 15 12:53:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:53:01 charge power: 2373W
Aug 15 12:53:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:53:01 pv power: 5935W
Aug 15 12:53:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:53:01 battery power: -3W
Aug 15 12:53:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:53:01 grid power: -3253W
Aug 15 12:53:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:53:01 battery soc 0: 90%
Aug 15 12:53:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:53:01 site power: -3256W
Aug 15 12:53:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:53:01 charge currents: [10 0.011 0.01]A
Aug 15 12:53:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:53:01 detected phases: 1p
Aug 15 12:53:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:53:01 charger status: C
Aug 15 12:53:02 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:53:02 vehicle soc: 92%
Aug 15 12:53:02 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:53:02 vehicle range: 400km
Aug 15 12:53:02 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:53:02 pv charge current: 26.2A = 12A + 14.2A (-3256W @ 1p)

andig, 2022-08-15:
Bitte formatieren.

andig, 2022-08-15:
> It does not switch to 3p after reducing to 1p. No matter how much pv energy ist produced.

Timestamp?

Flexo0815, 2022-08-15:
Was meinst du damit?

Flexo0815, 2022-08-16:
Um 12:52:48 Uhr etwa ist es passiert. Dort steht dann etwas von „recovering from panic logger“. 

Sorry, ich habe leider keine Ahnung wie ich den Code/Config als Code formatieren soll…

Flexo0815, 2022-08-16:
Ah, dafür scheint man nur alles markieren zu müssen und dann die Pfeile ;-)

Flexo0815, 2022-08-18:
Fehlt jetzt noch etwas von meiner Seite? Oder ist es tatsächlich ein Bug, der auch an anderer Stelle auftaucht? #4089

andig, 2022-08-18:
> Um 12:52:48 Uhr etwa ist es passiert. Dort steht dann etwas von „recovering from panic logger“.

Das panic hat nix mit 1p3p zu tun. Wo sehe ich das 1p3p Problem im Log? Woran machst Du das "Problem" fest?

Flexo0815, 2022-08-18:
Ok. Das log war ja ursprünglich etwas länger. Ich hatte es dann auf den Moment gekürzt, ab dem plötzlich nur noch mit 1p geladen wird, obwohl genug Überschuss vorhanden ist. Das bleibt ab dann auch dauerhaft bei 1p stehen. Es erfolgt erst wieder ein Laden mit 3p, wenn ich EVCC ausschalte und wieder auf PV stelle. Dann geht das Spiel von vorne los. 

Bis zu diesem Zeitpunkt wurde mit 3p geladen. 

Die Umschaltung von 3p auf 1p passiert immer nach ein paar Minuten ohne ersichtlichen Grund. Danach bleibt EVCC bei 1p „hängen“, egal wieviel Überschuss vorhanden ist. Selbst bei 7000w wird dann nur mit 3,8KW und 1p geladen.

andig, 2022-08-18:
Meine Glaskugel ist leider zur Reparatur ;) Zeig mir bitte das komplette Log mit den Zeitpunkten wo Du Fehlverhalten siehst.

Flexo0815, 2022-08-19:
Es ist garnicht so einfach das zu liefern, was du benötigst ;-) 

Vor allem weiß ich ja noch nicht einmal, was benötigt wird. Ich kann nur beschreiben, was aus meiner Sicht ein falsches Verhalten der Software ist. Bei weitaus genug Überschuss sollte doch auf 3p umgeschaltet werden, oder? 

Hier nochmal das LOG in ausführlicher Form. Bis 12:52:00 Uhr läuft alles normal. Dann passiert irgendwas (ich habe keinen Schimmer was) und es wird auf 1P heruntergeschaltet. Danach geht es ohne Neustart auch nicht wieder auf 3P zurück. Egal, wieviel Überschuss vorhanden ist. 

```
Aug 15 12:51:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:41 charge power: 4900W
Aug 15 12:51:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:41 pv power: 5864W
Aug 15 12:51:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:41 battery power: -3W
Aug 15 12:51:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:41 grid power: -576W
Aug 15 12:51:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:41 battery soc 0: 90%
Aug 15 12:51:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:41 site power: -579W
Aug 15 12:51:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:41 charge currents: [7.04 7.03 7.03]A
Aug 15 12:51:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:41 detected phases: 3p
Aug 15 12:51:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:41 charger status: C
Aug 15 12:51:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:41 vehicle soc: 91%
Aug 15 12:51:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:41 vehicle range: 393km
Aug 15 12:51:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:41 pv charge current: 7.84A = 7A + 0.839A (-579W @ 3p)
Aug 15 12:51:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:41 max charge current: 7A
Aug 15 12:51:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:51 ----
Aug 15 12:51:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:51 charge power: 4900W
Aug 15 12:51:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:51 pv power: 5863W
Aug 15 12:51:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:51 battery power: -3W
Aug 15 12:51:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:51 grid power: -583W
Aug 15 12:51:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:51 battery soc 0: 90%
Aug 15 12:51:51 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:51:51 site power: -586W
Aug 15 12:51:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:51 charge currents: [7.04 7.03 7.03]A
Aug 15 12:51:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:51 detected phases: 3p
Aug 15 12:51:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:51 charger status: C
Aug 15 12:51:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:51 vehicle soc: 91%
Aug 15 12:51:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:51 vehicle range: 393km
Aug 15 12:51:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:51 pv charge current: 7.85A = 7A + 0.849A (-586W @ 3p)
Aug 15 12:51:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:51:51 max charge current: 7A
Aug 15 12:52:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:01 ----
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 charge power: 4900W
Aug 15 12:52:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:01 pv power: 5872W
Aug 15 12:52:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:01 battery power: -3W
Aug 15 12:52:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:01 grid power: -629W
Aug 15 12:52:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:01 battery soc 0: 90%
Aug 15 12:52:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:01 site power: -632W
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 charge currents: [7.04 7.03 7.03]A
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 detected phases: 3p
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 charger status: C
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 vehicle soc: 91%
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 vehicle range: 393km
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 pv charge current: 7.92A = 7A + 0.916A (-632W @ 3p)
Aug 15 12:52:02 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:02 max charge current: 7A
Aug 15 12:52:11 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:11 ----
Aug 15 12:52:11 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:11 charge power: 4900W
Aug 15 12:52:11 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:11 pv power: 5881W
Aug 15 12:52:11 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:11 battery power: -3W
Aug 15 12:52:11 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:11 grid power: -586W
Aug 15 12:52:11 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:11 battery soc 0: 90%
Aug 15 12:52:11 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:11 site power: -589W
Aug 15 12:52:11 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:11 charge currents: [7.04 7.03 7.03]A
Aug 15 12:52:11 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:11 detected phases: 3p
[... cut ...]
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 vehicle range: 393km
Aug 15 12:52:51 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:51 pv charge current: 26.2A = 12A + 14.1A (-3252W @ 1p)
Aug 15 12:53:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:53:01 ----
Aug 15 12:53:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:53:01 charge power: 2373W
Aug 15 12:53:01 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:53:01 pv power: 5935W

```

andig, 2022-08-19:
> Hier nochmal das LOG in ausführlicher Form. Vollständig geht nicht, da bei irgendwie 65000 Zeichen Schluss ist.

Du kannst Dateien anhängen und den sinnlosen Präfix raus nehmen, dann geht auch mehr. Also:

```
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 charge currents: [7.04 7.03 7.03]A
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 detected phases: 3p
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 charger status: C
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 vehicle soc: 91%
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 vehicle range: 393km
Aug 15 12:52:01 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:01 pv charge current: 7.92A = 7A + 0.916A (-632W @ 3p)
Aug 15 12:52:02 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:02 max charge current: 7A

```
Laden mit 3p, alles schön

```
Aug 15 12:52:21 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:21 charger status: B
Aug 15 12:52:21 raspberrypi evcc[11091]: [lp-1  ] INFO 2022/08/15 12:52:21 stop charging <-
```

Wallbox sagt "Ladung beendet". Achtung: evcc tut hier nix, wir sehen das nur!

```
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 charger status: C
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] INFO 2022/08/15 12:52:31 start charging ->
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 pv charge current: 10.1A = 2.01A + 8.1A (-5592W @ 3p)
Aug 15 12:52:31 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:31 max charge current: 10A
Aug 15 12:52:41 raspberrypi evcc[11091]: [site  ] DEBUG 2022/08/15 12:52:41 ----
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 charge currents: [6.05 0.011 0.01]A
Aug 15 12:52:41 raspberrypi evcc[11091]: [lp-1  ] DEBUG 2022/08/15 12:52:41 detected phases: 1p
```

Ladung geht weiter, wir sehen aber nur noch 1p.

**Diagnose**: die Easee hat von sich aus auf 1p reduziert. Evcc sieht das nicht, wir denken weiter 3p! Deshalb wird auch nicht mehr hoch geschaltet.

**Vorschlag**: Ticket bei Easee. Warum handelt die Box eigenmächtig?

@premultiply: sollten wir so ein Abfallen 3p->1p erkennen und dann unseren internen Status ebenfalls auf 1p oder unbekannt adaptieren damit wir eine Chance haben das zu erkennen?

Flexo0815, 2022-08-19:
Das Auto ist gerade „leider“ voll geladen ;-)

Vielleicht klappt es Sonntag.

Flexo0815, 2022-08-23:
```
Also hier ist der Moment, wo trotz genügend Überschuss plötzlich von 3p auf 1p reduziert wird. Ich habe aber in dem Moment den Herd eingeschaltet. Trotzdem sollte doch nicht sofort reduziert werden?! In der yaml habe ich für das Reduzieren 5m eingestellt. 

Aug 23 14:08:43 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:08:43 charge power: 5609W
Aug 23 14:08:43 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:08:43 pv power: 6404W
Aug 23 14:08:43 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:08:43 battery power: 816W
Aug 23 14:08:43 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:08:43 grid power: 1W
Aug 23 14:08:44 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:08:44 battery soc 0: 90%
Aug 23 14:08:44 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:08:44 site power: 817W
Aug 23 14:08:44 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:08:44 charge currents: [8.02 7.99 8.03]A
Aug 23 14:08:44 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:08:44 detected phases: 3p
Aug 23 14:08:44 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:08:44 charger status: C
Aug 23 14:08:44 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:08:44 vehicle soc: 76%
Aug 23 14:08:44 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:08:44 vehicle range: 330km
Aug 23 14:08:44 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:08:44 pv charge current: 6.82A = 8A + -1.18A (817W @ 3p)
Aug 23 14:08:44 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:08:44 max charge current: 6A
Aug 23 14:09:03 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:09:03 ----
Aug 23 14:09:03 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:09:03 charge power: 114W
Aug 23 14:09:03 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:09:03 pv power: 6390W
Aug 23 14:09:03 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:09:03 battery power: -3W
Aug 23 14:09:03 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:09:03 grid power: -3441W
Aug 23 14:09:04 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:09:04 battery soc 0: 90%
Aug 23 14:09:04 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:09:04 site power: -3443W
Aug 23 14:09:04 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:09:04 charge currents: [1.5 0.01 0.01]A
Aug 23 14:09:04 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:09:04 detected phases: 1p
Aug 23 14:09:04 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:09:04 charger status: C
Aug 23 14:09:04 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:09:04 vehicle soc: 76%
Aug 23 14:09:04 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:09:04 vehicle range: 330km
Aug 23 14:09:04 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:09:04 pv charge current: 18.5A = 3.5A + 15A (-3443W @ 1p)

Es bleibt dann dauerhaft bei 1p trotz Überschuss. 

Aug 23 14:15:23 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:15:23 charge power: 3727W
Aug 23 14:15:23 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:15:23 pv power: 6404W
Aug 23 14:15:23 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:15:23 battery power: -2W
Aug 23 14:15:23 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:15:23 grid power: -2395W
Aug 23 14:15:24 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:15:24 battery soc 0: 90%
Aug 23 14:15:24 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:15:24 site power: -2398W
Aug 23 14:15:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:15:24 charge currents: [15.8 0.01 0.01]A
Aug 23 14:15:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:15:24 detected phases: 1p
Aug 23 14:15:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:15:24 charger status: C
Aug 23 14:15:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:15:24 vehicle soc: 78%
Aug 23 14:15:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:15:24 vehicle range: 338km
Aug 23 14:15:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:15:24 pv charge current: 26.4A = 16A + 10.4A (-2398W @ 1p)

Wenn ich dann in der UI auf "schnell" schalte, bleibt es ebenfalls bei 1p. Es ändert sich nichts. 

Aug 23 14:31:23 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:31:23 charge power: 3751W
Aug 23 14:31:23 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:31:23 pv power: 6395W
Aug 23 14:31:23 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:31:23 battery power: -2W
Aug 23 14:31:23 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:31:23 grid power: -2419W
Aug 23 14:31:24 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:31:24 battery soc 0: 90%
Aug 23 14:31:24 raspberrypi evcc[27764]: [site  ] DEBUG 2022/08/23 14:31:24 site power: -2422W
Aug 23 14:31:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:31:24 charge currents: [16 0.023 0.01]A
Aug 23 14:31:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:31:24 detected phases: 1p
Aug 23 14:31:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:31:24 charger status: C
Aug 23 14:31:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:31:24 vehicle soc: 80%
Aug 23 14:31:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:31:24 vehicle range: 344km
Aug 23 14:31:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:31:24 pv charge current: 26.5A = 16A + 10.5A (-2422W @ 1p)
Aug 23 14:31:25 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:31:25 set charge mode: now
[... cut ...]
Aug 23 14:34:24 raspberrypi evcc[27764]: [lp-1  ] DEBUG 2022/08/23 14:34:24 max charge current: 16A

Wenn ich das zusammenfasse, dann beginnen die Probleme damit, dass im Haus an anderer Stelle mehr verbraucht wird. Auch wenn das nur ganz kurz der Fall ist, wird sofort und ab diesem Moment nur noch mit 1p im PV Modus geladen. Außer ich schalte auf "schnell" und vorher "aus". 

Der "PV" Modus ist dann dauerhaft auf 1p "festgefahren". Ich sehe da auch nicht den Grund, dass die Easee Wallbox den Fehler produziert. Die hält sich an das, was ich vorgebe. Bis auf das 1p Problem bei "PV". 
```

andig, 2022-08-23:
Das ist jetzt leider sehr unbefriedigend, aber. Im Logfile ist kein einziges Vorkommen von `switched phases`. Das bedeutet, dass hier nicht evcc umschaltet, sondern Deine Easee das selber macht- warum auch immer.

An der Stelle muss ich Dich leider an den Easee Support verweisen :( Wäre mal interessant herauszufinden, ob und welche Stellungnahme Dir Easee zu den jeweiligen Timestamps geben kann...