evcc-io/evcc #4127: UI: wakeup visualisieren

andig, 2022-08-16:
Hallo,

was ich so lese und selber feststelle, hat der VW ID ein Problem mit Ladeunterbrechnungen. Wird der Ladevorgang bei PV Überschussladen zweimal unterbrochen, startet der ID das Laden nicht mehr. Ich muss es in der App manuell starten. Dann geht es wieder.

Wäre es möglich dies über die API zu tun? Sprich wenn genügend Strom zur Verfügung steht und EVCC ist im Status "Warten auf Fahrzeug", dann über die API dem Auto zu sagen, das Laden zu starten?

Viele Grüße

Carsten

_Originally posted by @diomedon in https://github.com/evcc-io/evcc/discussions/4125_

naltatis, 2022-08-16:
@andig kannst du beschreiben welche Idee du hier hast? Mir ist das noch nicht ganz klar. Manuelle Option einen Refresh anzutriggern? Also ähnlich Fahrzeug wechseln?

andig, 2022-08-16:
Nur Visualisierung wenn ein Refresh läuft. Manuelles Triggern würd ich gerne vermeiden.

diomedon, 2022-08-16:
Bin gespannt 😀
Log kommt morgen. Brauche genügend Sonne und ein paar Wolken.

naltatis, 2022-08-17:
Also eine Meldung wie "Fahrzeug wird geweckt" anstatt "Warte auf Fahrzeug"?

Was wir ja auch noch auf der Liste haben ist beim Fahrzeug eine aktualisieurngszeit anzuzeigen. Hatten wir in den Entwürfen auch schon mal dein: "vor 10 Minuten aktualisiert"

Würde euch das helfen? Das fände ich nämlich ein generell sinnvolles Feature.

andig, 2022-08-17:
Ich schau mal was wir einfach anbieten könnten. Die Aktualisierungszeit aus mehreren Lagen Caches wird eher schwierig, dazu müsste das egtl. Der Timestamp sein zu dem das API zuletzt Daten vom Fahrzeug erhalten hatte. Typischerweise gibts da nicht „den einen“ Timestamp was es noch unübersichtlicher macht…

naltatis, 2022-08-17:
> Der Timestamp sein zu dem das API zuletzt Daten vom Fahrzeug erhalten hatte.

Das wäre in der Tat aber das Schönste :D

premultiply, 2022-08-17:
Ich denke hier sind die beiden Timestamps "letzten Datenupdate in der API vom Fahrzeug" und "letzter API-Abruf (d.h. Cachedatenstand) durch evcc" am interessantesten.
Bekommen wir ersteren Timestamp irgendwie bei allen APIs?

andig, 2022-08-17:
Nope…

andig, 2022-08-17:
Laut Code gibt es einen Timer innerhalb dessen wir schauen ob Ladung freigegeben aber Ladestart nicht begonnen. Eine Art Watchdog der läuft bis Charger/Auto nochmal getriggert werden. Die Phase könnten wir Visualisieren. https://github.com/evcc-io/evcc/pull/2543 versucht dann nochmal, einen oben drauf zu setzen und das gestaffelt zu machen.

Wenn wir das nicht visualisieren wollen können wir hier zu machen.

premultiply, 2022-08-17:
Ich fände eine Wakeup-Vis schon hilfreich.

andig, 2022-09-04:
@naltatis Hatten wir denn jetzt eine Idee, wie wir das Wakeup signalisieren wollen? Also "will ping vehicle in x seconds"?

naltatis, 2022-09-04:
Ja, ich würde das erstmal rein über den Statustext lösen. Heute zeigen wir da ja schon "Warte auf Fahrzeug" an. Solange wir aktiv über API versuchen das Fahrzeug zu wecken würde ich da "Fahrzeug wird geweckt" / "Waking up vehicle" anzeigen. Ich bräuchte halt nen Flag/Bool in der API womit ich das erkennen kann. Danach (egal ob Wecken erfolgreich war oder nicht) sollte das Flag dann wieder aus gehen.

andig, 2022-09-04:
Ich würde dann einen Timer schicken, analog pvEnable/disable? Das event selbst dann eher nicht oder willst du das auch im Ui (als was) darstellen?

naltatis, 2022-09-05:
Vielleicht verstehe ich den Kontext noch nicht ganz. Ich hätte jetzt gedacht, dass wir versuchen das Fahrzeug zu wecken und dass wir hier über einen kurzen Zeitraum (wenige Minuten) sprechen.

Ob das jetzt durch einen oder mehrere API Calls passiert und wie oft wir nen Retry machen halte ich aus Nutzersicht für eher nicht so wichtig. Das verhält sich von Hersteller zu Hersteller ja auch unterschiedliche. An dieser Stelle würde ich die Nutzerkommunikation reduziert halten. Im Sinne von "Dein Fahrzeug schläft, aber wir kümmern uns gerade darum, dass es wach wird".

naltatis, 2022-09-05:
Ok

closed 2022-11-01 by andig