evcc-io/evcc #4134: Delay message "ignoring inconsistent phases"

MarkusGH, 2022-08-17:
For some reason my openWB based wallbox needs up to 90 sec to switch phases.

I suggest delaying the message ignoring inconsistent phases (loadpoint.go:1101) by 120 sec:

if phases > 0 && phases < measuredPhases && lp.clock.Since(lp.phaseTimer) > lp.Disable.Delay + 120 {
	lp.log.WARN.Printf("ignoring inconsistent phases: %dp < %dp observed active", phases, measuredPhases)
}

andig, 2022-08-17:
Das ist allgemeingültig nicht anwendbar. Da es sich *nur* um eine Warnung handelt eigentlich auch erstmal kein Problem?

MarkusGH, 2022-08-17:
Unnötige Warnungen verschönern das UI nicht gerade. Und User auf das ignorieren von Warnungen zu trainieren ist generell keine gute Idee.

MarkusGH, 2022-08-17:
Und warum soll das allgemeingültig nicht anwendbar sein?

andig, 2022-08-17:
> Und warum soll das allgemeingültig nicht anwendbar sein?

Deine Zeitkonstante bezieht sich auf die OpenWB.

> Beim Enabled state wird dieselbe Strategie angewendet.

Guter Punkt. Da Du eh schon im Code bist- gerne direkt als PR!

MarkusGH, 2022-08-17:
120 sec sind 120 sec
Um zu forken und einen einen PR zu stellen fehlt mir die Zeit und die Übung.

andig, 2022-08-17:
Die Zeit fehlt mir leider auch...

MarkusGH, 2022-08-17:
Für Dich sind das 30 Sekunden, für mich 60 Minuten...

panzerdev, 2022-08-19:
Ich würde da gern helfen hätte nur evtl. einen anderen Ansatz.

bei `openwb-pro` steht folgendes für den phasenswitch
```
func (wb *OpenWBPro) Phases1p3p(phases int) error {
	return wb.set(fmt.Sprintf("phasetarget=%d", phases))
}
```

Wäre es nicht denkbar den returnwert hier so lange hinauszuzögern bis die Wallbox den erfolgten switch bestätigt und wenn es nicht klappt einen Fehler zurückzugeben?

Bei meiner eigenen Implementierung für meine WB habe ich das auch so gemacht, dass in diesem Aufruf mehr passiert als nur der Switch und erst wenn alles durch ist, dann wird returned.

andig, 2022-08-19:
> Wäre es nicht denkbar den returnwert hier so lange hinauszuzögern 

Nein. Wenn wir hier warten stehen alle anderen Ladepunkte. Das *muss* asynchron passieren. Wir reden hier über ein rein optisches Problem!

premultiply, 2022-08-20:
Hilft es bei der Box vielleicht den Ladevorgang vorher aktiv zu unterbrechen oder dort ein Ticket aufzumachen?
Ich sehe das eher als Bug der Box an.

andig, 2022-09-17:
> Hilft es bei der Box vielleicht den Ladevorgang vorher aktiv zu unterbrechen oder dort ein Ticket aufzumachen?

Das machen wir immer.

> Ich sehe das eher als Bug der Box an.

Selbst wenn- hier gilt es einfach zu entscheiden ob wir das eine zeitlang tolerieren wollen? Fein für mich wenn nicht.

andig, 2022-09-25:
@premultiply wie machen wir hier weiter?

closed 2022-10-22 by github-actions[bot]