evcc-io/evcc #4138: openWB: charger out of sync

MarkusGH, 2022-08-17:
### Describe the bug

Messages "charger out of sync: expected disabled, got enabled" after evcc stops charge in pv mode.

I assume the reason is that the implementations of Enable() and Enabled() in openwb.go do not match:

func (m *OpenWB) Enable(enable bool) error {
	var current int64
	if enable {
		current = m.current
	}

	return m.currentS(current)
}

func (m *OpenWB) Enabled() (bool, error) {
	// current, err := m.enabledG()
	return m.current > 0, nil
}

Enable() setter never updates m.current but the getter queries m.current.

It seems either openWB should be queried for the actual currents or Enable() has to update m.current (old value is lost then):

func (m *OpenWB) Enable(enable bool) error {
	var current int64
	if enable {
		current = m.current
	} else {
		m.current = current
	}
	return m.currentS(current)
}

Or another state variable "enabled" is necessary to mirror openWB's state.

### Steps to reproduce

Use openWB and stop charging

### Configuration details

openWB

### Log details

Aug 17 17:40:11 evcc[3320281]: [lp-1  ] DEBUG 2022/08/17 17:40:11 charge power: 0W
Aug 17 17:40:11 evcc[3320281]: [lp-2  ] DEBUG 2022/08/17 17:40:11 charge power: 0W
Aug 17 17:40:11 evcc[3320281]: [lp-2  ] DEBUG 2022/08/17 17:40:11 charge currents: [0 0 0]A
Aug 17 17:40:11 evcc[3320281]: [lp-2  ] DEBUG 2022/08/17 17:40:11 charger status: A
Aug 17 17:40:12 evcc[3320281]: [openwb] TRACE 2022/08/17 17:40:12 recv openWB/system/Timestamp: '1660750812'
Aug 17 17:40:13 evcc[3320281]: [openwb] TRACE 2022/08/17 17:40:13 send openWB/set/isss/heartbeat: '1'
Aug 17 17:40:14 evcc[3320281]: [openwb] TRACE 2022/08/17 17:40:14 send openWB/set/isss/heartbeat: '1'
Aug 17 17:40:14 evcc[3320281]: [lp-1  ] DEBUG 2022/08/17 17:40:14 charge power: 0W
Aug 17 17:40:14 evcc[3320281]: [lp-2  ] DEBUG 2022/08/17 17:40:14 charge power: 0W
Aug 17 17:40:14 evcc[3320281]: [lp-1  ] DEBUG 2022/08/17 17:40:14 charge currents: [0 0 0]A
Aug 17 17:40:14 evcc[3320281]: [lp-1  ] DEBUG 2022/08/17 17:40:14 charger status: B
Aug 17 17:40:14 evcc[3320281]: [lp-1  ] WARN 2022/08/17 17:40:14 charger out of sync: expected disabled, got enabled
Aug 17 17:40:14 evcc[3320281]: [openwb] TRACE 2022/08/17 17:40:14 send openWB/set/isss/Lp2Current: '0'
Aug 17 17:40:14 evcc[3320281]: [lp-1  ] DEBUG 2022/08/17 17:40:14 pv charge current: 0.556A = 0A + 0.556A (-128W @ 1p)
Aug 17 17:40:14 evcc[3320281]: [lp-1  ] DEBUG 2022/08/17 17:40:14 climater active: false, target temp: 20.0°C, outside temp: 20.0°C

### What type of operating system are you running?

Linux

### Version

0.100

andig, 2022-08-17:
Bitte formatieren

MarkusGH, 2022-08-17:
Was meinst Du genau?

andig, 2022-08-17:
Nutz mal bitte die Funktionen des Texteditors um Code/Log lesbar zu machen:
[image]

MarkusGH, 2022-08-17:
Danke - kannte ich nicht.

andig, 2022-08-17:
> It seems either openWB should be queried for the actual currents 

Wenns denn ginge- ist leider nicht im Angebot :(

> Enable() setter never updates m.current but the getter queries m.current.

Mit Absicht. Entweder setzen wir Strom `0` (=disable) oder den letzten (=enable). Und genau das gibt `Enabled()` dann wieder.

Ich sehe da kein Problem?

andig, 2022-08-17:
...und ich versteh leider nicht, wo/welches Problem Du in dem Logschnipsel siehst?

MarkusGH, 2022-08-17:
> ...und ich versteh leider nicht, wo/welches Problem Du in dem Logschnipsel siehst?

WARN 2022/08/17 17:40:14 charger out of sync: expected disabled, got enabled

andig, 2022-08-17:
Ich sehe da keinen Zusammenhang zu Deiner Ursachendiagnose. Worin siehst Du den? Was der Code tut hab ich oben erklärt. 

> Ich sehe da kein Problem?

Welches siehst Du und wo finde ich das im Log?

MarkusGH, 2022-08-17:
> > It seems either openWB should be queried for the actual currents

> Wenns denn ginge- ist leider nicht im Angebot :(

Es gibt doch currentsG? Der holt doch die Ströme?

> > Enable() setter never updates m.current but the getter queries m.current.
> 
> Mit Absicht. Entweder setzen wir Strom `0` (=disable) oder den letzten (=enable). Und genau das gibt `Enabled()` dann wieder.
> 
> Ich sehe da kein Problem?

Ich schon.

Stellen wir uns vor m.current ist 6.
Enable(false) ändert m.current nicht.
Beim nächsten Aufruf gibt dann Enabled() true zurück und es kommt die "charger out of sync" Warnung.

andig, 2022-08-17:
> Es gibt doch currentsG? Der holt doch die Ströme?

Die gemessenen. Hier brauchts den Freigegebenen. Aber weder `AConfigured` noch `ChargePointEnabled` scheinen im "Nur Ladepunkt" Modus zu funktionieren. Feedback: keins (https://github.com/snaptec/openWB/issues/1761).

> Ich schon.

Sehr gut! Dank etwas mehr Kontext hab ichs jetzt auch verstanden...

andig, 2022-08-17:
Vorschlag: hak doch mal bei dem Ticket ein. Vielleicht findest Du als Kunde ja Gehör...

andig, 2022-08-17:
https://openwb.de/forum/viewtopic.php?f=4&t=5646 wäre bei der Gelegenheit auch noch super :)

MarkusGH, 2022-08-17:
> Vorschlag: hak doch mal bei dem Ticket ein. Vielleicht findest Du als Kunde ja Gehör...

Eher nicht - meine openWB ist Eigenbau.

MarkusGH, 2022-08-17:
> > Es gibt doch currentsG? Der holt doch die Ströme?
> 
> Die gemessenen. Hier brauchts den Freigegebenen. Aber weder `AConfigured` noch `ChargePointEnabled` scheinen im "Nur Ladepunkt" Modus zu funktionieren. Feedback: keins ([snaptec/openWB#1761](https://github.com/snaptec/openWB/issues/1761)).

Habe gerade das betreffende Modul in der openWB Software angeschaut (isss.py). Sieht ganz so aus als ob weder AConfigured noch ChargePointEnabled auf MQTT geschrieben werden. Und ich habe auch keine Stelle gefunden wo Cpulp1 gelesen wird. Geschrieben wird es allerdings schon wenn die Software im "Master" Modus läuft.

MarkusGH, 2022-08-17:
> > Es gibt doch currentsG? Der holt doch die Ströme?
> 
> Die gemessenen. Hier brauchts den Freigegebenen. 

Naja - anstatt sich zu merken was man zuletzt geschrieben hat wäre es immer noch besser herauszufinden was openWB tut.
Insofern wären die gemessenen Ströme immer noch besser als nix.

andig, 2022-08-17:
Das täuscht :/

andig, 2022-08-17:
Closed in https://github.com/evcc-io/evcc/commit/67bf75de8e043c8bec95146e929c9a1fe9d32d72

closed 2022-08-17 by andig

MarkusGH, 2022-08-18:
> Das täuscht :/

Hab nochmals darüber nachgedacht - Du hast zu 50% recht..

Ladestrom > 0 -> Charger ist enabled.
Ladestrom == 0 -> Zustand unbekannt.

Also könnte man in Enabled() den Ladestrom mit currentsG prüfen, wenn dieser auf allen Phasen 0 ist kann man immer noch m.enabled zurückgeben.