evcc-io/evcc #4159: Ladelog umsetzen

andig, 2022-08-19:
Anforderungen:

- Quelldaten durch Anwender bearbeitbar (=bekanntes Format)
- exportierbar
- über UI einsehbar

Weiß jemand, in welcher Form z.B. `openWB` das umgesetzt hat?

Wie wäre z.B. eine sqlite Datenbank?

premultiply, 2022-08-20:
Ich hätte gerne noch ein paar mehr Ausgabedaten, da diese u.a. für Statistiken und zur Validierung (z. B. perspektivisch zu Abrechnungszwecken) mindestens nützlich bis notwendig sein können:

- [ ] connected datetime
- [ ] disconnected datetime
- [ ] connect duration
- [ ] charge start datetime
- [ ] charge end datetime
- [ ] charge duration
- [ ] vehicle identifier
- [ ] vehicle title
- [ ] soc start
- [ ] soc end
- [ ] charge meter total start
- [ ] charge meter total end
- [ ] charged energy
- [ ] mean charge power
- [ ] max charge power

bedmanforever, 2022-08-20:
Hi,

wenn ihr die ein Ladelog umgesetzt, wäre es auch sinnvoll zusätzlich die geladene Energie von PV und extern zu unterscheiden:
- charged energy total
- charged energy PV
- charged energy grid

Das wäre hilfreich, falls man den über PV geladenen Strom auf die THG Quote anrechnen kann.

andig, 2022-08-20:
Wie soll das gehen bei mehreren Ladepunkten?

bedmanforever, 2022-08-20:
PV und Grid total sollte reichen. Falls es bloß pro Loadpoint bzw. Fahrzeug
geht, könnte man über dynamische Logeinträge der Art:

- [ ] energy-<total/pv/grid>-<loadpoint-id>-<vehicle-id>

nachdenken?

andig, 2022-08-20:
@premultiply die meisten deiner Daten hatte ich oben schon erwähnt bzw. lassen sich leicht errechnen. Welche fehlen wirklich?

Hofyyy, 2022-09-01:
Hallo,

Ich finde die initial Liste der Informationen sehr gut und mehr Werte kann man ja immer noch einbauen.
Für privates Wallbox Sharing, wäre es hilfreich in dem Rahmen z.B. die letzten 5 Ladevorgänge o.ä. im UI nachschauen zu können. Erstmal müsste das aus meiner Sicht auch nicht persistent sein (auch wenn das natürlich schön wäre).

Mein Usecase dazu ist im Moment, das öffentlich stellen der Wallbox mit &Charge. Wo man recht pragmatisch und manuell die geladene KWh mit dem "Kunden" abrechnet.

Also das Issue hier ist top.

JFI: (ein wenig offtopic)
Ich habe einen Dev Zugang und Kontakte bei &Charge angefragt, da es der einzige Partner ist um nicht geeichte Boxen zu sharen. Aktuell können die nur autom. mit Easee arbeiten. Allerdings halte ich es für simpel letzte Ladevorgang mit einer ID von &Charge er HTTP irgendwo hinzuschicken, gerade wenn wir dieses Issue sauber umsetzen, dann hat man ja die Daten.

closed 2022-10-05 by andig