evcc-io/evcc #4205: Liviness probe

poggenpower, 2022-08-23:
**Is your feature request related to a problem? Please describe.**
The Dockerfile has a HEALTHCHECK configured, unfortunately this is not executed when running this image in a Kubernetes Cluster. Kubernetes has the concept of liviness and health probes which should be supported. 
 
**Describe the solution you'd like**
As evcc already exposed a webservice it would be nice if there is a route like `/health` that allows healthchecks of the service

**Describe alternatives you've considered**
If there is already another route exposing an health state, this could be used instead. 

**Additional context**
Even hat home the container environments moving from docker compose to kubernets as an orchestrator. Would be great if this would be supported. An health check route could also be use by external monitoring systems to get the state of evcc.

andig, 2022-08-23:
Ich hab leider keine Ahnung von Kubernetes- magst Du einen PR erstellen?

poggenpower, 2022-08-23:
Leider sind meine GO-Kenntnisse sehr, sehr eingeschraenkt. Aber vielleicht ist das ja ein guter Einstieg.

andig, 2022-08-23:
Lass uns mal vorne anfangen. Das Api gibts ja schon. Was fehlt konkret?

poggenpower, 2022-08-23:
Sorry ich haette bevor ich hier einen feature Request abwerfe genauer recherchieren muessen. 
Ich habe mich jetzt selber noch ein bisschen schlau gemacht, was Kubernetes eigentlich checkt. 
Nach ein bisschen Recherce habe ich gelernt, dass Kubernetes auch ein Kommando fuer den Healtcheck ausfuehren kann, aehnlich wie Docker.
`          livenessProbe:
            exec:
              command:
              - evcc
              - health
            initialDelaySeconds: 60
            periodSeconds: 60
            timeoutSeconds: 30
`
sollte genau das gleiche machen wie der Docker Healthcheck. 
Ist natuerlich ein bisschen mehr Overhead als ein HTTP request. Bei den HTTP(S) liveness checks muss der Service einfach nur einen Status-Code von 200 bis 399 zurueckgeben, dann gilt der Service als "alive".
Wenn ich mir die API anschaue koennte `/api/health` sich ja schon genau so verhalten.  
Wenn ich Zeile https://github.com/evcc-io/evcc/blob/1b1358cb3f8b04da02e0d9ab825de9153a044d36/server/http_handler.go#L74 richtig verstehe ist das auch so. 

Gibt es eine Moeglichkeit evcc kuenstlich in einen "unhealthy" Zustand zu versetzen, dann kann ich das mal testen.

andig, 2022-08-23:
Einfach Prozess abschiessen? Scheint als hätten wir schon alles an Board- ich verschiebe daher nach Discussions.