evcc-io/evcc #4214: InfluxDB Intervall konfigurierbar machen

Chris8er, 2022-08-24:
**Is your feature request related to a problem? Please describe.**
Ich beschäftige mich gerade mit Statistiken zu meiner PV Anlage / Stromverbrauch. Ich habe mit Solaranzeige, IOBroker Adaptern etc. rum probiert. Aber irgendwie ist es gar nicht so leicht, die Daten des Sunny Homemanagers und allen anderen Geräten aufbereitet in die Influx DB zu bekommen (vor allem der derzeitige Verbrauch des Hauses). Dann habe ich die InfluxDB Option von evcc entdeckt und war heilfroh, denn evcc stellt hierüber genau die Daten zur Verfügung, die ich haben möchte.
Ich möchte aber nun eigentlich gar nicht alle 10s Daten in die Influx schreiben. Ich möchte für das Regeln des Wallboxen ein kleines und für die Statistik ein größeres Intervall haben.

**Describe the solution you'd like**
Ich möchte im Influx Config-Block der evcc.yaml ein eigenes Influx-Sync Intervall angeben können.

**Describe alternatives you've considered**
Retention Policies und Continuous Queries brächten theoretisch Abhilfe. Allerdings sind diese relativ schwer und aufwändig zu konfigurieren (vor allem für so viele measurements) und es bliebe die Last auf dem Speicher.

andig, 2022-08-24:
> Alternativ wäre ggf. eine Prometheus kompatible Seite, die alle Daten beinhaltet.

Das können wir auf jeden Fall machen, ist allerdings ein ziemliches Techiefeature- den Bedarf für "normale" Anwender sehe ich nur sehr eingeschränkt.

Für InfluxDB wäre mir nicht klar, wie wir das machen sollen. Einfach Werte weg lassen? Welche? Alle "zwischenzeitlichen"?

premultiply, 2022-08-24:
Ich verstehe denn Mehrwert eines niedrigeren Samplingintervalls auch noch nicht so richtig.
Ist es nicht gerade ein Feature von Zeitreihendatenbanken wie Influx dass man da bei der Abtastrate der Ausgabe sehr flexibel ist?

andig, 2022-08-28:
> Für InfluxDB wäre mir nicht klar, wie wir das machen sollen. Einfach Werte weg lassen? Welche? Alle "zwischenzeitlichen"?

@Chris8er ?

Chris8er, 2022-08-29:
Primär ging es mir nur darum, nicht so viele Daten zu speichern. So ein Raspi hat ja nicht unendlich resourcen. Bei der Auswertung reicht mir eine geringere Auflösung.

Im IOBroker Inflxu-Adapter gibt es z.B. folgende Option:

[image]

Gab es dazu nicht schon evtl. mal Überlegungen? Siehe:

https://github.com/evcc-io/evcc/blob/1464b755cd898e4b8d34f94ff47b2ba7b4d02d55/cmd/setup.go#L87

markuspulch, 2022-09-09:
@Chris8er Influxdb ist eigentlich sehr effizient darin viele Daten mit geringer IO-Last in Filesystem zu schreiben. Das sollte also kein Problem sein (ich lasse selbst eine Vergleichsweise große InfluxDB auf einem Raspberry Pi4 laufen).
Falls Storage Volume das Problem sein sollte: schreib die Daten in ein fine-granulares Bucket mit einer kurzen retention period (24h). Parallel läßt do eine continous query laufen welche ein downsampling ein ein anderes Bucket mit einer langen retention period macht, zb. stündliche samples macht und 1 Jahr aufhebt.
Das ist bei Influx eigentlich best practice.

Markus

sebnaf, 2022-09-16:
> schreib die Daten in ein fine-granulares Bucket mit einer kurzen retention period (24h). Parallel läßt do eine continous query laufen welche ein downsampling ein ein anderes Bucket mit einer langen [...]

Genau so :)

andig, 2022-09-30:
Keine weiteren Erkenntnisse

closed 2022-09-30 by andig