evcc-io/evcc #4220: Ladung stoppt nicht bei vorgegebenem Ladeziel

stdt, 2022-08-24:
### Describe the bug

Ladeziel bei Fahrzeug Mokka_e und Wallbox Wallbe wird z.B. auf 80 % gesetzt.
Ladung stoppt nicht bei 80%.

### Steps to reproduce

Ladung auf Min+PV stellen. 
Mokka_e mit Ladestand 79% (oder geringer) anstecken.
Ladeziel auf 80% festlegen.
Warten.
Ladung stoppt nicht bei 80% 
Debug log vom Ladepunkt (lp-1 - siehe log details weiter unten) gibt immer den Anfangs SoC des Fahrzeugs wieder:
vehicle soc: 79%

### Configuration details

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  lp-1: debug
  opel: debug

interval: 10s # control cycle interval

meters:
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 10.11.12.14  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  timeout: 30s  
  name: grid1
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 10.11.12.14  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  timeout: 30s  
  name: pv2

chargers:
- type: template
  template: wallbe 
  host: 10.11.12.15  
  port: 502  
  name: wallbox4

vehicles:
- type: template
  template: opel 
  title: Mokka_e  
  user: xxxxx@xxxxx.de  
  password: 123456qwertzuiop  
  vin: VXKTRALALA0815  
  capacity: 46  
  name: ev3

loadpoints:
- title: Garage
[... 114 lines cut ...]
Aug 24 12:20:22 dahaampi evcc[7126]: [lp-1  ] DEBUG 2022/08/24 12:20:22 pv charge current: 6.73A = 7.85A + -1.12A (776W @ 3p)
Aug 24 12:20:22 dahaampi evcc[7126]: [lp-1  ] DEBUG 2022/08/24 12:20:22 max charge current: 6.73A
^C
pi@dahaampi:~ $ evcc -l debug vehicle
[main  ] INFO 2022/08/24 12:20:30 evcc 0.100
[main  ] INFO 2022/08/24 12:20:30 using config file /etc/evcc.yaml
SoC:            85%
Charge status:  C
Range:          232km
Odometer:       31695km
Finish time:    2022-08-24 12:59:00 +0200 CEST
Climate active: false
Outside temp:   20.0°C
Target temp:    20.0°C
Position:       not available
Capacity:       46kWh
pi@dahaampi:~ $ 

### What type of operating system are you running?

Linux

### Version

evcc version 0.100

premultiply, 2022-08-24:
https://docs.evcc.io/docs/reference/configuration/loadpoints/#estimate

andig, 2022-08-24:
Stoppt es denn überhaupt oder gar nicht? Siehe auch https://github.com/evcc-io/evcc/issues/3977

stdt, 2022-08-24:
Lädt durch bis 100%.
Also stoppt gar nicht. 
Es kann auch auf 90% Ladeziel eingestellt werden. Ladung stoppt nicht, Ladung ist erst beendet, wenn Auto dann bei 100% voll ist.

premultiply, 2022-08-24:
@stdt Kannst du das bitte auch nochmal mit aktivem Estimator gegenchecken?

Ich gehe davon aus dass da „nur“ keine Updates über die Vehicle-API reinkommen.

stdt, 2022-08-24:
Bin gerade dabei ...

premultiply, 2022-08-24:
Relevant ist nur
```yaml
  soc:
    estimate: true
```

andig, 2022-08-24:
> Also stoppt gar nicht.

@premultiply wir wissen ja mittlerweile, dass der Estimator funktioniert. Sollen wir den mal als Standard setzen und nur bei Bedarf *deaktivierbar* machen, per Default aber "on"?

premultiply, 2022-08-24:
Genau das hab ich gerade eben auch gedacht :)
Die Fälle wo er einen negativen Einfluss hätte und das Fahrzeug Daten im Minutentakt liefert kann man ja an einer Hand abzählen.

stdt, 2022-08-24:
Also mit estimate hört er beim angegebenen Ladeziel (90%) auf.

`Aug 24 15:01:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:13 vehicle soc: 89%
Aug 24 15:01:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:13 vehicle range: 234km
Aug 24 15:01:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:13 pv charge current: 2.38A = 6A + -3.62A (2498W @ 3p)
Aug 24 15:01:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:23 charge power: 4140W
Aug 24 15:01:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:23 charger status: C
Aug 24 15:01:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:23 soc estimated: 89.88% (vehicle: 85.00%)
Aug 24 15:01:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:23 vehicle soc: 89%
Aug 24 15:01:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:23 vehicle range: 234km
Aug 24 15:01:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:23 pv charge current: 2.28A = 6A + -3.72A (2566W @ 3p)
Aug 24 15:01:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:33 charge power: 4140W
Aug 24 15:01:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:33 charger status: C
Aug 24 15:01:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:33 soc estimated: 89.90% (vehicle: 85.00%)
Aug 24 15:01:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:33 vehicle soc: 89%
Aug 24 15:01:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:33 vehicle range: 234km
Aug 24 15:01:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:33 pv charge current: 2.24A = 6A + -3.76A (2592W @ 3p)
Aug 24 15:01:43 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:43 charge power: 4140W
Aug 24 15:01:43 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:43 charger status: C
Aug 24 15:01:43 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:43 soc estimated: 89.93% (vehicle: 85.00%)
Aug 24 15:01:43 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:43 vehicle soc: 89%
Aug 24 15:01:43 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:43 vehicle range: 234km
Aug 24 15:01:43 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:43 pv charge current: 1.08A = 6A + -4.92A (3394W @ 3p)
Aug 24 15:01:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:53 charge power: 4140W
Aug 24 15:01:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:53 charger status: C
Aug 24 15:01:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:53 soc estimated: 89.95% (vehicle: 85.00%)
Aug 24 15:01:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:53 vehicle soc: 89%
Aug 24 15:01:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:53 vehicle range: 234km
Aug 24 15:01:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:01:53 pv charge current: 2.23A = 6A + -3.77A (2600W @ 3p)
Aug 24 15:02:03 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:03 charge power: 4140W
Aug 24 15:02:03 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:03 charger status: C
Aug 24 15:02:03 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:03 soc estimated: 89.97% (vehicle: 85.00%)
Aug 24 15:02:03 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:03 vehicle soc: 89%
Aug 24 15:02:03 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:03 vehicle range: 234km
Aug 24 15:02:03 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:03 pv charge current: 1.38A = 6A + -4.62A (3186W @ 3p)
Aug 24 15:02:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:13 charge power: 4140W
Aug 24 15:02:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:13 charger status: C
Aug 24 15:02:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:13 soc estimated: 89.99% (vehicle: 85.00%)
Aug 24 15:02:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:13 vehicle soc: 89%
Aug 24 15:02:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:13 vehicle range: 234km
Aug 24 15:02:13 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:13 pv charge current: 2.19A = 6A + -3.81A (2628W @ 3p)
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 charge power: 4140W
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 charger status: C
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 soc estimated: 90.02% (vehicle: 85.00%)
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 vehicle soc: 90%
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 vehicle range: 234km
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 targetSoC reached: 90.0 > 90
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 climater active: false, target temp: 20.0°C, outside temp: 20.0°C
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 charger disable
Aug 24 15:02:23 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:23 wake-up timer: stop
Aug 24 15:02:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:33 charge power: 0W
Aug 24 15:02:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:33 charger status: B
Aug 24 15:02:33 dahaampi evcc[12822]: [lp-1  ] INFO 2022/08/24 15:02:33 stop charging <-
Aug 24 15:02:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:33 soc estimated: 90.02% (vehicle: 85.00%)
Aug 24 15:02:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:33 vehicle soc: 90%
Aug 24 15:02:33 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:33 vehicle range: 234km
[... cut ...]
Aug 24 15:02:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:53 charge power: 0W
Aug 24 15:02:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:53 charger status: B
Aug 24 15:02:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:53 targetSoC reached: 90.0 > 90
Aug 24 15:02:53 dahaampi evcc[12822]: [lp-1  ] DEBUG 2022/08/24 15:02:53 climater active: false, target temp: 20.0°C, outside temp: 20.0°C`

Wobei das Auto selbst 91% anzeigt. Was jetzt aber kein Problem für mich wäre. ;-)

stdt, 2022-08-24:
PS: warum zeigt er den Code so bescheiden an?

VolkerK62, 2022-08-24:
Hängt das evtl hiermit zusammen?
https://docs.evcc.io/docs/guides/vehicles#psa-bei-meinem-peugeotopel-wird-der-ladezustand-nur-aktualisiert-wenn-ich-die-hersteller-app-benutze

stdt, 2022-08-24:
Das kann ich die Tage gerne mal ausprobieren, wenn der Ladestand wieder unten ist...

stdt, 2022-08-24:
Interessant auch, dass das Auto nicht gleich per _evcc -l debug vehicle_ korrekt den Ladestand übermittelt.
Gleich nach der Ladung wurde ein Ladestand von 85% übermittelt, erst später dann 90%.
Das könnte für @VolkerK62 s Hinweis sprechen, oder?

`pi@dahaampi:/etc $ evcc -l debug vehicle  
[main  ] INFO 2022/08/24 15:03:45 evcc 0.100  
[main  ] INFO 2022/08/24 15:03:45 using config file /etc/evcc.yaml  
**SoC:            85%**  
Charge status:  B  
Range:          234km  
Odometer:       31695km  
Finish time:    2022-08-24 12:58:00 +0200 CEST  
Climate active: false  
Outside temp:   20.0°C  
Target temp:    20.0°C  
Position:       not available  
Capacity:       46kWh  
pi@dahaampi:/etc $ evcc -l debug vehicle  
[main  ] INFO 2022/08/24 15:05:25 evcc 0.100  
[main  ] INFO 2022/08/24 15:05:25 using config file /etc/evcc.yaml  
**SoC:            86%**  
Charge status:  C  
Range:          236km  
Odometer:       31695km  
Finish time:    2022-08-24 18:12:00 +0200 CEST  
Climate active: false  
Outside temp:   20.0°C  
Target temp:    20.0°C  
Position:       not available  
Capacity:       46kWh  
pi@dahaampi:/etc $ evcc -l debug vehicle  
[main  ] INFO 2022/08/24 15:09:05 evcc 0.100  
[main  ] INFO 2022/08/24 15:09:05 using config file /etc/evcc.yaml  
**SoC:            91%**  
Charge status:  B  
Range:          254km  
Odometer:       31695km  
Finish time:    2022-08-24 15:05:00 +0200 CEST  
Climate active: false  
Outside temp:   20.0°C  
Target temp:    20.0°C  
Position:       not available  
Capacity:       46kWh   
pi@dahaampi:/etc $ `

andig, 2022-08-24:
Auto-Apis sind träge. Deshalb sollten wir lieber den Estimator nutzen.

andig, 2022-08-24:
> PS: warum zeigt er den Code so bescheiden an?

Falsch formatiert

closed 2022-08-24 by andig