evcc-io/evcc #4228: Zugriff auf die evcc.yaml Datei

die4M, 2022-08-24:
Liebe Community,

ich bin total neu im Linux Feld und stolz auf mich, wie weit ich schon mit evcc gekommen bin. Nun verzweifle ich jedoch bei sowas simplen, wie auf die evcc.yaml zuzugreifen, um sie nach dem evcc config Befehl nochmal anzupassen. Wie komme ich da ran????
Hatte mir einen sftp Programm geholt aber damit kommt man ja wohl nicht auf die root Einstellung und somit auf den /etc.
Auf welchen Weg greif man am besten auf die evcc.yaml zu?

Danke

RTTTC, 2022-08-24:
For me this works: `sudo nano /etc/evcc.yaml`

die4M, 2022-08-24:
Many thanks, now I can see the content of the file and can edit it. But how can I save it? it tells me "^O" is save but nothing happens when I enter this combination. Sorry I am really a linux newbie...

RTTTC, 2022-08-24:
`^x` to close, click `y` to save changes

die4M, 2022-08-24:
Thanks! It works :-)