evcc-io/evcc #4238: RCT Power Batterie SOC wird nicht ausgelesen

diomedon, 2022-08-26:
### Describe the bug

Hallo,

ich habe gestern die Batterie für mein PV System gekommen. Nun versuche ich sie in EVCC einzubinden. Das Auslesen des SOC funktioniert leider nicht. Ich habe eine RCT Power Anlage

Als Fallback habe ich auch versucht die SOC über meinen ioBroker zu beziehen. Dies läuft auch auf eine Fehlermeldung. 

Viele Grüße

Carsten

### Steps to reproduce

Connect EVCC to RCT Power battery

### Configuration details

- type: template
  template: rct-power
  usage: battery
  host: 192.168.1.56
  name: battery3

**fallback:**

- name: battery3
  type: custom
  power:
    source: http
    method: GET
    uri: http://192.168.1.96:8087/rct.0.g_sync.p_acc_lp
  soc:
    source: http
    method: GET
    uri: http://192.168.1.96:8087/rct.0.battery.soc

### Log details

**via template**

Aug 26 13:16:29 raspberrypi evcc[4004502]: [site  ] DEBUG 2022/08/26 13:16:29 battery power: 0W
Aug 26 13:16:29 raspberrypi evcc[4004502]: [site  ] DEBUG 2022/08/26 13:16:29 grid power: -8183W
Aug 26 13:16:29 raspberrypi evcc[4004502]: [site  ] ERROR 2022/08/26 13:16:29 battery soc 0: Invalid response Cmd Response (05) Id Battery power [W] (400F015B) Data [193 130 87 64]
Aug 26 13:16:29 raspberrypi evcc[4004502]: [site  ] DEBUG 2022/08/26 13:16:29 site power: -7183W
Aug 26 13:16:29 raspberrypi evcc[4004502]: [lp-1  ] DEBUG 2022/08/26 13:16:29 charge currents: [0.03 0.03 0.07]A
Aug 26 13:16:29 raspberrypi evcc[4004502]: [lp-1  ] DEBUG 2022/08/26 13:16:29 charger status: E

**Fallback Lösung**

Error message im UI:
battery soc 0: strconv.ParseFloat: parsing \"{\\n \\\"error\\\": \\\"command rct.0.battery.soc unknown\\\",\\n \\\"getPlainValue\\\": \\\"http://192.168.1.96:8087/getPlainValue/stateID\\\",\\n \\\"get\\\": \\\"http://192.168.1.96:8087/get/stateID/?prettyPrint\\\",\\n \\\"getBulk\\\": \\\"http://192.168.1.96:8087/getBulk/stateID1,stateID2/?prettyPrint\\\",\\n \\\"set\\\": \\\"http://192.168.1.96:8087/set/stateID?value=1&prettyPrint\\\",\\n \\\"toggle\\\": \\\"http://192.168.1.96:8087/toggle/stateID&prettyPrint\\\",\\n \\\"setBulk\\\": \\\"http://192.168.1.96:8087/setBulk?stateID1=0.7&stateID2=0&prettyPrint\\\",\\n \\\"setValueFromBody\\\": \\\"http://192.168.1.96:8087/setValueFromBody?stateID1\\\",\\n \\\"objects\\\": \\\"http://192.168.1.96:8087/objects?pattern=system.adapter.admin.0*&prettyPrint\\\",\\n \\\"states\\\": \\\"http://192.168.1.96:8087/states?pattern=system.adapter.admin.0*&prettyPrint\\\",\\n \\\"search\\\": \\\"http://192.168.1.96:8087/search?pattern=system.adapter.admin.0*&prettyPrint\\\",\\n \\\"query\\\": \\\"http://192.168.1.96:8087/query/stateID1,stateID2/?dateFrom=2019-06-06T12:00:00.000Z&dateTo=2019-06-06T12:00:00.000Z&noHistory=false&aggregate=minmax&count=3000&prettyPrint\\\"\\n}\": invalid syntax

Konsole:
Aug 26 12:46:36 raspberrypi evcc[4002704]: [site  ] ERROR 2022/08/26 12:46:36 battery soc 0: strconv.ParseFloat: parsing "{\n  \"error\": \"command rct.0.battery.soc unknown\",\n  \"getPlainValue\": \"http://192.168.1.96:8087/getPlainValue/stateID\",\n  \"get\": \"http://192.168.1.96:8087/get/stateID/?prettyPrint\",\n  \"getBulk\": \"http://192.168.1.96:8087/getBulk/stateID1,stateID2/?prettyPrint\",\n  \"set\": \"http://192.168.1.96:8087/set/stateID?value=1&prettyPrint\",\n  \"toggle\": \"http://192.168.1.96:8087/toggle/stateID&prettyPrint\",\n  \"setBulk\": \"http://192.168.1.96:8087/setBulk?stateID1=0.7&stateID2=0&prettyPrint\",\n  \"setValueFromBody\": \"http://192.168.1.96:8087/setValueFromBody?stateID1\",\n  \"objects\": \"http://192.168.1.96:8087/objects?pattern=system.adapter.admin.0*&prettyPrint\",\n  \"states\": \"http://192.168.1.96:8087/states?pattern=system.adapter.admin.0*&prettyPrint\",\n  \"search\": \"http://192.168.1.96:8087/search?pattern=system.adapter.admin.0*&prettyPrint\",\n  \"query\": \"http://192.168.1.96:8087/query/stateID1,stateID2/?dateFrom=2019-06-06T12:00:00.000Z&dateTo=2019-06-06T12:00:00.000Z&noHistory=false&aggregate=minmax&count=3000&prettyPrint\"\n}": invalid syntax

### What type of operating system are you running?

Linux

### Version

0.100

diomedon, 2022-08-26:
Hallo,
Ich muss die Batterie Werte im iobroker erst einmal durchlauferhitzen und in neue Objekte schreiben. Dann geht das Auslesen per http. 

Ich denke EVCC hat beim nativen RCT Auslesen ein ähnliches Problem. Ich teste auch nochmal die direkte RCT Gridmeter und Wechselrichter Anbindung. Dort setze ich auch nicht mehr auf die native Anbindung. RCT liefert zu oft Müll zurück, so dass ich ständig ein rotes Warndreieck auf der EVCC Oberfläche habe. 

Viele Grüße

Carsten

premultiply, 2022-08-26:
Es kann und darf nur ein einziges System auf das RCT-System zugreifen.
Also nicht mit Hausautomation und evcc parallel.

andig, 2022-08-26:
Die Fehlermeldung sieht tatsächlich aus als würde die Antwort nicht zum Request passen. Mehrere Abfragen?! Ich konvertiere das hier mal in eine Diskussion, ein Fehler scheint es nicht zu sein.