evcc-io/evcc #4244: Add support for Shelly EM (2 channels) meter

efrencd, 2022-08-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2022-08-27:
/cc @thierolm

VolkerK62, 2022-08-27:
@efrencd  I use this also. You know, that you can use the http-plugin for integration?

efrencd, 2022-08-27:
I didn't know. I'll try. 

Anyway a direct integration with a template would be welcomed.

Thanks!

thierolm, 2022-08-30:
Hi @efrencd ,
have U tested the shelly charger or shelly-pm1 meter template?
U are able to define the channel in order to select P1 (channel = 0) or P2 (channel = 1) meter ...

VolkerK62, 2022-08-30:
Hi @thierolm ,
I have Shelly EM in use.
I tried shelly-1pm template. It doesn´t work.
```
- name: WP_power # Verbrauch (charger) Wärmepumpe
  type: template
  template: shelly-1pm
  host: 192.168.178.21
  channel: 1
```
result `evcc meter`
```
WP_power
--------
Power: invalid channel, missing power meter
```

VolkerK62, 2022-08-30:
So funktioniert es (custom)
```
- name: WP_power # Verbrauch (charger) Wärmepumpe
  type: custom
  power:         # Shelly EM Stromzange 2
    source: http
    uri: http://192.168.178.21/emeter/1
    jq: .power  
```

thierolm, 2022-08-31:
Hi @VolkerK62 , could U pls provide me a trace log: `evcc meter -l trace`

VolkerK62, 2022-08-31:
which version: template or custom?

VolkerK62, 2022-08-31:
Ahh, meter vs emeter??
`"num_meters":0,"num_emeters":2`

thierolm, 2022-08-31:
> Ahh, meter vs emeter?? `"num_meters":0,"num_emeters":2`

Jepp, da haben die Shelly Entwickler mal wieder nix von JSON Standardisierung gehalten ... :-(

@andig:
Ich schau mal, dass ich am WE einen entsprechenden Erweiterungs-PR aufmache.

thierolm, 2022-08-31:
@VolkerK62 + @efrencd 
Could U pls check test the PR? I do not have an Shelly EM device ...

VolkerK62, 2022-08-31:
Sorry, da hab ich absolut keine Ahnung, wie ich das machen könnte .... das hatten wir vor einiger Zeit schon mal ;-)
Wenn andig es ins nightly nimmt, kann ich es gleich morgen früh testen.

andig, 2022-09-02:
Ist drin

closed 2022-09-02 by andig

VolkerK62, 2022-09-02:
werd ich morgen mal testen.
Aber vorher noch ne blöde Frage: nehm ich das `Shelly-1pm` Template oder hat es ein eigenes bekommen?

thierolm, 2022-09-02:
nimm das Shelly-1pm template

VolkerK62, 2022-09-02:
@thierolm 
ist `usage: pv` optional oder zwingend?
Ich nutze den Shelly EM als `battery` und als `charge` meter.
Aktuell als `custom` ohne `usage`.

thierolm, 2022-09-02:
Im Meter Template ist pv aktuell die einzige Option und muss so gesetzt werden. Du müsstest das so definierte Meter aber auch als battery meter in der evcc.yaml definieren können.
Wenn du den Shelly als Charger nutzen willst, musst du das `shelly` Charger Template nehmen.

VolkerK62, 2022-09-03:
Top job.
Funktioniert prima.
Ich bin immer wieder erstaunt, was und wie schnell alles möglich ist ;-)

thierolm, 2022-09-03:
Thx fürs Testen! Schönes WE!

VolkerK62, 2025-05-20:
Du bist hier in einem fast 3 Jahre alten, geschlossenen Issue.
Mach mit deiner Frage bitte eine neue Diskussion auf, da ist die Antwortwahrscheinlichkeit bedeutend größer.

PR #4273 (2022-09-02): Support for Shelly EM (2 channels) meter
Fix #4244