evcc-io/evcc #4256: Add "script" message type

andig, 2022-08-29:
Hi,

Bei den Plugins für `meter`, `charger `und `vehicles `gibt es "Shell Script"
https://docs.evcc.io/docs/reference/plugins#shell-script-lesenschreiben
um beliebige Anbindungen zu ermöglichen.

Es wäre schön, wenn man beim `messaging `type 
https://docs.evcc.io/docs/reference/configuration/messaging#unterst%C3%BCtzte-dienste
auch script nutzen könnte.
So könnte man ebenfalls beliebige Dienste nutzen.
(Ich nutze sonst Signal über https://github.com/AsamK/signal-cli/)

VG
Sirko

_Originally posted by @SirkoVZ in https://github.com/evcc-io/evcc/discussions/4247_

SirkoVZ, 2022-09-07:
Hi,

ich hab schon mal angefangen. Funktioniert auch soweit. Allerdings sind meine Go-Fähigkeiten begrenzt, das meiste ist copy&paste vom Script Plugin.
Aber vielleicht hilft es
https://github.com/SirkoVZ/evcc/compare/master...SirkoVZ:evcc:ScriptMessenger?expand=1

andig, 2022-09-07:
Sieht erstmal plausibel aus. Funktionierts denn? Dann gerne PR!

SirkoVZ, 2022-09-09:
PR ist da.

closed 2022-09-10 by andig