evcc-io/evcc #4259: evcc restart nach Fahrzeugwechsel im UI

VolkerK62, 2022-08-29:
### Describe the bug

Ich habe aus Versehen im UI ein Fahrzeug ausgewählt. 
Habe dann direkt wieder Gastfahrzeug (= kein Fahrzeug) ausgewählt. 
Dann bekam ich kurzzeitig die Anzeige "keine Verbindung zu Server". 
Später habe ich dann in der Ersparnisanzeige gesehen, dass es einen Server Neustart gegeben hat.

### Steps to reproduce

Im UI ein Fahrzeug auswählen.
Innerhalb von ein paar Sekunden wieder zurück auf Gastfahrzeug wechseln.

### Configuration details

Es ist kein Defaultfahrzeug definiert

### Log details

```
Aug 29 14:18:55 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:18:55 ----
Aug 29 14:18:55 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:18:55 charge power: 0W
Aug 29 14:18:55 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:18:55 pv power: 4998W
Aug 29 14:18:55 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:18:55 battery power: -928W
Aug 29 14:18:55 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:18:55 grid power: -2794W
Aug 29 14:18:55 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:18:55 battery soc 0: 84%
Aug 29 14:18:55 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:18:55 giving priority to battery charging at soc: 84%
Aug 29 14:18:55 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:18:55 site power: -2694W
Aug 29 14:19:02 raspberrypi evcc[30310]: [lp-1  ] INFO 2022/08/29 14:19:02 vehicle updated: unknown -> (T)Raumschiff
Aug 29 14:19:02 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:02 set charge mode: pv
Aug 29 14:19:02 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:02 pv timer elapse
Aug 29 14:19:02 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:02 pv timer inactive
Aug 29 14:19:02 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:02 set target soc: 90
Aug 29 14:19:02 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:19:02 ----
Aug 29 14:19:02 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:02 charge power: 0W
Aug 29 14:19:03 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:19:03 pv power: 4987W
Aug 29 14:19:03 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:19:03 battery power: -903W
Aug 29 14:19:03 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:19:03 grid power: -2779W
Aug 29 14:19:03 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:19:03 battery soc 0: 84%
Aug 29 14:19:03 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:19:03 giving priority to battery charging at soc: 84%
Aug 29 14:19:03 raspberrypi evcc[30310]: [site  ] DEBUG 2022/08/29 14:19:03 site power: -2679W
Aug 29 14:19:03 raspberrypi evcc[30310]: [hyundai] TRACE 2022/08/29 14:19:03 retry stamps download, last attempt: 2022-08-27 16:37:07.546603656 +0200 CEST m=+0.410934734
Aug 29 14:19:03 raspberrypi evcc[30310]: [hyundai] TRACE 2022/08/29 14:19:03 POST https://prd.eu-ccapi.hyundai.com:8080/api/v1/user/oauth2/token
Aug 29 14:19:04 raspberrypi evcc[30310]: [hyundai] TRACE 2022/08/29 14:19:04 grant_type=refresh_token&redirect_uri=https%3A%2F%2Fwww.getpostman.com%2Foauth2%2Fcallback&refresh_token=TD4E9ZL4VHU0KWPNKJIGHW
Aug 29 14:19:04 raspberrypi evcc[30310]: --
Aug 29 14:19:04 raspberrypi evcc[30310]: {"access_token":"eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCJ9.eyJwaWQiOiI1YmIyYzcyMDlhOTk0MzA0MTk0MzZlYzQiLCJ1aWQiOiI5MjExYTM0YS1lMjhmLTRkMjktOWFlNi03ZWM3NDZjNTc2MGQiLCJzaWQiOiI2ZDQ3N2MzOC0zY2E0LTRjZjMtOTU1Ny0yYTE5MjlhOTQ2NTQiLCJleHAiOjE2NjE4NjE5NDQsImlhdCI6MTY2MTc3NTU0NCwiaXNzIjoiYmx1ZWxpbmsifQ.e9i6Ozg1EODYeBEJt4P1mGsNIP4EBQdPC3p4qref4AQA6jC1WdnN9s4irwELt6oR9vj_ie-wikU1jYTPxLCSQuzaCMRNhfnP6ZUZXHqFeGMax1IF-6lv1IxPOXcE0Y-LkiCVfd9B9YW6x9sAfiDMkS7EKyj-9jxzYIhZ0T7E6Dzu6a63M-5Ov7shkfOsaKuJWETRPKvv4gQ75ivpmihL2vX-1HZ-L-uV3sxqqr1q5nDA4EW3jzp-RtyENjuRDx_jjJBBkS8oA3Cl8dovct2gO1p3l7VgErpIF1BrPsjZEGd0deT7D7fTwId-O3TvyoM6ofvvZZQOi1mnrNlWNp6b6Q","token_type":"Bearer","expires_in":86400}
Aug 29 14:19:04 raspberrypi evcc[30310]: [hyundai] TRACE 2022/08/29 14:19:04 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Aug 29 14:19:04 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:04 set charge mode: off
Aug 29 14:19:04 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:04 pv timer elapse
Aug 29 14:19:04 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:04 pv timer inactive
Aug 29 14:19:04 raspberrypi evcc[30310]: [hyundai] TRACE 2022/08/29 14:19:04 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":49.309922,"lon":8.539017,"alt":0,"type":0},"head":333,"speed":{"value":0,"unit":0},"accuracy":{"hdop":0,"pdop":0},"time":"20220827102029"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":72,"batteryPlugin":0,"remainTime2":{"etc1":{"value":4,"unit":1},"etc2":{"value":510,"unit":1},"etc3":{"value":120,"unit":1},"atc":{"value":510,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":353,"unit":1},"totalAvailableRange":{"value":353,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"0000","timeSection":0}},"end":{"day":0,"time":{"time":"0000","timeSection":0}}},"targetSOClist":[{"targetSOClevel":100,"plugType":1},{"targetSOClevel":80,"plugType":0}]},"chargePortDoorOpenStatus":2,"batteryPreconditioning":false},"ign3":true,"hoodOpen":false,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":76,"batState":0,"sjbDeliveryMode":0,"batSignalReferenceValue":{"batWarning":0},"powerAutoCutMode":2},"lampWireStatus":{"stopLamp":{"stopLampStatus":false,"leftLamp":false,"rightLamp":false},"headLamp":{"headLampStatus":false,"leftLowLamp":false,"rightLowLamp":false,"leftHighLamp":false,"rightHighLamp":false,"leftBifuncLamp":false,"rightBifuncLamp":false},"turnSignalLamp":{"turnSignalLampStatus":false,"leftFrontLamp":false,"rightFrontLamp":false,"leftRearLamp":false,"rightRearLamp":false}},"smartKeyBatteryWarning":false,"washerFluidStatus":false,"breakOilStatus":false,"sleepModeCheck":false,"time":"20220827174321","remoteWaitingTimeAlert":{"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"07:26:01"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"odometer":{"value":9897.4,"unit":1}}},"msgId":"5d6626d0-8c42-4066-afd1-19b95278b5bc"}
Aug 29 14:19:04 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:04 vehicle odometer: 9897km
Aug 29 14:19:04 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:04 charge currents: [0 0 0]A
Aug 29 14:19:04 raspberrypi evcc[30310]: [lp-1  ] DEBUG 2022/08/29 14:19:04 charger status: A
Aug 29 14:19:04 raspberrypi evcc[30310]: [hyundai] TRACE 2022/08/29 14:19:04 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Aug 29 14:19:04 raspberrypi evcc[30310]: [hyundai] TRACE 2022/08/29 14:19:04 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":49.309922,"lon":8.539017,"alt":0,"type":0},"head":333,"speed":{"value":0,"unit":0},"accuracy":{"hdop":0,"pdop":0},"time":"20220827102029"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":72,"batteryPlugin":0,"remainTime2":{"etc1":{"value":4,"unit":1},"etc2":{"value":510,"unit":1},"etc3":{"value":120,"unit":1},"atc":{"value":510,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{" [... cut ...]
Aug 29 14:19:28 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Aug 29 14:19:28 raspberrypi systemd[1]: evcc.service: Consumed 7min 28.942s CPU time.
Aug 29 14:19:38 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart counter is at 1.
Aug 29 14:19:38 raspberrypi systemd[1]: Stopped evcc.
Aug 29 14:19:38 raspberrypi systemd[1]: evcc.service: Consumed 7min 28.942s CPU time.
Aug 29 14:19:38 raspberrypi systemd[1]: Started evcc.
```

### What type of operating system are you running?

Linux

### Version

v0.100 (0b87def2)

andig, 2022-08-30:
Closed in https://github.com/evcc-io/evcc/commit/90b55dbdb82658958b21772ba91eb7c1486eb9ed. `socEstimator` is not guarded by mutex, hence might expire during charger or vehicle calls.

closed 2022-08-30 by andig

VolkerK62, 2022-08-31:
@andig Super. Klappt wieder. Danke