evcc-io/evcc #4276: Make Target Charging recurring

stefaanbolle, 2022-09-01:
The Target charging feature is really a nice feature.
What would be very useful to me and hopefully others is to allow a recurrent schedule. For example on weekdays be charged at 7AM when plugged in.
Now it only allows a Target charging at a specific day and time.

naltatis, 2022-09-01:
Thanks for the request. We thought about implementing this but started with the simple "one target time" approach. 

Would be interesting to get more feedback here from evcc users that might use this "schedule"-feature. Another question is how sophisticated/complex such a configuration should be and to hear what the use cases are. I guess "driving to work at the same time on a set number of weekdays" is the most common one, but maybe there are other cases as well.

stefaanbolle, 2022-09-01:
> Thanks for the request. We thought about implementing this but started with the simple "one target time" approach.
>
> Would be interesting to get more feedback here from evcc users that might use this [...]

Indeed going to work would be a main use case. I see it as a clickable week (Mon-Sun) where you click/highlight the days when target charging is needed and then select the target time as we can select now.

andig, 2022-09-10:
This is perfectly  doable using home automation, therefore wontfix in evcc for time being.

closed 2022-09-10 by andig