evcc-io/evcc #4277: aktuelles nightly: cannot create meter 'grid': cannot create meter 'modbus': invalid measurement for power: POWER

Caibuk, 2022-09-01:
### Describe the bug

Ich habe heute meinen nightly-docker-Container geupdatet. Es kann nun - bei unveränderter Konfig - kein grid meter (KSEM) mehr erstellt werden (siehe Betreff).

Weder meine seit (zwei?) Jahren bewährte noch die in der Doku vermerkte auf einem template basierende Konfig funktionieren.

Ich habe zur Fehlereingrenzung den aktuellen Repository-Stand auf einem WIndows-PC kompiliert und evcc mit meiner (docker-) Konfig ausgeführt: derselbe Fehler.

Mein letzter funktionierender Stand war das nightly vom 30.08.2022.

### Steps to reproduce

Start des aktuellen nightlys `evcc 0.100 (d9802b73)` mit meiner bisher problemlos funktionieren Konfig.

### Configuration details

Da die Fehlermeldung den konkreten Fehlerort angibt, kopiere ich hier nur den entsprechenden Teil der Konfig ein. Bei Bedarf kann die komplette Konfig nachgereicht werden.

### alte (bisher funktionierende) Konfig # 1:
```
meters:
  - name: grid
    type: modbus
    model: kostal
    uri: 10.0.1.57:502
    id: 255
    power: Power
    energy: Export
```
### alternativ getestete Konfig gemäß Doku # 2 (funktioniert auch nicht):
```
meters:
  - name: grid
    type: template
    template: kostal-ksem
    usage: grid
    modbus: tcpip
    id: 255
    host: 10.0.1.57
    port: 502
```

### Log details

###Konfig # 1:

```
[main  ] INFO 2022/09/01 09:20:38 evcc 0.100 (d9802b73)
[main  ] INFO 2022/09/01 09:20:38 using config file /etc/evcc.yaml
[main  ] INFO 2022/09/01 09:20:38 listening at :7070
[mqtt  ] INFO 2022/09/01 09:20:38 connecting evcc-1612669389 at tcp://10.0.1.97:1883
[mqtt  ] DEBUG 2022/09/01 09:20:38 tcp://10.0.1.97:1883 connected
[main  ] FATAL 2022/09/01 09:20:38 cannot create meter 'grid': cannot create meter 'modbus': invalid measurement for power: POWER
```

###Konfig # 2
```
[main  ] INFO 2022/09/01 09:30:44 evcc 0.100 (d9802b73)
[main  ] INFO 2022/09/01 09:30:44 using config file /etc/evcc.yaml
[main  ] INFO 2022/09/01 09:30:44 listening at :7070
[mqtt  ] INFO 2022/09/01 09:30:44 connecting evcc-1504953181 at tcp://10.0.1.97:1883
[mqtt  ] DEBUG 2022/09/01 09:30:44 tcp://10.0.1.97:1883 connected
[main  ] FATAL 2022/09/01 09:30:44 cannot create meter 'grid': cannot create meter 'template': cannot create meter 'modbus': invalid measurement for power: POWER
```

### What type of operating system are you running?

Docker container

### Version

`evcc 0.100 (d9802b73)`

Caibuk, 2022-09-01:
Die Windows-Variante des 0.100-Releases (https://github.com/evcc-io/evcc/releases/tag/0.100) funktioniert mit beiden Konfigs.

premultiply, 2022-09-01:
Probier mal bitte folgendes:
meters:
```yaml
meters:
- name: grid
  type: modbus
  model: sunspec
  uri: 10.0.1.57:502
  id: 255
  power: Power
  energy: Import
  currents:
  - CurrentL1
  - CurrentL2
  - CurrentL3
```

Caibuk, 2022-09-01:
Hallo @premultiply,

vielen Dank fürs Übernehmen!

Das
```yaml
meters:
  - name: grid
    type: modbus
    model: sunspec
    uri: 10.0.1.57:502
    id: 255
    power: Power
    energy: Import
```

führt zu
```
[main  ] INFO 2022/09/01 12:14:36 evcc 0.100 (d9802b73)
[main  ] INFO 2022/09/01 12:14:36 using config file /etc/evcc.yaml
[main  ] INFO 2022/09/01 12:14:36 listening at :7070
[mqtt  ] INFO 2022/09/01 12:14:36 connecting evcc-926786305 at tcp://10.0.1.97:1883
[mqtt  ] DEBUG 2022/09/01 12:14:36 tcp://10.0.1.97:1883 connected
[main  ] FATAL 2022/09/01 12:14:36 cannot create meter 'grid': cannot create meter 'modbus': invalid measurement for power: POWER
```
Leider keine Änderung. Er hat scheint ja auch ein Problem mit `power` und nicht mit `energy` zu haben ...

premultiply, 2022-09-01:
Es geht ums model...

Caibuk, 2022-09-01:
> Es geht ums model...

Klar, sorry für meinen unqualifizierten Einwurf.

Mit der `release`-Version klappt es auch mit Deinem Vorschlag (habe nur Ex- statt Import genommen):
```yaml
  - name: grid
    type: modbus
    model: sunspec
    uri: 10.0.1.57:502
    id: 255
    power: Power
    energy: Export
```

premultiply, 2022-09-01:
Könntest du bitte auch mal vollständig
```yaml
meters:
- name: grid
  type: modbus
  model: sunspec
  uri: 10.0.1.57:502
  id: 255
  power: Power
  energy: Import
  currents:
  - CurrentL1
  - CurrentL2
  - CurrentL3
```
testen und die `evcc meter` Ausgabe davon hierher kopieren.

Dann könnte ich das Template anpassen.

Caibuk, 2022-09-01:
Ergebnis mit Deinem vollständigen Vorschlag (aktuelle docker-nightly):
```
[main  ] INFO 2022/09/01 13:30:52 evcc 0.100 (d9802b73)
[main  ] INFO 2022/09/01 13:30:52 using config file /etc/evcc.yaml
[main  ] INFO 2022/09/01 13:30:52 listening at :7070
[mqtt  ] INFO 2022/09/01 13:30:52 connecting evcc-1281498333 at tcp://10.0.1.97:1883
[mqtt  ] DEBUG 2022/09/01 13:30:52 tcp://10.0.1.97:1883 connected
[main  ] FATAL 2022/09/01 13:30:52 cannot create meter 'grid': cannot create meter 'modbus': invalid measurement for power: POWER
```

`evcc meter` und Deinem Konfigvorschlag (mit der Windows-nightly):
```
PS C:\Software\evcc-master> .\evcc.exe meter
[main  ] INFO 2022/09/01 13:32:16 evcc 0.0.1-alpha
[main  ] INFO 2022/09/01 13:32:16 using config file C:\Software\evcc-master\evcc.yaml
[mqtt  ] INFO 2022/09/01 13:32:16 connecting evcc-242684909 at tcp://10.0.1.97:1883
[mqtt  ] DEBUG 2022/09/01 13:32:16 tcp://10.0.1.97:1883 connected
[main  ] FATAL 2022/09/01 13:32:16 cannot create meter 'grid': cannot create meter 'modbus': invalid measurement for power: POWER
```
 `evcc meter` und Deinem Konfigvorschlag (mit der Windows-Release-Version):
```
PS C:\Software\evcc-master> .\evcc_rel.exe meter
[main  ] INFO 2022/09/01 13:33:13 evcc 0.100
[main  ] INFO 2022/09/01 13:33:13 using config file C:\Software\evcc-master\evcc.yaml
[mqtt  ] INFO 2022/09/01 13:33:13 connecting evcc-1127118774 at tcp://10.0.1.97:1883
[mqtt  ] DEBUG 2022/09/01 13:33:13 tcp://10.0.1.97:1883 connected
grid
----
Power:          -6260W
Energy:         14444914.0kWh
Current L1..L3: 8.54A 8.8A 9.01A

pv-dc-kostal
------------
Power: 6997W

battery-pv
----------
Power: 6W
SoC:   100%

lp1
---
[mqtt  ] DEBUG 2022/09/01 13:33:15 mbmd/sdm1-20/Power wait for initial value
[mqtt  ] DEBUG 2022/09/01 13:33:17 mbmd/sdm1-20/Import wait for initial value
[mqtt  ] DEBUG 2022/09/01 13:33:17 mbmd/sdm1-20/Current/L2 wait for initial value
Power:          0W
Energy:         2542.2kWh
Current L1..L3: 0A 0A 0A

lp2
---
[mqtt  ] DEBUG 2022/09/01 13:33:27 mbmd/sdm2301-10/Import wait for initial value
Power:  0W
Energy: 9132.4kWh
```

premultiply, 2022-09-01:
Ist Energy um den Faktor 1000 zu groß?

Caibuk, 2022-09-01:
scheint so:
[image]

andig, 2022-09-01:
`POWER` sollte im Master behoben sein, sorry. Falls es hier ein zweites (anderes) Problem gibt müssen wir wieder auf machen.

premultiply, 2022-09-01:
Ja, die Einheit der Energie stimmt auch nicht.

andig, 2022-09-01:
Das Template hat gar kein Energy, kann also auch kein Bug sein ;) Allerdings haben wir bei Sunspec eigentlich kein Einheitenproblem- und wenn wäre es eher eins in mbmd.

Caibuk, 2022-09-02:
> `POWER` sollte im Master behoben sein, sorry. Falls es hier ein zweites (anderes) Problem gibt müssen wir wieder auf machen.

Mit dem aktuellen `nightly`-Build funktioniert mein Gridmeter wieder:
```
[main  ] INFO 2022/09/02 07:29:30 evcc 0.100 (a2132897)
[site  ] INFO 2022/09/02 07:29:38   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/09/02 07:29:38     grid:      power ✓ energy ✓ currents ✓
```
Vielen Dank für den schnellen Bugfix!

andig, 2022-09-03:
Der bug ist erstmal behoben, PR für Energy ist ein anderes Thema und braucht Unterstützung aus mbmd.

PR #4281 (2022-09-08): KSEM: Fix template readings
Fixes #4277
closed 2022-09-03 by andig