evcc-io/evcc #4288: Q.HOME Integration

masterdot, 2022-09-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

I do have a q.home qcells pv installed. it seems to be solarx compatible, but has some differences. 
I did post the api call and the json reply. 
If anything is needed, just ask. Ill do my best to answer it. 

API Url
https://qhome-ess-g3.q-cells.eu/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=20240812316545356441670812&sn=SVNAR2Z9LE

JSON Response
{"success":true,"exception":"Query success!","result":{"inverterSN":"J36C28I2893433","sn":"SVNAR2Z9LE","acpower":300.0,"yieldtoday":11.2,"yieldtotal":310.3,"feedinpower":0.0,"feedinenergy":1.96,"consumeenergy":98.59,"feedinpowerM2":0.0,"soc":82.0,"peps1":0.0,"peps2":0.0,"peps3":0.0,"inverterType":"14","inverterStatus":"102","uploadTime":"2022-09-01 23:08:17","batPower":-253.0,"powerdc1":0.0,"powerdc2":0.0,"powerdc3":null,"powerdc4":null,"batStatus":"0"}}

masterdot, 2022-09-01:
Error Messages in the Web GUI
vor 2 Sekunden
grid meter: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 2 Sekunden
battery meter 0: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 3 Sekunden
pv meter 0: add[0]: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 12 Sekunden
grid meter: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 12 Sekunden
battery meter 0: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 13 Sekunden
pv meter 0: add[0]: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 22 Sekunden
grid meter: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 22 Sekunden
battery meter 0: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 23 Sekunden
pv meter 0: add[0]: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 32 Sekunden
grid meter: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 32 Sekunden
battery meter 0: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 33 Sekunden
pv meter 0: add[0]: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 42 Sekunden
grid meter: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 42 Sekunden
battery meter 0: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 43 Sekunden
pv meter 0: add[0]: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 52 Sekunden
grid meter: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 52 Sekunden
battery meter 0: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 53 Sekunden
pv meter 0: add[0]: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 1 Minute
grid meter: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 1 Minute
battery meter 0: strconv.ParseFloat: parsing \"<nil>\": invalid syntax
vor 1 Minute
pv meter 0: add[0]: strconv.ParseFloat: parsing \"<nil>\": invalid syntax

andig, 2022-09-02:
How do you get the error messages? What config did you try?