evcc-io/evcc #4296: Add SENEC.Wallbox pro s

andig, 2022-09-03:
Refs https://github.com/evcc-io/evcc/discussions/4287

premultiply, 2022-09-03:
Es sind wohl inzwischen zu viele Geräte. Wir scheinen den Überblick verloren zu haben. :grinning: 
Die Box ist schon längst artig im Portfolio eingepflegt:
https://github.com/evcc-io/evcc/blob/f388eeeebd7a11f382dddef691e3bfd9cf7c389f/templates/definition/charger/heidelberg.yaml#L6-L8
https://docs.evcc.io/docs/devices/chargers#wallbox-pro-s--

closed 2022-09-03 by premultiply

andig, 2022-09-03:
Wtf 😂

bastian1607, 2023-06-21:
Hallo zusammen,

ich würde das Thema doch gerne wieder aufgreifen. Wäre es möglich zusätzlich eine Cloud-Anbindung zu implementieren? 
Hintergrund: Bei einigen ist die Wallbox direkt an dem Speicher angebunden. 

Die API ist hier beschrieben: 
https://documenter.getpostman.com/view/10329335/UVCB9ihW#25cbd682-7ea2-41e9-ad86-df56697d8736

Über die API habe ich meine Wallbox steuern können über die Seite: 
https://reqbin.com/curl 

Gerne stelle ich meine Wallbox zur Verfügung zum Testen. 

Hier wurde der Weg über einen manuellen Charger mit dem http-Plugin angeregt, aber ich bin zu unwissend um das umzusetzen: 
https://github.com/evcc-io/evcc/discussions/6242

andig, 2023-06-21:
> Bei einigen ist die Wallbox direkt an dem Speicher angebunden.

Ja, und?

premultiply, 2023-06-21:
Kabel ab, Problem gelöst.
Die ABL-Box wird von evcc direkt unterstützt.
Es macht keinen Sinn da noch ausgerechnet ein Senec-System zwischenzuschalten.

bastian1607, 2023-06-21:
> > Bei einigen ist die Wallbox direkt an dem Speicher angebunden.
> 
> Ja, und?

Die Wallboxen kann man nicht mit der rs485 erreichen ohne eine ggf bauliche, aber in jedem Fall Änderung der Schaltung bei der Wallbox. Das heißt, die Wallbox müsste aus dem Speicher genommen werden und die Schaltung innerhalb der Wallbox geändert werden. 

So verstehe ich es auch in https://docs.evcc.io/docs/devices/chargers/#wallbox-pro-s- . Nur ein "Kabel raus" würde nicht reichen.

premultiply, 2023-06-21:
Es muss keine "Schaltung" geändert werden.
Kabel am Senec ab und mit RS485-USB-Dongle an das evcc-System dran, fertig.
So wie bei vielen anderen auch schon demonstriert.

andig, 2023-06-21:
Dann schließen wir hier mal da schon arg OT. Wenn noch was fehlt gerne Discussion öffnen.