evcc-io/evcc #4298: EVCC ignoriert prioritySoC

Stargate85, 2022-09-04:
### Describe the bug

Obwohl ich prioritySoC auf 10 gesetzt habe wird bei einem Batteriestand von über 10% trotzdem die Batterie priorisiert und nur mit 6A geladen. Dabei ist es egal ob ich PV oder Min+PV als Modus nutze.

### Steps to reproduce

Laden bei einem Batteriestand von mehr als 10% im Modus PV oder Min+PV

### Configuration details

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: ###

meters:
- type: template
  template: e3dc
  usage: grid
  host: 192.168.1.8
  port: 502
  name: grid1
- type: template
  template: e3dc
  usage: pv
  host: 192.168.1.8
  port: 502
  name: pv2
- type: template
  template: e3dc
  usage: battery
  host: 192.168.1.8
  port: 502
  name: battery3

chargers:
- type: template
  template: easee
  user: ###
  password: ###
  charger: ###
  name: wallbox5

vehicles:
- type: template
  template: tesla
  title: Tesla Model 3
  accessToken: ##
  refreshToken: ##
  capacity: 77
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  vehicles:
  - ev4
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 20
  resetOnDisconnect: false
  soc:
    poll:
      mode: charging
      interval: 60m
    min: 0
    target: 90
    estimate: false

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  residualPower: -100
  prioritySoC: 10
  bufferSoC: 98

### Log details

Sep 04 09:00:49 nuc-evcc evcc[434]: [main  ] INFO 2022/09/04 09:00:49 evcc 0.100
Sep 04 09:00:49 nuc-evcc evcc[434]: [main  ] INFO 2022/09/04 09:00:49 using config file /etc/evcc.yaml
Sep 04 09:00:49 nuc-evcc evcc[434]: [main  ] INFO 2022/09/04 09:00:49 listening at :7070
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] WARN 2022/09/04 09:00:54 locking phase config to 1p for switchable charger
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] INFO 2022/09/04 09:00:54 site config:
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] INFO 2022/09/04 09:00:54   meters:      grid ✓ pv ✓ battery ✓
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] INFO 2022/09/04 09:00:54     grid:      power ✓ energy ✗ currents ✗
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] INFO 2022/09/04 09:00:54     pv 1:      power ✓ energy ✗ currents ✗
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] INFO 2022/09/04 09:00:54     battery 1: power ✓ energy ✗ currents ✗ soc ✓
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] INFO 2022/09/04 09:00:54 loadpoint 1:
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] INFO 2022/09/04 09:00:54   mode:        pv
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] INFO 2022/09/04 09:00:54   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] INFO 2022/09/04 09:00:54   meters:      charge ✓
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] INFO 2022/09/04 09:00:54     charge:    power ✓ energy ✗ currents ✓
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] INFO 2022/09/04 09:00:54   vehicles:    ✓
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] INFO 2022/09/04 09:00:54     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:00:54 ----
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:00:54 charge power: 1457W
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:00:54 pv power: 4694W
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:00:54 battery power: -2565W
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:00:54 grid power: -4W
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:00:54 battery soc 0: 56%
Sep 04 09:00:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:00:54 site power: -2669W
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:00:54 charge currents: [6.09 0.011 0.009]A
Sep 04 09:00:54 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:00:54 charger status: A
Sep 04 09:01:04 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:04 ----
Sep 04 09:01:04 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:04 charge power: 1457W
Sep 04 09:01:04 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:04 pv power: 4713W
Sep 04 09:01:04 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:04 battery power: -2607W
Sep 04 09:01:04 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:04 grid power: 8W
Sep 04 09:01:04 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:04 battery soc 0: 56%
Sep 04 09:01:04 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:04 site power: -2699W
Sep 04 09:01:04 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:04 charge currents: [6.09 0.011 0.009]A
Sep 04 09:01:04 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:04 charger status: A
Sep 04 09:01:14 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:14 ----
Sep 04 09:01:14 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:14 charge power: 1457W
Sep 04 09:01:14 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:14 pv power: 4718W
Sep 04 09:01:14 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:14 battery power: -2631W
Sep 04 09:01:14 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:14 grid power: 30W
Sep 04 09:01:14 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:14 battery soc 0: 56%
Sep 04 09:01:14 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:14 site power: -2701W
Sep 04 09:01:14 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:14 charge currents: [6.09 0.011 0.009]A
Sep 04 09:01:14 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:14 charger status: A
Sep 04 09:01:24 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:24 ----
Sep 04 09:01:24 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:24 charge power: 1457W
Sep 04 09:01:24 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:24 pv power: 4739W
Sep 04 09:01:24 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:24 battery power: -2625W
Sep 04 09:01:24 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:24 grid power: -6W
Sep 04 09:01:24 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:24 battery soc 0: 56%
Sep 04 09:01:24 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:24 site power: -2731W
Sep 04 09:01:24 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:24 charge currents: [6.09 0.011 0.009]A
Sep 04 09:01:24 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:24 charger status: A
Sep 04 09:01:34 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:34 ----
Sep 04 09:01:34 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:34 charge power: 1457W
Sep 04 09:01:34 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:34 pv power: 4755W
Sep 04 09:01:34 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:34 battery power: -2619W
Sep 04 09:01:34 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:34 grid power: 7W
Sep 04 09:01:34 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:34 battery soc 0: 56%
Sep 04 09:01:34 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:34 site power: -2712W
Sep 04 09:01:34 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:34 charge currents: [6.09 0.011 0.009]A
Sep 04 09:01:34 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:34 charger status: A
Sep 04 09:01:44 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:44 ----
Sep 04 09:01:44 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:44 charge power: 1457W
Sep 04 09:01:44 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:44 pv power: 4751W
Sep 04 09:01:44 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:44 battery power: -2634W
Sep 04 09:01:44 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:44 grid power: -9W
Sep 04 09:01:44 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:44 battery soc 0: 56%
Sep 04 09:01:44 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:44 site power: -2743W
Sep 04 09:01:44 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:44 charge currents: [6.09 0.011 0.009]A
Sep 04 09:01:44 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:44 charger status: A
Sep 04 09:01:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:54 ----
Sep 04 09:01:54 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:54 charge power: 1457W
Sep 04 09:01:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:54 pv power: 4723W
Sep 04 09:01:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:54 battery power: -2517W
Sep 04 09:01:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:54 grid power: 4W
Sep 04 09:01:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:54 battery soc 0: 57%
Sep 04 09:01:54 nuc-evcc evcc[434]: [site  ] DEBUG 2022/09/04 09:01:54 site power: -2613W
Sep 04 09:01:54 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:54 charge currents: [6.09 0.011 0.009]A
Sep 04 09:01:54 nuc-evcc evcc[434]: [lp-1  ] DEBUG 2022/09/04 09:01:54 charger status: A

### What type of operating system are you running?

Linux

### Version

0.100

premultiply, 2022-09-04:
`residualPower: 100`

closed 2022-09-04 by premultiply

Stargate85, 2022-09-04:
@premultiply  Habe ich wie von dir geschrieben geändert. Das verhalten ist aber das gleiche:
`site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  residualPower: 100
  prioritySoC: 10
  bufferSoC: 98
`

`Sep 04 10:02:37 nuc-evcc evcc[493]: [lp-1  ] DEBUG 2022/09/04 10:02:37 charge power: 1451W
Sep 04 10:02:37 nuc-evcc evcc[493]: [site  ] DEBUG 2022/09/04 10:02:37 pv power: 5686W
Sep 04 10:02:37 nuc-evcc evcc[493]: [site  ] DEBUG 2022/09/04 10:02:37 battery power: -3372W
Sep 04 10:02:37 nuc-evcc evcc[493]: [site  ] DEBUG 2022/09/04 10:02:37 grid power: 0W
Sep 04 10:02:37 nuc-evcc evcc[493]: [site  ] DEBUG 2022/09/04 10:02:37 battery soc 0: 58%
Sep 04 10:02:37 nuc-evcc evcc[493]: [site  ] DEBUG 2022/09/04 10:02:37 site power: -3272W
Sep 04 10:02:37 nuc-evcc evcc[493]: [lp-1  ] DEBUG 2022/09/04 10:02:37 charge currents: [6.08 0.011 0.008]A
Sep 04 10:02:37 nuc-evcc evcc[493]: [lp-1  ] DEBUG 2022/09/04 10:02:37 charger status: A
`

premultiply, 2022-09-04:
> `Sep 04 10:02:37 nuc-evcc evcc[493]: [lp-1 ] DEBUG 2022/09/04 10:02:37 charger status: A`

Kein Fahrzeug angeschlossen. Was sollte da geladen werden???