evcc-io/evcc #4310: Solarmax WR

mobbl, 2022-09-05:
openWB unterstützt seit kurzem die Wechselrichter von Solarmax, ist eine Umsetzung auch für evcc geplant/möglich?

andig, 2022-09-05:
Kann Solarmax kein Sunspec?

closed 2022-09-05 by andig

premultiply, 2022-09-05:
@mobbl Hast du so ein Gerät? Ich würde da gerne mal per ModbusTCP direkt draufschauen und das Template noch verbessern.
Ich vermute da geht noch mehr.

mobbl, 2022-09-05:
> @mobbl Hast du so ein Gerät? Ich würde da gerne mal per ModbusTCP direkt draufschauen und das Template noch verbessern. Ich vermute da geht noch mehr.

Ein Bekannter hat so Geräte von Solarmax und wir sind gerade dabei eine Überschussladung hinzubekommen. 
Wir können gerne mal versuchen was wir aus dem WR + Speicher rausbekommen.