evcc-io/evcc #4325: Wrong SunSpec energy unit

premultiply, 2022-09-05:
If SunSpec energy register values are read the default scaling factor is wrong by 1000.
The raw value (unit: Wh) is currently read as kWh.

If the device returns the exported energy 15369 [Wh] (=15,369 kWh) (SF 1) evcc reads this as 15369,0 kWh.

andig, 2022-09-06:
Ich denke das hätte man nie so (=Nutzung von mbmd als Zwischenschicht) bauen dürfen. Schöner wäre, wir würden das überall rauswerfen und durch Model/Block/Point für Sunspec ersetzen.

premultiply, 2022-09-06:
Sprich mbmd und samt Aliasen ganz raus? 😧

andig, 2022-09-06:
Drei Alternativen:

- alles umbauen, Aliase weg
- Aliase um `scale` ergänzen
- Aliase reparieren

Ich tendiere  kurzfristig zur letzten Variante

andig, 2022-09-06:
@premultiply kannst Du mal probieren, obs mit dem PR geht?

premultiply, 2022-09-06:
Du meinst erstmal rein in mbmd?

andig, 2022-09-07:
Ich meinte egtl. die mbmd Version aus dem PR go getten und hier ausprobieren. Wenns nicht geht merge ich die sonst auch.

andig, 2022-09-07:
Ich habs gemerged, jetzt brauchst Du einfach 

    go get github.com/volkszaehler/mbmd@master

premultiply, 2022-09-08:
`go build` meldet dann
```
# github.com/evcc-io/evcc/provider
provider\modbus.go:180:18: cannot use dev.QueryPoint(m.conn, m.op.SunSpec.Model, m.op.SunSpec.Block, m.op.SunSpec.Point) (value of type float64) as type meters.MeasurementResult in assignment
```

andig, 2022-09-08:
Yep, musst du mit ändern. Der float ist der ausgelesene Wert.

premultiply, 2022-09-08:
Yep. War in
```
# github.com/evcc-io/evcc/meter
meter\modbus.go:165:15: cannot use dev.QueryPoint(m.conn, op.SunSpec.Model, op.SunSpec.Block, op.SunSpec.Point) (value of type float64) as type meters.MeasurementResult in assignment
```
auch noch drin.

premultiply, 2022-09-08:
Jetzt stimmt es.
```
evcc meter
[main  ] INFO 2022/09/08 21:26:20 evcc 0.0.1-alpha
[main  ] INFO 2022/09/08 21:26:20 using config file L:\git\evcc\evcc.yaml
pv1
---
Power:  0W
Energy: 15194.9kWh

pv2
---
Power:  0W
Energy: 4440.2kWh
```

andig, 2022-09-08:
Nice 👍🏻

PR #4347 (2022-09-08): Fix SunSpec energy units
Fixes #4325
closed 2022-09-08 by andig