evcc-io/evcc #4326: Min+PV reduziert die Leistung nicht

dirktebie, 2022-09-06:
### Describe the bug

hallo,

ich habe mir mal am Sonntag die Zeit genommen, und evcc und das Auto beim Laden beobachtet.

Mode: Min+PV
Version: 0.100 (aber auch bereits bei 0.98 gesehen)
Auto: VW-ID3, 45kwh, 2~AC(max 16A), Min 6A, Max 16A
Wallbox: EVSE-DIN (Stark-in-Strom Easy)

Die PV machte gute 5kw.
evcc startet mit 2,8kw. (2x 1380W / 230V = 6A / Phase)

nach kurzer zeit wird auf den Überschuss von ca. 4,4kw (10A) hochgeschaltet.

dann weiter, ca 5,3kw abgelesen.
dann weiter bis ca 7,3kw (16A).
PV-Leistung aber weiterhin 5kw, Überschuss ca. 4,5kw.

ich schalte das Laden Aus.

### Steps to reproduce

nach einer minute schalte ich wieder ein, Min+PV.

die Ladeleistung geht sofort wieder auf 7,3kw, 16A.

Noch die Variante PV probiert:
Lädt entweder gar nicht, oder benötigt "ewig" (viel länger als die eingestellte Verzögeungszeit).

Probiert habe ich:
Auto trennen und wieder anstecken.
-wieder 7,3kw(max).

Wallbox Aus - 1 Minute warten - Ein.
- wieder 7,3kw.

Raspi Neustart (Aus - 10 sekunden - Ein)
- Laden startet wieder mit 2,8kw (Min).

### Configuration details

[evcc.zip](https://github.com/evcc-io/evcc/files/9496497/evcc.zip)

### Log details

[log.txt](https://github.com/evcc-io/evcc/files/9496511/log.txt)

wenn ich keins habe...

### What type of operating system are you running?

Linux

### Version

0.100

RTTTC, 2022-09-06:
You have exposed your car passwords !!

closed 2022-09-06 by premultiply

RTTTC, 2022-09-06:
As far as your problem -  you have set your vehicle as
` minSoC: 60` 
So EVCC will do maximum to get to this minimum state of charge as soon as possible via any means possible.

And I have a question about the charger - what kind of charger do you have? Do you get "charger: serial: timeout" errors ? I have EVSE-DIN and do get these errors a few times per day.

dirktebie, 2022-09-07:
minSoC: 60
I had a SOC above 75% at the time, i noticed this.

 "charger: serial: timeout": i have this too. but how i can optimize it, cable, shield, lenght of cable ?

RTTTC, 2022-09-07:
Regarding ocasional charger errors - nothing can be done about it, it’s the nature of the beast, or poor design.

About your original error - your log file is empty, and you need better logging configuration too.
Hard to say without the log.

RTTTC, 2022-09-14:
Hopefully this will be the answer to EVSE-DIN charger: serial: timeout errors ! https://github.com/evcc-io/evcc/discussions/4182

premultiply, 2022-09-14:
https://github.com/evcc-io/evcc/discussions/4182#discussioncomment-3441995