evcc-io/evcc #4346: add selfConsumption (and other calculated values) to influxdb

kai-morich, 2022-09-08:
**Is your feature request related to a problem? Please describe.**
Calculating the selfConsumption can be done in Grafana e.g. with:

    import "math"
    import "interpolate"
    from(bucket: "evcc")
      |> range(start: v.timeRangeStart, stop:v.timeRangeStop)
      |> filter(fn: (r) => r._measurement == "homePower" or r._measurement == "pvPower")
      |> interpolate.linear(every: 15s)
      |> pivot(rowKey: ["_time"], columnKey: ["_measurement"], valueColumn: "_value")
      |> map(fn: (r) => ({r with selfPower: math.mMin(x:r.homePower, y:r.pvPower) }))
      |> filter(fn: (r) => exists r.selfPower)
      |> integral(column:"selfPower", unit:1h)

but this gets very slow for larger time ranges.

**Describe the solution you'd like**
Add calculated values that are shown in UI like `selfConsumption` to influxdb data model

**Describe alternatives you've considered**
I did calculation in Grafana as shown above, but to slow. 

**Additional context**
Each metric is is reported with own timestamp, so first normalized values have to be created with `interpolate` to be able to calculate the minumim. I asume the own timestamps are not a bug, but are the basically independent timestamps when the data was collected.

Additionally I store absolute home/pv/self-power meter readings on a daily basis in influxdb, so I have correct values for long term statistics.  Integral of your home/pv/...-power values comes relatively close to these values and easily allows short term analysis.

isarrider, 2022-09-09:
Hint: dont do it in the grafana query - do it in Transform as binary operation...
I add my PV Power and the measured power of the energymeter - the result is my selfconsumption
(as the value of the measured power is negative...)
Works like a charm for me...

andig, 2022-09-30:
Keine weiteren Erkenntnisse

closed 2022-09-30 by andig