evcc-io/evcc #4425: Support the newer NRGKick chargers

florian-h05, 2022-09-13:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2022-09-13:
NRGKick has not published an API, hence wontfix.

closed 2022-09-13 by andig

florian-h05, 2022-09-13:
@andig 
I accidentally opened the issue before I was done writing the description, please read the description.

florian-h05, 2022-09-13:
They are planning to release a new API.

andig, 2022-09-13:
Feel free to reopen when released.

frlp542, 2023-10-09:
a custom API appears to be available now https://www.nrgkick.com/wp-content/uploads/2023/10/20231005_PV-Anleitung-Custom-API_03_web.pdf

florian-h05, 2023-10-09:
As far as I understand it right, this is no API that actually allows external systems to control the NRGKick — it is meant for integrating other solar systems than the officially supported ones.
It spins up a web server that the NRGKick requests data from.

frlp542, 2023-10-09:
ok now I understand. thanks for the update. 
As they sell a license now to upgrade your NRGKick to support the PV charging I think this topic for a new API will remain closed... forever.

florian-h05, 2023-10-09:
Yeah, I guess this will stay closed. However in https://github.com/evcc-io/evcc/discussions/4048#discussioncomment-7233264 some progress is made in using Bluteooth to control the new NRGKick chargers, so probably one day it will be possible to control the new ones using Bluetooth.

I am very disappointed by DiniTech/NRGKick that they don't release a real API and that they sell a license for PV charging. I am done with NRGKick, I do not need them for PV charging as I had to buy a new charger for the KfW 442 funding.

rpoisel, 2023-11-01:
I will soon contact the [VKI](https://vki.at/) to enforce warranty claims regarding communication capability: [Link](https://github.com/evcc-io/evcc/discussions/4048#discussioncomment-7440466). If anyone wants to join, please just get in touch with me.

dupondje, 2023-11-20:
I have a NRGKick (new) for almost a year, and bought it with the idea that an API was going to follow soon.
As we have some kind of capacity tarif in Belgium, I really need the API to loadbalance charging.

So I got frustrated, the PV plugin didn't solve the issue at all, so started to debug the communication myself :)
Result: https://github.com/dupondje/nrgkick-proto
-> This is the proto file for the protobuf communication between the NRGKick and the Android app over WiFi (most likely same protocol over bluetooth).

Also written a Home Assistant integration which uses that code to give details about the charger and be able to set charging rate etc :)
https://github.com/dupondje/nrgkick-ha

As evcc does not have a loadbalancing option yet, I did not look into writing a EVCC plugin for it. But with this information it should be quite easy to write one.

aichingerk, 2024-03-03:
Is it planned to realise an integration for NRGKick wallboxes with the information from @dupondje ?
If you need someone to test it, I'd be happy to do so, as I've been waiting ages for NRGKick to be integrated into EVCC.

andig, 2024-03-03:
We won't implement this: the API is not official and NRG is more or less hostile to opening its API. Best way forward here is to use the HA solution as "custom" charger in evcc.

@dupondje if you disagree and want to implement the NRGkick in evcc you're welcome. Core team won't touch this.

dupondje, 2024-03-03:
> @dupondje if you disagree and want to implement the NRGkick in evcc you're welcome. Core team won't touch this.

I don't use evcc, so likely won't spend time on this. But the proto code is open, so it should be easy to implement this in evcc for anyone with some coding experience.

wenzul, 2024-06-10:
Since february there is a [custom API documentation](https://www.nrgkick.com/wp-content/uploads/2024/02/20240220_Anleitung-zum-PV-Laden-mit-NRGkick_custom-API.pdf) online. Is there a chance to use the mentioned `Custom_API_Webserver.py` to implement the similar behaivour in evcc?

premultiply, 2024-06-10:
See https://github.com/evcc-io/evcc/issues/4425#issuecomment-1753634370

rpoisel, 2024-06-10:
> I will soon contact the [VKI](https://vki.at/) to enforce warranty claims regarding communication capability: [Link](https://github.com/evcc-io/evcc/discussions/4048#discussioncomment-7440466). If anyone wants to join, please just get in touch with me.

For your reference: I was successful with my claim. It took some help from the [Austrian Chamber of Labour](https://www.arbeiterkammer.at/beratung/konsumentenschutz/) (many thanks to them), but I was able to return the NRGkick and opted for a go-e charger instead, which works **flawlessly** with evcc.

wenzul, 2024-06-10:
> See [#4425 (comment)](https://github.com/evcc-io/evcc/issues/4425#issuecomment-1753634370)

Thanks. I didn't realize that it's an older version as the link is dead.

> but I was able to return the NRGkick

How long after the purchase were you able to return it?

rpoisel, 2024-06-10:
> How long after the purchase were you able to return it?

In Austria there is a [2-year guarantee](https://ooe.arbeiterkammer.at/beratung/konsumentenschutz/konsumentenrecht/konsumentenrecht/Gewaehrleistung_ist_gesetzlich_vorgeschrieben.html). It's even better if you try to return a product within the first 6 months of purchase. I returned it within the first year. But it was quite a hassle, and it was only possible because lawyers from the Austrian Chamber of Labor intervened.