evcc-io/evcc #4431: bufferSoC: add warning when disable threshold <= 0?

andig, 2022-09-14:
**Update**

Ok, ich sehe es:

	lp.log.DEBUG.Printf("pv charge current: %.3gA = %.3gA + %.3gA (%.0fW @ %dp)", targetCurrent, effectiveCurrent, deltaCurrent, sitePower, activePhases)

	// in MinPV mode or under special conditions return at least minCurrent
	if (mode == api.ModeMinPV || batteryBuffered || lp.climateActive()) && targetCurrent < minCurrent {
		return minCurrent
	}

Weil `batteryBuffered` geben wir die Ladung mit 6A frei. Danach passiert:

```
[lp-1  ] DEBUG 2022/09/14 16:52:48 site power 10W >= 0W disable threshold
[lp-1  ] DEBUG 2022/09/14 16:52:48 pv disable timer start: 3m0s
```

Das macht tatsächlich ganz wenig sinn und ist einfach ein Konfigurationsfehler! Bei Hausbatterie muss der Arbeitspunkt eigentlich immer leicht in Richtung Netzbezug verschoben sein. Diesen direkt als `disable threshold` zu nutzen ist natürlich unsinnig!

@premultiply sollten wir bei `bufferSoC` eine Warnung ausgeben falls `disable threshold <= 0`?

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/4430#discussioncomment-3646105_

VolkerK62, 2022-09-14:
Was ist hier mit `disable threshold` gemeint?
Der reine Wert aus der Config oder ist das unter Berücksichtigung von `residualpower`?

andig, 2022-09-14:
Eigentlich der aus der Config. Wer an `residualpower` rumdreht sollte wissen, was er tut?

VolkerK62, 2022-09-14:
ok. Ich habe nix angegeben, also default=0 und hatte noch nie Probleme.
Aber ich nutze auch kein `bufferSoC`. Vergiss das ich gefragt habe ;-)

andig, 2022-09-14:
Ohne `bufferSoC` rechnen wie die Batterie ja raus. *Mit* liegt der Arbeitspunkt aber bei aktiver Batterie um 0W. Da fehlt dann halt völlig die Hysterese.

...oder ich hab das falsch interpretiert. Waiting for @premultiply

andig, 2022-09-14:
Wichtig ist übrigens, dass wir bei `bufferSoC` davon ausgehen, dass `minCurrent` freigegeben werden kann. Es wäre also sinnig (Dokuthema) wenn die Batterie die passende Leistung bei dem Wert dann auch liefern könnte...

VolkerK62, 2022-09-14:
... und wenn die Batterie das nicht kann, wird mit Netzstrom weitergeladen bis `bufferSoC` erreicht ist.
So habe ich es verstanden  https://github.com/evcc-io/evcc/discussions/4236

premultiply, 2022-09-14:
Der bufferSoC ist wirklich die allereinfachste Implementierung:
Batterie-SoC > bufferSoC: PV-Modus entspricht dem Min-PV-Modus.

Was genau die Thresholds dabei machen habe ich ehrlich gesagt noch nie betrachtet und noch nie genutzt.
Aber auch hier: Da überhaupt nicht abgeschaltet wird braucht man dabei eigentlich auch keine Hysterese.

VolkerK62, 2022-09-15:
@premultiply  Nur zum Verständnis, folgendes Szenario:
Fahrzeug lädt im Modus PV. 
BufferSoC ist noch nicht erreicht. 
PV-Erzeugung sinkt, evcc stoppt die Ladung.
Hausbatterie wird weiter geladen; BufferSoC wird überschritten.
Aufgrund eines größeren Verbrauch im Haus, springt die Batterie ein und schickt Strom ins Haus.
Fängt dann das Auto wieder an zu Laden? (Stichwort `batteryBuffered`)

premultiply, 2022-09-15:
Sobald der `bufferSoC` erreicht ist verhält es sich wie wenn man manuell auf Min+PV schalten würde.
Also ja, das Auto fängt dann mindestens mit minCurrent an zu laden.

andig, 2022-09-15:
Danke für die Erklärung. Dann können wir hier zumachen weil ich einfach auf den falschen Parameter geschaut habe. Weiter in https://github.com/evcc-io/evcc/discussions/4430#discussioncomment-3646105- da passt meine Erklärung dann nämlich nicht zum Problem :(

closed 2022-09-15 by andig

premultiply, 2022-09-15:
Genau. Die Leistung usw. spielt überhaupt keine Rolle. Sobald der bufferSoC der Batterie erreicht ist wird wie bei Min+PV geladen.
Der Ladevorgang wird auch gestartet wenn zuvor nicht geladen oder oder sogar schon massiver Netzbezug da ist.