evcc-io/evcc #4440: Unable to update from 0.100 to 0.101 on Hassio

alexebner, 2022-09-14:
### Describe the bug

Unable to update.

### Steps to reproduce

Press Update in Home Assistant

### Configuration details

 Home Assistant 2022.9.4 
Supervisor 2022.08.6 
Operating System 8.5 
Frontend 20220907.2 

### Log details

2022-09-14 20:52:00.622 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139627002332880] Error updating evcc: Can't install andig/evcc:0.101: 404 Client Error for http+docker://localhost/v1.41/images/create?tag=0.101&fromImage=andig%2Fevcc&platform=linux%2Famd64: Not Found ("manifest for andig/evcc:0.101 not found: manifest unknown: manifest unknown")
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 167, in async_install
    await async_update_addon(self.hass, slug=self._addon_slug, backup=backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 45, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Can't install andig/evcc:0.101: 404 Client Error for http+docker://localhost/v1.41/images/create?tag=0.101&fromImage=andig%2Fevcc&platform=linux%2Famd64: Not Found ("manifest for andig/evcc:0.101 not found: manifest unknown: manifest unknown")

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 200, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 676, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 713, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 152, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 411, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 169, in async_install
    raise HomeAssistantError(f"Error updating {self.title}: {err}") from err
homeassistant.exceptions.HomeAssistantError: Error updating evcc: Can't install andig/evcc:0.101: 404 Client Error for http+docker://localhost/v1.41/images/create?tag=0.101&fromImage=andig%2Fevcc&platform=linux%2Famd64: Not Found ("manifest for andig/evcc:0.101 not found: manifest unknown: manifest unknown")

### What type of operating system are you running?

Linux

### Version

0.100

LeoWinterDE, 2022-09-14:
Same here:
`Can't install andig/evcc:0.101: 404 Client Error for http+docker://localhost/v1.41/images/create?tag=0.101&fromImage=andig%2Fevcc&platform=linux%2Farm64: Not Found ("manifest for andig/evcc:0.101 not found: manifest unknown: manifest unknown")`

andig, 2022-09-15:
Sollte jetzt klappen. Da fehlte das getaggte Docker image.

closed 2022-09-15 by andig

alexebner, 2022-09-15:
Danke funktioniert.

alexebner, 2022-09-17:
Same problem on 0.101.1