evcc-io/evcc #4449: guardDuration: Show in UI or remove?

premultiply, 2022-09-15:
In dem Kontext hier ist mir aber heute auch noch aufgefallen dass während der Guard-Timer läuft keine passende Textmeldung in der UI erscheint und man sich dann fragt warum gerade trotz ausreichend Überschuss und sonstigen Bedingungen nicht geladen wird im PV-Modus.

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/4430#discussioncomment-3648268_

VolkerK62, 2022-09-15:
Guard-Timer = contactor delay?
Ich finde die UI Hinweise zum `enable timer` und `disable timer` ganz praktisch. Da wäre es nur konsequent, die "Pause" ebenfalls anzuzeigen.

andig, 2022-09-15:
Ein Bug ist es unabhängig davon nicht- geht aber sicher anwenderorientierter.

RTTTC, 2022-09-15:
I would say - remove, with explanation in documentation AND in Yaml, the importance of setting enable/disable delay that is not too short, to protect the contactor.

andig, 2022-09-15:
You could still kill it using the api. You can't do that today...

RTTTC, 2022-10-31:
I see in some configurations posted here that some people select very short `enable delay` and `disable delay` .. 1 minute for example. So at this time only `guardDuration `is protecting them.. Some more user education is needed to better understand what it means and what it does. (limited contactor cycle life etc)

pauxus, 2022-10-31:
Totally agree. But of course, the default values (1m enable, 3m disable, 5m guard) are already far from optimal. I would suggest making them way more sensible.

andig, 2022-10-31:
More sensible or optimal in terms of what? Suggestions welcome.

pauxus, 2022-11-02:
In terms of equipment life.

For comparison, E3DC's S10 has a default setting of 2 hours (somewhat similar to guard duration) - which might be a little far on the safe side. I think at least 15 min or better 30min (or even more) for guard duration would be safer for the equipment. Also, I agree with @RTTTC that this should be more prominent in the documentation (Right now, it is buried in the reference)

Perhaps it could also be upgraded to a site parameter (defaultGuardDuration) and be asked for in `evcc configure`

andig, 2022-11-03:
> has a default setting of 2 hours

Noch längere Details machen evcc unbenutzbar. Kein Konsens das Delay auszubauen oder zu visualisieren, daher wontfix.

closed 2022-11-03 by andig