evcc-io/evcc #4460: Duplicate (notification) messages Alfen charging session

hyperbart, 2022-09-15:
### Describe the bug

At the end of a charging session when a driver opens the Polestar 2 car and pushes the button to unlock the cable multiple messages are being duplicated. See result below.

translated
September 14 
Car connected
Charging session started
Charger finished session, kwh delivered, duration

today september 15
charging session started
Charger finished session, kwh delivered, duration
car disconnected
car connected
car disconnected

[image]

### Steps to reproduce

Alfen EVSE
EVCC
Configure EVCC to do messaging:

### Configuration details

```
messaging:
  events:
    start:
      title: Laadbeurt gestart
      msg: Er is een laadbeurt gestart op xxxxxxx
    stop:
      title: Laadbeurt gestopt
      msg: |
        Oplader {{.title}} heeft een laadbeurt voltooid
        Er is {{round (divf .chargedEnergy 1000) 2 }} kWh geleverd op {{.chargeDuration}}.
    connect:
      title: Wagen aangesloten op xxxxxxx
      msg: Er is een auto aangesloten op de lader in xxxxxx
    disconnect:
      title: Wagen ontkoppeld op xxxxxx
      msg: De wagen is ontkoppeld van de lader in xxxxx
```

### Log details

```
[site  ] DEBUG 2022/09/15 00:06:00 ----
[lp-1  ] DEBUG 2022/09/15 00:06:00 charge power: 0W
[site  ] DEBUG 2022/09/15 00:06:00 pv power: 0W
[site  ] DEBUG 2022/09/15 00:06:00 grid power: 487W
[site  ] DEBUG 2022/09/15 00:06:00 site power: 587W
[lp-1  ] DEBUG 2022/09/15 00:06:00 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/09/15 00:06:00 charger status: B
[site  ] DEBUG 2022/09/15 00:06:01 ----
[lp-1  ] DEBUG 2022/09/15 00:06:01 charge power: 0W
[site  ] DEBUG 2022/09/15 00:06:01 pv power: 0W
[site  ] DEBUG 2022/09/15 00:06:01 grid power: 482W
[site  ] DEBUG 2022/09/15 00:06:01 site power: 582W
[lp-1  ] DEBUG 2022/09/15 00:06:01 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/09/15 00:06:01 charger status: B
[site  ] DEBUG 2022/09/15 00:06:02 ----
[lp-1  ] DEBUG 2022/09/15 00:06:02 charge power: 0W
[site  ] DEBUG 2022/09/15 00:06:02 pv power: 0W
[site  ] DEBUG 2022/09/15 00:06:02 grid power: 484W
[site  ] DEBUG 2022/09/15 00:06:02 site power: 584W
[lp-1  ] DEBUG 2022/09/15 00:06:02 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/09/15 00:06:02 charger status: B
[... 397 lines cut ...]
[site  ] DEBUG 2022/09/15 00:06:58 ----
[lp-1  ] DEBUG 2022/09/15 00:06:58 charge power: 0W
[site  ] DEBUG 2022/09/15 00:06:58 pv power: 0W
[site  ] DEBUG 2022/09/15 00:06:58 grid power: 477W
[site  ] DEBUG 2022/09/15 00:06:58 site power: 577W
[lp-1  ] DEBUG 2022/09/15 00:06:58 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/09/15 00:06:59 charger status: A
[lp-1  ] DEBUG 2022/09/15 00:06:59 charger disable: contactor delay 4m50s
[site  ] DEBUG 2022/09/15 00:06:59 ----
[lp-1  ] DEBUG 2022/09/15 00:06:59 charge power: 0W
[site  ] DEBUG 2022/09/15 00:06:59 pv power: 0W
[site  ] DEBUG 2022/09/15 00:06:59 grid power: 477W
[site  ] DEBUG 2022/09/15 00:06:59 site power: 577W
[lp-1  ] DEBUG 2022/09/15 00:06:59 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Linux

### Version

Ubuntu 20.04 LTS
Docker
EVCC 0.101

andig, 2022-09-15:
Please show a log for the screenshot. Happy to reopen with that informatoin.

hyperbart, 2022-09-15:
> 

What do you mean "Show a log for the screenshot"? The log of EVCC is there?

andig, 2022-09-15:
The log has totally different timestamps?

hyperbart, 2022-09-16:
> The log has totally different timestamps?

The messages in Telegram are sent at midnight, 00:06 and the logs are also from that time, I grep'ed them from the logs?

Or do you mean you want to see the earlier logs too? Sure! I have added the logs for every event/message.