evcc-io/evcc #4487: evcc 0.101.1: cannot create vehicle template: invalid element

emcepe, 2022-09-17:
### Describe the bug

just updated to 0.101.1 and get the following errors during evcc vehicle:

```
root@raspberry-pi-4b:~# evcc vehicle
[main  ] INFO 2022/09/17 10:24:17 evcc 0.101.1
[main  ] INFO 2022/09/17 10:24:17 using config file /etc/evcc.yaml
[mqtt  ] INFO 2022/09/17 10:24:17 connecting evcc-1363470488 at tcp://localhost:1883
.SoC:     vehicle not available: cannot create vehicle 'template': invalid element 'name: minsoc'
Capacity: 0kWh

root@raspberry-pi-4b:~# evcc vehicle
[main  ] INFO 2022/09/17 10:25:18 evcc 0.101.1
[main  ] INFO 2022/09/17 10:25:18 using config file /etc/evcc.yaml
[mqtt  ] INFO 2022/09/17 10:25:18 connecting evcc-1705392814 at tcp://localhost:1883
.SoC:     vehicle not available: cannot create vehicle 'template': invalid element 'name: targetsoc'
Capacity: 0kWh
```

versions before worked fine.

commenting minSoc and targetSoc in evcc.yaml "fixes" that error of evcc vehicle.

### Steps to reproduce

"evcc vehicle" with "minSoc and targetSoc" in evcc.yaml

### Configuration details

```
vehicles:
- type: template
  template: hyundai
  title: Hyundai KONA
  user: bla
  password: blub
  vin: foobar
  # capacity of SoC in kWh
  capacity: 64
  language: de
  name: ev
  # charge to at least 21% independent of charge mode
  minSoC: 21
  # limit charge to 80%
  targetSoC: 80
  # ev phases (default 3)
  phases: 0
  # cache API response
  cache: 10m
```

### Log details

N/A

### What type of operating system are you running?

Linux

### Version

evcc version 0.101.1

Diavolezza, 2022-09-17:
I also upgraded to 0.101.1 today and I get a very similar error-message:

sundance@raspberrypi:~ $ sudo evcc -l debug vehicle
[main  ] INFO 2022/09/17 10:31:43 evcc 0.101.1
[main  ] INFO 2022/09/17 10:31:43 using config file /etc/evcc.yaml
.SoC:     vehicle not available: cannot create vehicle 'template': invalid element 'name: accesstoken'
Capacity: 0kWh

That was working yesterday with 0.101

andig, 2022-09-17:
Fixed

closed 2022-09-17 by andig

emcepe, 2022-09-17:
thank you for the very fast fix. Have a nice sunday.