evcc-io/evcc #4496: Vehicle template .targetsoc and minsoc not working in 0.101.2

pauxus, 2022-09-17:
The following configuration worked in 0.101:

```yaml
vehicles:                                                                                                                                                                                          
- name: opel                                                                                                                                                                                       
  type: template                                                                                                                                                                                   
  template: opel                                                                                                                                                                                   
  title: e-Corsa                                                                                                                                                                                   
  user: xxx                                                                                                                                                               
  password: xxx                                                                        
  capacity: 46
  targetSoC: 80
```

However after updating to 0.101.2, it fails with:

```
[main  ] INFO 2022/09/17 16:52:10 evcc 0.101.2
[main  ] INFO 2022/09/17 16:52:10 using config file /etc/evcc.yaml
.SoC:     vehicle not available: cannot create vehicle 'template': invalid element 'name: targetsoc'
Capacity: 0kWh
```

gosmarty, 2022-09-17:
Ich habe das selbe Problem.
```yaml
vehicles:
- name: bmw
  type: template
  template: bmw
  title: BMW i3
  capacity: 38 # kWh
  user: 
  password: 
  vin: 
  mode: pv
  minSoC: 0
  targetSoC: 80
```
`evcc vehicle` gibt folgendes aus:
`.SoC:     vehicle not available: cannot create vehicle 'template': invalid element 'name: minsoc' Capacity: 0kWh`
Sobald ich minSoC und targetSoC auskommentiere funktioniert es wieder...

tb646, 2022-09-17:
I have the same Problem. Deactivating SoC and current lines by setting them as comments solves the problem.

andig, 2022-09-17:
Should be fixed in 0.101.3

closed 2022-09-17 by andig

gosmarty, 2022-09-17:
I can confirm - is fixed!
Vielen Danke @andig für das schnelle Korrigieren und überhaupt für dieses tolle Projekt!