evcc-io/evcc #4518: Trennen des Autos wird nicht erkannt und Wert under "Geladen" ist der vom vorherigen Ladevorgang

Robbe64, 2022-09-19:
### Describe the bug

Wird das Fahrzeug getrennt, so wird dieses nun nicht (mehr) erkannt. Das Fahrzeug wird weiterhinn als Verbunden dargestellt. Erst bei einem Neustart von EVCC wird das Fahrzeug wieder als getrennt erkannt. Dazu kommt, dass nach dem Verbinden der Wet unter "Geladen" der ist vom vorherigen Ladevorgang was dann durch den Estimator auch noch zu einem falschen SoC führt.

Zeitlicher Ablauf passend zum Log:
08:03: Auto abgesteckt
08:15: Auto wieder angestekct
08:45: Auto abgestekct
08:56: Auto wieder angeschlossen

Zwiscehndurch hat es dann um 08:20 von selbst angefangen zu laden obwohl im Status PV und kein Überschuss vorhanden. Warum weiß ich nicht und auch nicht ob das was mit dem Problem zu tun hat. Stoppen konnte ich es nur über die Easee-App.

Die Probleme fingen bei mir an als ich von 0.100 auf auf 0.102.1 aktualisiert habe. Geleichzeitig habe ich aber auch das default Vehicle konfiguriert. Ich kann aber auch nicht ausschließen, dass das wieder ein merkwürdiges verhalten der Easee ist wobei ich da eigentlich aktuell nichts ausßergewöhnliches feststellen konnte. In der Easee App sowie im Portal stimmt die Anzeige.

### Steps to reproduce

1. Auto anschließen -> Fahrzeug wird korrekt erkannt.
2. Auto etwas laden lassen. (optional für fehlerhafte Fahrzeugerkennung jedoch notwendig für falschen Wert unter "Geladen")
3. Auto abstecken (während es nicht lädt) -> Fahrzeug wird weiterhinn als Verbunden dargestellt.
4. EVCC neu starten -> Fahrzeug wird wieder korrekt als getrennt dargestellt
5. Auto anschließen -> Fahrzeug wird korrekt erkannt, jedoch wird unter "Geladen" kurz darauf der Wert der beim vorherigen Ladevorgang geladen wurde angezeigt -> Ladestand wird falsch dargestellt.

### Configuration details

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/9596693/evcc.yaml.txt)

### Log details

[evcc.log](https://github.com/evcc-io/evcc/files/9596788/evcc.log)

### What type of operating system are you running?

Docker container

### Version

0.102.1

andig, 2022-09-19:
Die Box liefert von 8:11 bis 8:14 den Status A, sonst nicht. Wenn Easee das Abstecken nicht erkennt können wir nichts machen.

closed 2022-09-19 by andig

Robbe64, 2022-09-19:
Nach dem Neustart von EVCC wird es aber zuverlässig korrekt erkannt. Da muss es doch dann irgend einen Unterschied geben. Oder wird beim Neustart aktiv der Status nachgefragt und ansonsten wird darauf gewartet, dass Easee von sich aus eine Änderung meldet?

andig, 2022-09-19:
Wir warten immer auf Easee. Was die schickt siehst Du im Trace Log. Vllt. Ist die Verbindung „stecken“ geblieben, auch da sollte der Easee Support aussagefähig sein.