evcc-io/evcc #4521: Problems since Update for PSA Vehilce SoC

sven0110, 2022-09-19:
After update from V0.100 to 0.102.1 evcc can‘t get information for my Peugeot e2008 Form the PSA-API:

vehicle soc: Get \"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/xxx/status?client_id=xxx\": Post \"https://idpcvs.peugeot.com/am/oauth2/access_token\": net/http: request canceled (Client.Timeout exceeded while awaiting headers) (Client.Timeout exceeded while awaiting headers)

Till V0.100 it works without problems.

Is Theresa anything to change in the config?

Thanks and best regards,
Sven

andig, 2022-09-19:
Problem with the PSA api.

marcsauter75, 2022-10-11:
have the same issue since 0.102 now on 0.104 still the same
ehicle soc: Get \"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/1566c497a5655746153316861546b6f324f5449774f446b6a497a63304d32466a4d446b344e47517a4f4745334d6d5a6b5a544d334f4455334e325534597a466a4f5745324f444d784d4455785a44646c4e44566d4f4459325a6a566a5a5451774d474d314f474d334f5451354e574d3d/status?client_id=1eebc2d5-5df3-459b-a624-20abfcf82530\": oauth2: cannot fetch token: 400 Bad Request\nResponse: {\"error\":\"invalid_grant\",\"error_description\":\"grant is invalid\"}

andig, 2022-10-11:
Different error, new topic.