evcc-io/evcc #4537: Zusätzlicher MIN+PV aber Zeitgesteuert.

elektro-freak, 2022-09-21:
Hallo zusammen,

ich würde mir wünschen das man MIN+PV auch Zeitlich aktivieren kann.

Hintergrund:

Meine Speicher soll sich morgens erstmal aufladen um dann ab 9 das Heißwasser von der Wärmepumpe zu unterstützen.
Danach würde ich gerne zu dieser Jahreszeit (Herbst-Winter) automatisch zeitabhänig die Wallbox ab 10:00 bis 15:30 auf MIN+PV setzen, damit das Auto möglicht günstig geladen werden kann. Ab 15:30 wenn vermutlich die Leistung meiner PV (11,3kw) runtergeht kann dann der Rest in den Akku. Dies hätte halt den Vorteil das bei gerade ausreichender Erzeugung weiter geladen wird und man nicht Manuell eingreifen muß. (In den Sommermonaten steht bei mir die Ladeautomatik nur auf PV)

Danke.

andig, 2022-09-21:
Dazu gibts eigentlich keinen Bedarf- die Zeitsteuerung kannst Du wunderbar mit Hilfe der APIs (siehe Doku) über eine Heimautomatisierung Deiner Wahl machen. Als Kern von evcc sehen wir das nicht an!

closed 2022-09-21 by andig

naltatis, 2022-09-22:
Was allerdings schon öfters in der Diskussion war ist eine Art "Wintermodus" bei dem nicht erst gewartet wird bis die volle Ladeleistung durch PV gedeckt wird sondern das Laden beginnt sobald auch nur ein bisschen Überschuss da ist. Das wäre nicht das gleich wie Min+PV (wird bspw. Mittags gekocht würde das Auto ggf. pausieren). Du hättest aber den Effekt, dass (im Gegensatz zu Min+PV) das Auto nicht die Nacht durchlädt sondern nach Sonnenuntergang stoppt.

elektro-freak, 2022-09-22:
> Dazu gibts eigentlich keinen Bedarf- die Zeitsteuerung kannst Du wunderbar mit Hilfe der APIs (siehe Doku) über eine Heimautomatisierung Deiner Wahl machen. Als Kern von evcc sehen wir das nicht an!

Das finde ich aber doch sehr schade, Ihr hebt ja auch eine Endzeit hinterlegt. Ich dachte das EVCC hier eingenstädig ist und meine Frau kommt mit dem Frontend wunderbar klar. Kern des Tools ist doch nunmal die Überschußladung zu optimieren. Daher dachte ich schon das man hier auch im EVCC diese funktionalöität der Zeitsteuerung haben müsste. Dies nun zuverlässig über eine Zeites tool zu steuern finde aufwenidger als man gerade denkt. Ich habe zwar ein node-red lauf aber das gehört aus meiner sich tatsächlich eher in EVCC als in meine Haussteuerung. Die ja aktuell mit der PV nix am kopf hat.

Ich denke das dieses Feature viele glücklich machen würde. Aber ist ja nicht meine Entscheidung sondern eure.

andig, 2022-09-22:
Wir müssen leider unsere knappen Ressourcen priorisieren. Für's erste ist wichtig, dass es hier Workarounds gibt. In anderen Bereichen ist das schwieriger und dort liegt erstmal unser Fokus. Danke für's Verständnis!