evcc-io/evcc #4542: Release counts as latest version instead of nightly

andig, 2022-09-22:
Mhhm. Das heißt also, dass das Patchlevel *älter* gewertet wird als eine Releaseversion (mit impliziten Patchlevel 0). Wir könnten jetzt:

- an die Releases ein Patchlevel 0 anhängen. Das erscheint mir aber ziemlich merkwürdig.
- die Versionsnummer der Nightlies bewusst um eine Minor Version anheben (ala "pre"). Das wäre unintuitiv.

Gibts einen Linuxguru der noch eine bessere Idee hätte? Auf jeden Fall ein Bug im Releasemanagement.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/4055#discussioncomment-3709010_

andig, 2022-09-23:
@Pitillo hättest Du evtl. eine Idee? Du scheinst Dich da besser auszukennen ;)

Pitillo, 2022-09-23:
Wenn ich es im Debian Manual jetzt richtig überflogen habe und auch das nightly naming scheme von EVCC richtig verstehe, dann muss die Tilde ~ in der Version nur durch ein + ersetzt werden. Die Tilde wird als less than nothing interpretiert, also als kleiner als wenn kein Ahnung mit Tilde vorhanden ist.

```
p    0.102.1~1663561686
p    0.102.1~1663600355
p    0.102.1~1663647475
p    0.102.1~1663734253
p    0.102.1
```

Muss also werden zu
```
p    0.102.1
p    0.102.1+1663561686
p    0.102.1+1663600355
p    0.102.1+1663647475
p    0.102.1+1663734253
```

Kann man hier lesen: https://www.debian.org/doc/manuals/maint-guide/first.en.html#namever

Ich hoffe das hilft als Hinweis, hab aktuell keinen Rechner zur Hand um es selbst umzusetzen oder zumindest auch mit dem dort beschriebenen dpkg Befehl auszuprobieren, vielleicht könnte das jemand anderes übernehmen 🙂

andig, 2022-09-23:
Aaaaah, super! Ich habs mal geändert und Build angeworfen.

@VolkerK62 in 10min kannst Du testen!

VolkerK62, 2022-09-23:
Funktioniert. Kleines Zeichen, große Wirkung ;-)
Vielen Dank

andig, 2022-09-23:
Nice- danke an @Pitillo 💚

closed 2022-09-23 by andig