evcc-io/evcc #4575: Current reading of Alfen connected to 1 phase is wrong

treynaer, 2022-09-24:
### Describe the bug

My Alfen Eve charger is setup and configured in 1 Phase mode. Unfortantly this causes an error in the UI.

`Failed to parse web socket data: JSON.parse: unexpected character at line 1 column 32 of the JSON data [{"loadpoints.0.chargeCurrents":}]`

Rootcause is that the charger is reporting nothing on P2 and P3:

`[lp-1  ] DEBUG 2022/09/24 20:04:50 charge currents: [6.13 NaN NaN]A`

### Steps to reproduce

Connect an Alfen Eve charger in connected to 1 phase.
Open the UI.

### Configuration details

```
levels:
  site: debug
  lp-1: debug

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

sponsortoken: 

site:
  - title: Home # display name for UI
    meters:
      grid: my_grid # grid meter reference
      pvs: my_pv # first pv meter reference
      batteries: my_battery # battery meter reference
    residualPower: 120
    bufferSoC: 80
    prioritySoC: 50

loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # Wallbox Referenz
    #vehicle: my_car # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)
    maxcurrent: 32
    phases: 1

chargers:
  - name: my_charger
    type: template
    template: alfen

    # Modbus TCP
    modbus: tcpip
    id: 1
    host:  # Hostname
    port: 502 # Port

meters:
  - name: my_grid
    type: template
    template: solaredge-hybrid
    usage: grid
    host:  # IP-Adresse oder Hostname
    port: 1502 # Port 502 (SetApp) oder 1502 (LCD) # Optional
    timeout: 10s # Optional
  - name: my_pv
    type: template
    template: solaredge-hybrid
    usage: pv
    host:  # IP-Adresse oder Hostname
    port: 1502 # Port 502 (SetApp) oder 1502 (LCD) # Optional
    timeout: 10s # Optional
  - name: my_battery
    type: template
    template: solaredge-hybrid
    usage: battery
    host:  # IP-Adresse oder Hostname
    port: 1502 # Port 502 (SetApp) oder 1502 (LCD) # Optional
    timeout: 10s # Optional
```

### Log details

```
Using config file:  /config/evcc.yaml
starting evcc: 'evcc --config  /config/evcc.yaml'
[main  ] INFO 2022/09/24 20:10:32 evcc 0.103
[main  ] INFO 2022/09/24 20:10:32 using config file: /config/evcc.yaml
[main  ] INFO 2022/09/24 20:10:32 listening at :7070
[lp-1  ] WARN 2022/09/24 20:10:34 locking phase config to 1p for switchable charger
[site  ] INFO 2022/09/24 20:10:34 site config:
[site  ] INFO 2022/09/24 20:10:34   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/09/24 20:10:34     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/09/24 20:10:34     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/09/24 20:10:34     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/09/24 20:10:34 loadpoint 1:
[lp-1  ] INFO 2022/09/24 20:10:34   mode:        pv
[lp-1  ] INFO 2022/09/24 20:10:34   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/09/24 20:10:34   meters:      charge ✓
[lp-1  ] INFO 2022/09/24 20:10:34     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2022/09/24 20:10:34 phase timer inactive
[lp-1  ] DEBUG 2022/09/24 20:10:34 pv timer inactive
[lp-1  ] DEBUG 2022/09/24 20:10:34 max charge current: 6A
[site  ] DEBUG 2022/09/24 20:10:34 ----
[lp-1  ] DEBUG 2022/09/24 20:10:34 charge power: 1433W
[site  ] DEBUG 2022/09/24 20:10:34 pv power: 10W
[site  ] DEBUG 2022/09/24 20:10:34 battery power: 0W
[site  ] DEBUG 2022/09/24 20:10:34 grid power: 1649W
[site  ] DEBUG 2022/09/24 20:10:34 battery soc 0: 0%
[site  ] DEBUG 2022/09/24 20:10:34 site power: 1769W
[lp-1  ] DEBUG 2022/09/24 20:10:34 charge currents: [6.16 NaN NaN]A
[lp-1  ] DEBUG 2022/09/24 20:10:35 charger status: C
[lp-1  ] DEBUG 2022/09/24 20:10:35 charge start energy: 6.21e+03kWh
[lp-1  ] INFO 2022/09/24 20:10:35 start charging ->
[lp-1  ] DEBUG 2022/09/24 20:10:35 charger disable
[lp-1  ] DEBUG 2022/09/24 20:10:35 wake-up timer: stop
[lp-1  ] DEBUG 2022/09/24 20:10:35 phase timer inactive
[lp-1  ] DEBUG 2022/09/24 20:10:35 pv timer elapse
[lp-1  ] DEBUG 2022/09/24 20:10:35 pv timer inactive
[lp-1  ] DEBUG 2022/09/24 20:10:35 switched phases: 1p
[site  ] DEBUG 2022/09/24 20:10:44 ----
```

### What type of operating system are you running?

Docker container

### Version

0.103

andig, 2022-09-24:
Loadpoint should convert all NaNs to zero.

treynaer, 2022-09-25:
I would expect the same but with a Modbus Poller connected I get following information on the registres containing the currents (320, 322, 324). No data is available on the registers matching P2 and P3.

[image]

Its the same for the Phase voltages (306,308,310):

[image]

andig, 2022-09-25:
Please

    evcc charger —log trace

treynaer, 2022-09-25:
```
[main  ] INFO 2022/09/25 10:45:48 evcc 0.103
[main  ] INFO 2022/09/25 10:45:48 using config file: /config/evcc.yaml
Power:          6W
Energy:         6223.1kWh
Current L1..L3: 0.04A NaNA NaNA
Charge status:  B
Enabled:        false
```

andig, 2022-09-25:
Log trace!

treynaer, 2022-09-25:
I understand, but setting log level to trace in config shows only an output up-to debug. Connection to sh on docker shows this output. I'm running evcc as Home Assistant Add-on.

treynaer, 2022-09-25:
This is better:

```
/config # evcc charger -l trace
[main  ] INFO 2022/09/25 11:01:48 evcc 0.103
[main  ] INFO 2022/09/25 11:01:48 using config file: /config/evcc.yaml
[alfen ] TRACE 2022/09/25 11:01:48 modbus: send 00 01 00 00 00 06 01 03 01 58 00 02
[alfen ] TRACE 2022/09/25 11:01:48 modbus: recv 00 01 00 00 00 07 01 03 04 40 c8 a3 d7
[alfen ] TRACE 2022/09/25 11:01:48 modbus: send 00 02 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2022/09/25 11:01:48 modbus: recv 00 02 00 00 00 0b 01 03 08 41 57 bd 3c 80 00 00 00
[alfen ] TRACE 2022/09/25 11:01:48 modbus: send 00 03 00 00 00 06 01 03 01 40 00 02
[alfen ] TRACE 2022/09/25 11:01:49 modbus: recv 00 03 00 00 00 07 01 03 04 3d 23 d7 0a
[alfen ] TRACE 2022/09/25 11:01:49 modbus: send 00 04 00 00 00 06 01 03 01 42 00 02
[alfen ] TRACE 2022/09/25 11:01:49 modbus: recv 00 04 00 00 00 07 01 03 04 ff ff ff ff
[alfen ] TRACE 2022/09/25 11:01:49 modbus: send 00 05 00 00 00 06 01 03 01 44 00 02
[alfen ] TRACE 2022/09/25 11:01:49 modbus: recv 00 05 00 00 00 07 01 03 04 ff ff ff ff
[alfen ] TRACE 2022/09/25 11:01:49 modbus: send 00 06 00 00 00 06 01 03 04 b1 00 05
[alfen ] TRACE 2022/09/25 11:01:49 modbus: recv 00 06 00 00 00 0d 01 03 0a 43 31 00 00 00 00 00 00 00 00
[alfen ] TRACE 2022/09/25 11:01:49 modbus: send 00 07 00 00 00 06 01 03 04 ba 00 02
[alfen ] TRACE 2022/09/25 11:01:49 modbus: recv 00 07 00 00 00 07 01 03 04 00 00 00 00
Power:          6W
Energy:         6223.1kWh
Current L1..L3: 0.04A NaNA NaNA
Charge status:  B
Enabled:        false
```

andig, 2022-09-25:
Die Ströme sind

```
[alfen ] TRACE 2022/09/25 11:01:48 modbus: send 00 03 00 00 00 06 01 03 01 40 00 02
[alfen ] TRACE 2022/09/25 11:01:49 modbus: recv 00 03 00 00 00 07 01 03 04 3d 23 d7 0a

[alfen ] TRACE 2022/09/25 11:01:49 modbus: send 00 04 00 00 00 06 01 03 01 42 00 02
[alfen ] TRACE 2022/09/25 11:01:49 modbus: recv 00 04 00 00 00 07 01 03 04 ff ff ff ff

[alfen ] TRACE 2022/09/25 11:01:49 modbus: send 00 05 00 00 00 06 01 03 01 44 00 02
[alfen ] TRACE 2022/09/25 11:01:49 modbus: recv 00 05 00 00 00 07 01 03 04 ff ff ff ff
```

closed 2022-09-25 by andig

MRstevice, 2023-01-27:
Dear community, I've just started with home assistant and activated evcc since yesterday . So apologies if I'm asking a stupid question here. However I have the same setup as above and receiving the same error. Do I need to change a file somewhere with the changes as provided in the fix via the HA file editor?

andig, 2023-01-27:
Please open new issue- this has been resolved months ago