evcc-io/evcc #4576: Orno OR-WE-514 Modbus Fehler (einphasiger Zähler)

L-Online, 2022-09-25:
### Describe the bug

Ich habe bei mir mehrere modbusfähige ORNO Zähler verbaut. die dreiphasigen funktionieren tadellos mit evcc. Nur der einphasige geht leider nicht.

### Steps to reproduce

Für die ORNO Zähler ist ein separater USB - RS485 Wandler vorhanden.

### Configuration details

pi@solaranzeige:/var/www/log $ cat /etc/evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 5s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1
  usage: grid
  host: 192.168.178.181
  name: grid1
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.178.181
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.184
  port: 502
  usage: pv
  modbus: tcpip
  name: pv3
- type: modbus
  model: orno3p
  id: 2
  device: /dev/ttyUSB1 # serial RS485 interface
  baudrate: 9600
  comset: "8E1"
  energy: Sum # default value, optionally override
  power: Power # default value, optionally override
  name: ORNO_SiS
- type: modbus
  model: orno3p
  id: 3
  device: /dev/ttyUSB1 # serial RS485 interface
  baudrate: 9600
  comset: "8E1"
  energy: Sum # default value, optionally override
  power: Power # default value, optionally override
  name: ErichMeter
- type: modbus
  model: orno1p
  id: 4
  device: /dev/ttyUSB1 # serial RS485 interface
  baudrate: 9600
  comset: "8E1"
  energy: Sum # default value, optionally override
  power: Power # default value, optionally override
  name: JuwelMeter

chargers:
- type: template
  template: evse-din
  id: 1
  device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1
  modbus: rs485serial
  name: wallbox3
- name: ErichCharger
  type: custom
  status:
   source: script
   cmd: /bin/sh -c 'echo C'
   timeout: 2s
  enabled:
   source: script
   cmd: /bin/sh -c 'echo 1'
   timeout: 2s
  enable:
   source: script
   cmd: /bin/sh -c 'echo 1'
   timeout: 2s
  maxcurrent:
   source: script
   cmd: /bin/sh -c 'echo ${maxcurrent}'
   timeout: 2s
- name: JuwelCharger
  type: custom
  status:
   source: script
   cmd: /bin/sh -c 'echo C'
   timeout: 2s
  enabled:
   source: script
   cmd: /bin/sh -c 'echo 1'
   timeout: 2s
  enable:
   source: script
   cmd: /bin/sh -c 'echo 1'
   timeout: 2s
  maxcurrent:
   source: script
   cmd: /bin/sh -c 'echo ${maxcurrent}'
   timeout: 2s

loadpoints:
- title: Garage
  charger: wallbox3
  meter: ORNO_SiS # charge meter
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true
- title: Erich
  charger: ErichCharger
  meter: ErichMeter # charge meter
  mode: now
  phases: 3
  mincurrent: 0
  maxcurrent: 32
  resetOnDisconnect: true
- title: Juwel
  charger: JuwelCharger
  meter: JuwelMeter # charge meter
  mode: now
  phases: 3
  mincurrent: 0
  maxcurrent: 32
  resetOnDisconnect: true

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    - pv3

# influx database
influx:
  url: http://localhost:8086
  database: evcc
  user:
  password:

### Log details

pi@solaranzeige:/var/www/log $ evcc meter -l trace
[main  ] INFO 2022/09/25 09:16:59 evcc 0.103
[main  ] INFO 2022/09/25 09:16:59 using config file: /etc/evcc.yaml
[main  ] FATAL 2022/09/25 09:16:59 cannot create meter 'JuwelMeter': cannot create meter 'modbus':

(Wenn ich einen 3 Phasigen Zähler in der YAML Konfiguriere (  "model: orno3p" statt   "model: orno1p", dann kann evcc das meter erzeugen, aber meckert halt wegen der Register):

 pi@solaranzeige:/var/www/log $ evcc meter -l trace
[main  ] INFO 2022/09/25 09:17:37 evcc 0.103
[main  ] INFO 2022/09/25 09:17:37 using config file: /etc/evcc.yaml
ErichMeter
----------
[modbus] TRACE 2022/09/25 09:17:37 modbus: send 03 03 00 1c 00 02 04 2f
[modbus] TRACE 2022/09/25 09:17:37 modbus: recv 03 03 04 40 2d f3 b6 99 7c
[modbus] TRACE 2022/09/25 09:17:37 modbus: send 03 03 01 00 00 02 c4 15
[modbus] TRACE 2022/09/25 09:17:37 modbus: recv 03 03 04 43 6e d9 9a 76 51
Power:  2718W
Energy: 238.9kWh

JuwelMeter
----------
[modbus] TRACE 2022/09/25 09:17:38 modbus: send 04 03 00 1c 00 02 05 98
[modbus] TRACE 2022/09/25 09:17:38 modbus: recv 04 83 02 d0 f0
[modbus] TRACE 2022/09/25 09:17:38 modbus: send 04 03 01 00 00 02 c5 a2
[modbus] TRACE 2022/09/25 09:17:38 modbus: recv 04 83 02 d0 f0
Power:  read failed: modbus: exception '2' (illegal data address), function '3'
Energy: read failed: modbus: exception '2' (illegal data address), function '3'

grid1
-----
[http  ] TRACE 2022/09/25 09:17:38 GET http://192.168.178.181/solar_api/v1/GetPowerFlowRealtimeData
[http  ] TRACE 2022/09/25 09:17:38 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 232,
               "E_Day" : 385,
               "E_Total" : 25983000,
               "E_Year" : 7009413,
               "P" : 730
            }
         },
         "Site" : {
            "E_Day" : 385,
            "E_Total" : 25983000,
            "E_Year" : 7009413,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : 1478.01,
            "P_Load" : -2208.0100000000002,
            "P_PV" : 730,
            "rel_Autonomy" : 33.061444468095701,
            "rel_SelfConsumption" : 100
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2022-09-25T09:17:38+02:00"
   }
}
Power: 1478W

pv2
---
[http  ] TRACE 2022/09/25 09:17:38 GET http://192.168.178.181/solar_api/v1/GetPowerFlowRealtimeData
[http  ] TRACE 2022/09/25 09:17:38 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 232,
               "E_Day" : 385,
               "E_Total" : 25983000,
               "E_Year" : 7009413,
               "P" : 730
            }
         },
         "Site" : {
            "E_Day" : 385,
            "E_Total" : 25983000,
            "E_Year" : 7009413,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : 1478.01,
            "P_Load" : -2208.0100000000002,
            "P_PV" : 730,
            "rel_Autonomy" : 33.061444468095701,
            "rel_SelfConsumption" : 100
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2022-09-25T09:17:38+02:00"
   }
}
Power: 730W

pv3
---
[modbus] TRACE 2022/09/25 09:17:38 modbus: send 00 01 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2022/09/25 09:17:38 modbus: recv 00 01 00 00 00 07 01 04 04 03 fd 00 00
[modbus] TRACE 2022/09/25 09:17:38 modbus: send 00 02 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2022/09/25 09:17:38 modbus: recv 00 02 00 00 00 07 01 04 04 ae ca 00 00
Power:  1021W
Energy: 4474.6kWh

ORNO_SiS
--------
[modbus] TRACE 2022/09/25 09:17:38 modbus: send 02 03 00 1c 00 02 05 fe
[modbus] TRACE 2022/09/25 09:17:38 modbus: recv 02 03 04 00 00 00 00 c9 33
[modbus] TRACE 2022/09/25 09:17:38 modbus: send 02 03 01 00 00 02 c5 c4
[modbus] TRACE 2022/09/25 09:17:38 modbus: recv 02 03 04 44 72 56 66 c2 52
Power:  0W
Energy: 969.3kWh

### What type of operating system are you running?

Linux

### Version

pi@solaranzeige:/var/www/log $ evcc -v
evcc version 0.103

andig, 2022-09-25:
Bitte

    evcc meter —log trace

L-Online, 2022-09-25:
`evcc meter -l trace`, wie von mir oben verwendet und und `evcc meter —log trace` machen doch das gleiche, oder?

andig, 2022-09-25:
Ahh, Moment. Du kannst den Zähler ja gar nicht anlegen. Das hatte ich missverstanden, schau gleich rein…

andig, 2022-09-25:
Ich bekomme erstmal:

```
gor main.go meter -c orno1p.yaml
[main  ] INFO 2022/09/25 10:59:39 evcc 0.0.1-alpha
[main  ] INFO 2022/09/25 10:59:39 using config file: orno1p.yaml
[main  ] FATAL 2022/09/25 10:59:39 cannot create meter 'JuwelMeter': cannot create meter 'modbus': invalid measurement for power: Power
exit status 1
```

Wieso fehlt das Detail bei Dir? Copy/paste Fehler?

andig, 2022-09-25:
Probier mal bitte `PowerL1` statt `Power`.

ochykysh, 2022-09-25:
Servus, die 514/15 Register Adressen stammen aus dieser PDF - https://b2b.orno.pl/download-resource/26063/ und 516/517 auf dieser https://b2b.orno.pl/download-resource/26064/

L-Online, 2022-09-25:
> Ich bekomme erstmal:
>
> ```
> gor main.go meter -c orno1p.yaml
> [main  ] INFO 2022/09/25 10:59:39 evcc 0.0.1-alpha
> [main  ] INFO 2022/09/25 10:59:39 using config file: orno1p.yaml
> [main  ] FATAL [...]

```
pi@solaranzeige:/var/www/log $ evcc meter -l trace
[main  ] INFO 2022/09/25 09:15:58 evcc 0.103
[main  ] INFO 2022/09/25 09:15:58 using config file: /etc/evcc.yaml
[main  ] FATAL 2022/09/25 09:15:58 cannot create meter 'JuwelMeter': cannot create meter 'modbus': invalid measurement for power: Power
pi@solaranzeige:/var/www/log $ evcc  -l trace

```

so ist es komplett :)

andig, 2022-09-25:
Ok, das passt.

> Probier mal bitte PowerL1 statt Power.

L-Online, 2022-09-25:
> Probier mal bitte `PowerL1` statt `Power`.

So startet es sauber:

```
pi@solaranzeige:/var/www/log $ evcc meter —l trace
[main  ] INFO 2022/09/25 11:51:54 evcc 0.103
[main  ] INFO 2022/09/25 11:51:54 using config file: /etc/evcc.yaml
ORNO_SiS
--------
Power:  0W
Energy: 969.3kWh

ErichMeter
----------
Power:  2019W
Energy: 241.9kWh

JuwelMeter
----------
Power:  0W
Energy: 111.5kWh

grid1
-----
Power: 1599W

pv2
---
Power: 1375W

pv3
---
Power:  1840W
Energy: 4477.0kWh

pi@solaranzeige:/var/www/log $ sudo systemctl start evcc

```
Vielen Dank :-)

andig, 2022-09-30:
Für's Erste nicht besser zu lösen. Der Workaround ist `PowerL1` zu verwenden.

closed 2022-09-30 by andig